zephyrproject-rtos/zephyr #54934: stm32u5, samples\boards\stm32\backup_sram not supported

JiiZR, 2023-02-17:
**Describe the bug**
It seems that support for U5 series BKPSRAM is missing.

I'am using b_u585i_iot02a board. My goal is to enable BKPSRAM using sample backup_sram. U5 is missing from
kconfig file so I added
|| SOC_SERIES_STM32U5X.

"
warning: STM32_BACKUP_SRAM (defined at soc/arm/st_stm32\common\Kconfig.soc:12) was assigned the
value 'y' but got the value 'n'. Check these unsatisfied dependencies: (SOC_SERIES_STM32F2X ||
SOC_SERIES_STM32F4X || SOC_SERIES_STM32F7X || SOC_SERIES_STM32H7X)
"

I also added to app.overlay:

	soc {
			backup_sram: memory@40036400 {
			compatible = "zephyr,memory-region", "st,stm32-backup-sram";
			reg = <0x40036400 DT_SIZE_K(2)>;
			clocks = <&rcc STM32_CLOCK_BUS_AHB1 0x10020000>;  /* BKPSRAMEN and RAMCFGEN */
			zephyr,memory-region = "BACKUP_SRAM";
			status = "disabled";
		};
	};
	
...

&backup_sram{
	status = "okay";
};

And now sample works. 

Also it would be nice to have driver support for BKPSRAM erase,
currently I'am using direct register writes to do it.	

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b b_u585i_iot02a_ns samples\boards\stm32\backup_sram

**Expected behavior**
build should be ok

**Impact**
medium

**Environment (please complete the following information):**
 - OS: Windows

FRASTM, 2023-02-17:
Thanks for reporting that missing node. 
I have put  a [branch](https://github.com/FRASTM/zephyr/tree/backup_sram) with slight changes to run the sample on the stm32u5 board.
$ west build -p auto -b nucleo_u575zi_q samples/boards/stm32/backup_sram
$ west flash
 
```
...
*** Booting Zephyr OS build v3.3.0-rc3-4-g1d84291b9fb5 ***                      
Current value in backup SRAM (0x0x40036400): 7                                  
Next reported value should be: 8                                                
Keep VBAT power source and reset the board now!                                 
*** Booting Zephyr OS build v3.3.0-rc3-4-g1d84291b9fb5 ***                      
Current value in backup SRAM (0x40036400): 8                                    
Next reported value should be: 9                                                
Keep VBAT power source and reset the board now!  
```

FRASTM, 2023-02-17:
Regarding the erase operation of the backup sram (at least) on the stm32u5x this is a soft erase (or of course a hw reset when VBat is off),  done during the  stm32_backup_sram_init since there is no API for that.
That could be a kconfig like CONFIG_ERASE_BKP_SRAM_AT_BOOT  or a property of the dts/bindings/memory-controllers/st,stm32-backup-sram.yaml

JiiZR, 2023-02-21:
Thanks for a fast response. Branch is also looking good. I'am aware that there is no API for  stm32_backup_sram_init but I was hoping that there will be at some day :)

JiiZR, 2023-02-22:
Proposed option is OK.