zephyrproject-rtos/zephyr #54937: can: iso-tp: implementation test failed on nucleo_g474re

FRASTM, 2023-02-17:
**Describe the bug**
Running the tests/subsys/canbus/isotp/implementation/ on the nucleo_g474re board, the test fails on
```
 - FAIL - [isotp_implementation.test_send_receive_net_single_blocks] duration = 1.066 seconds
 - FAIL - [isotp_implementation.test_send_receive_sf] duration = 1.013 seconds
 - FAIL - [isotp_implementation.test_send_receive_single_block] duration = 0.020 seconds

```

Please also mention any information which could help others to understand
the problem you're facing:
- could this error comes after the merge of https://github.com/zephyrproject-rtos/zephyr/pull/53634/
 
 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_g474re tests/subsys/canbus/isotp/implementation/
2. west flash
3. See error

**Expected behavior**
Test PASSED

**Impact**
Ci fails on this target paltform

**Logs and console output**
```
*** Booting Zephyr OS build v3.3.0-rc3-1-gbbefcd29a9dd ***
Running TESTSUITE isotp_implementation
===================================================================
START - test_bind_unbind
 PASS - test_bind_unbind in 2.509 seconds
===================================================================
START - test_buffer_allocation
 PASS - test_buffer_allocation in 1.551 seconds
===================================================================
START - test_buffer_allocation_wait
 PASS - test_buffer_allocation_wait in 10.151 seconds
===================================================================
START - test_send_receive_blocks
 PASS - test_send_receive_blocks in 0.454 seconds
===================================================================
START - test_send_receive_net_blocks
 PASS - test_send_receive_net_blocks in 1.019 seconds
===================================================================
START - test_send_receive_net_sf
 PASS - test_send_receive_net_sf in 0.006 seconds
===================================================================
START - test_send_receive_net_single_blocks

E: Reception of next FC has timed out
    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:289: isotp_implementation_test_send_receive_net_single_)
recv returned -14

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:55: send_complete_cb: (error_nr not equal to ISOTP_N_OK)
Sending failed (-2)
 FAIL - test_send_receive_net_single_blocks in 1.066 seconds
===================================================================
START - test_send_receive_sf

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:91: get_sf: (ret not equal to 1)
recv returned -14
 FAIL - test_send_receive_sf in 1.013 seconds
===================================================================
START - test_send_receive_single_block
E: No buffer for FF left

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:313: isotp_implementation_test_send_receive_single_bloc)
Binding failed (-11)
 FAIL - test_send_receive_single_block in 0.020 seconds
===================================================================
TESTSUITE isotp_implementation failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  66.67% [isotp_implementation]: pass = 6, fail = 3, skip = 0, total = 9 duration = 17.789 seconds
 - PASS - [isotp_implementation.test_bind_unbind] duration = 2.509 seconds
 - PASS - [isotp_implementation.test_buffer_allocation] duration = 1.551 seconds
 - PASS - [isotp_implementation.test_buffer_allocation_wait] duration = 10.151 seconds
 - PASS - [isotp_implementation.test_send_receive_blocks] duration = 0.454 seconds
 - PASS - [isotp_implementation.test_send_receive_net_blocks] duration = 1.019 seconds
 - PASS - [isotp_implementation.test_send_receive_net_sf] duration = 0.006 seconds
 - FAIL - [isotp_implementation.test_send_receive_net_single_blocks] duration = 1.066 seconds
 - FAIL - [isotp_implementation.test_send_receive_sf] duration = 1.013 seconds
 - FAIL - [isotp_implementation.test_send_receive_single_block] duration = 0.020 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain Zephyr SDK 0.15.0
 - Commit SHA v3.3.0-rc3   (bb8b74502091a90bff08de1b66b560a04b7c28e1)

henrikbrixandersen, 2023-02-22:
A Git bisect indeed points to 31dfd84fd5c89a068c00521902dcf47e1c295ed4 by @peter-mitsis as having introduced the failure.

peter-mitsis, 2023-02-22:
@henrikbrixander - Thanks for bringing this to my attention. I am taking a look at both the test and pipes code to see if I can spot anything.

henrikbrixandersen, 2023-02-22:
> @henrikbrixander - Thanks for bringing this to my attention. I am taking a look at both the test and pipes code to see if I can spot anything.

The failure seems to be board/SoC specific. The same test passes on at least the following boards:
- lpcxpresso55s16
- twr_ke18f
- mimxrt1024_evk
- frdm_k64f
- stm32f3_disco

peter-mitsis, 2023-02-22:
I don't have the hardware to test this specific failure. However, my initial investigation shows that the test does not use pipes (CONFIG_PIPES is not set). From this, the most likely culprit in the identified commit is its change to kernel/sched.c :: z_unpend_thread_no_timeout().

henrikbrixandersen, 2023-02-22:
> I don't have the hardware to test this specific failure. However, my initial investigation shows that the test does not use pipes (CONFIG_PIPES is not set). From this, the most likely culprit in the identified commit is its change to kernel/sched.c :: z_unpend_thread_no_timeout().

I can confirm that reverting the change to `kernel/sched.c` makes the test pass again on current `main` branch.

peter-mitsis, 2023-02-24:
While it is good to know that reverting that allows the test to pass again, it is not yet clear to me why it has that effect. As an aside, that particular addition to z_unpend_thread_no_timeout() was added to guard against a race condition in the pipes implementation.

My initial investigation into the kernel side of this issue shows that all uses of the `pended_on` field are guarded by the sched_spinlock (which on an UP system such as the nucleo degenerates into a mere interrupt lock). Unless there is some higher priority interrupt at play that is not masked by our standard interrupt locking (which is a scenario that I doubt is coming into play here), I can imagine a couple more potential conditions to search for that could lead to this behaviour. Unfortunately, I am not yet able to convert these ideas from abstract mental pictures into words for a proper description--but I will continue to dig.

FRASTM, 2023-03-08:
In any case the test is now PASSED since a while (SHA1 04386be57e8aa167ec7b3fd5748a6372fe305a2e)

peter-mitsis, 2023-03-20:
As a follow up to the addition to z_unpend_thread_no_timeout() change to guard against a pipes race condition ...

There have been some recent commits to the kernel to fix a similar type of race condition in kernel events that do not require that z_unpend_thread_no_timeout() change. I have a plan to leverage that solution and apply it to pipes, which then ought to remove the necessity of that previous change, and in turn allow for that change to be undone.

FRASTM, 2023-03-20:
OK, let's see on the next run

peter-mitsis, 2023-03-21:
FYI: I have posted PR #56038 that ought to be a cleaner fix for pipes, and one that does not that aforementionned `z_unpend_thread_no_timeout()` change.

FRASTM, 2023-04-17:
still failing

henrikbrixandersen, 2023-05-19:
> From one SHA to the other the test result is failed or passed (a question of memory mapping ?) For example on SHA1 [7bdae08](https://github.com/zephyrproject-rtos/zephyr/commit/7bdae081e3c0ad66e7edff0a03cfdcca2630672d) : failed but with a slight change like enabling a second can2 node to the nucleo_g474re.dts, the results is PASSED

This is very odd indeed.

henrikbrixandersen, 2023-06-01:
This appears to be fixed by #58731.

henrikbrixandersen, 2023-06-02:
@FRASTM I am closing this as I am unable to reproduce it after #58731 was merged. Please do not hesitate to reopen if you can still reproduce it - or if the problem reappears.

FRASTM, 2023-06-15:
Re-openning the issue : 
failure occurs  on SHA1 3104ad0e393c   when testing the tests/subsys/canbus/isotp/implementation/canbus.isotp.implementation, on the nucleo_g474re board
```

START - test_send_receive_single_block
E: Reception of next FC has timed out
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:321: isotp_implementation_test_send_receive_single_block: (ret not equal to send_len)
data should be received at once (ret: -14)
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:55: send_complete_cb: (error_nr not equal to ISOTP_N_OK)
Sending failed (-2)
FAIL - test_send_receive_single_block in 1.070 seconds
===================================================================
TESTSUITE isotp_implementation failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  88.89% [isotp_implementation]: pass = 8, fail = 1, skip = 0, total = 9 duration = 16.972 seconds
- PASS - [isotp_implementation.test_bind_unbind] duration = 2.510 seconds
- PASS - [isotp_implementation.test_buffer_allocation] duration = 1.551 seconds
- PASS - [isotp_implementation.test_buffer_allocation_wait] duration = 10.151 seconds
- PASS - [isotp_implementation.test_send_receive_blocks] duration = 0.461 seconds
- PASS - [isotp_implementation.test_send_receive_net_blocks] duration = 1.045 seconds
- PASS - [isotp_implementation.test_send_receive_net_sf] duration = 0.006 seconds
- PASS - [isotp_implementation.test_send_receive_net_single_blocks] duration = 0.172 seconds
- PASS - [isotp_implementation.test_send_receive_sf] duration = 0.006 seconds
- FAIL - [isotp_implementation.test_send_receive_single_block] duration = 1.070 seconds
------ TESTSUITE SUMMARY END ------
```

gramsay0, 2023-07-26:
> ```
> START - test_send_receive_single_block
> E: Reception of next FC has timed out
> Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:321: [...]

I think this is an issue with the test.

`isotp_send` is called on the same `struct isotp_send_ctx` before the last send is complete.

The test thread and the system workqueue have the same priority, so it is a bit arbitrary whether the workqueue cleans up the send context before the next send.

I think it shows up on MCAN because the loopback RX ISR occurs before the TX complete ISR (i.e. different order to can_loopback driver). The work queue yields after completing each work item, so the order is "WQ RX -> test recv + send -> WQ TX complete"

Try adding a "TX complete" semaphore to the test, and/or changing a thread priority (`CONFIG_ZTEST_THREAD_PRIORITY=0`)

FRASTM, 2023-07-28:
Changing the thread priority fixes the pb
```
diff --git a/tests/subsys/canbus/isotp/implementation/prj.conf b/tests/subsys/canbus/isotp/implementation/prj.conf
index af7b76b9fc3..4ab0fe51d38 100644
--- a/tests/subsys/canbus/isotp/implementation/prj.conf
+++ b/tests/subsys/canbus/isotp/implementation/prj.conf
@@ -7,3 +7,4 @@ CONFIG_ISOTP_ENABLE_CONTEXT_BUFFERS=y
 CONFIG_ISOTP_RX_BUF_COUNT=2
 CONFIG_ISOTP_RX_BUF_SIZE=56
 CONFIG_ISOTP_RX_SF_FF_BUF_COUNT=2
+CONFIG_ZTEST_THREAD_PRIORITY=0
```

closed 2023-07-31 by carlescufi