zephyrproject-rtos/zephyr #54955: RTT UART all channels write to channel 0

jm-deltaxp, 2023-02-18:
The Segger RTT UART driver appears to support multiple channels, however all the enumerated channels in my dts overlay write to channel 0.

 - Target platform nucleo-f411re with j-link onboard
 - Workaround: modify drivers/serial/uart_rtt.c by adding line after line 204:
     .channel = idx \

henrikbrixandersen, 2023-02-21:
Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

jm-deltaxp, 2023-02-21:
This bug looks like it was introduced when the driver was being migrated from using Kconfig symbols to devicetree here:
https://github.com/zephyrproject-rtos/zephyr/commit/3e33d73381ec76926900848a73b625bda5a6a6c1#diff-740da175c479bb6d24b3a5bec76e947469697828eb3483493992fdc010fb7212L98

It looks like there are no test cases that would test this configuration and does not seem to be a popular use case as this was refactored a couple years ago. Can the author @JordanYates review this issue?

JordanYates, 2023-02-25:
Thanks for reporting this, I have opened a PR that should fix this issue.

JordanYates, 2023-02-25:
I see that you had solved the issue yourself, I would encourage you to submit the fix yourself in the future.
https://docs.zephyrproject.org/latest/contribute/guidelines.html

PR #55183 (2023-02-25): drivers: serial: uart_rtt: fix multi-channel
Populate the `channel` index when constructing configuration structs for secondary RTT channels. Originally missed in #27704.

Fixes #54955.