zephyrproject-rtos/zephyr #54983: qemu: testing: qemu_kvm_arm64 is not runnable

PerMac, 2023-02-20:
**Describe the bug**
It seems wemu_kvm_arm64 definition https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm64/qemu_kvm_arm64/qemu_kvm_arm64.yaml is missing `simulation: qemu` entry. Without it twister doesn't know how to execute tests on this platform and always finish at building. I wanted to fix the issue by adding the missing field and the qemu was started, but always finishes with an error:

```
DEBUG   - run test: qemu_kvm_arm64/tests/kernel/common/kernel.common.nano64
DEBUG   - Spawning QEMUHandler Thread for qemu_kvm_arm64/tests/kernel/common/kernel.common.nano64
DEBUG   - Running qemu_kvm_arm64/tests/kernel/common/kernel.common.nano64 (qemu)
DEBUG   - Spawning QEMUHandler Thread for qemu_kvm_arm64/tests/kernel/common/kernel.common.nano64
DEBUG   - QEMU (27711) complete (unexpected eof) after 0.0017654895782470703 seconds
DEBUG   - No timeout, return code from QEMU (None): 1
DEBUG   - return code from QEMU (None): 1
DEBUG   - run status: qemu_kvm_arm64/tests/kernel/common/kernel.common.nano64 failed
```

**To Reproduce**
`./scripts/twister -T samples/hello_world/ --retry-failed 1 -v -p qemu_kvm_arm64`

**Impact**
Results from qemu_kvm_arm64 might give false negative impressions. Twister reports passes but these are always build-only.

nashif, 2023-02-22:
This target is not for simulation, it runs on top of a hypervisor and is not to be confused with other targets qemu we have in zephyr.

closed 2023-02-22 by nashif

PerMac, 2023-02-22:
I still don't understand this, sorry. The doc for qemu_kvm_arm64 says that for "programing and debugging refer to the qemu_cortex_a53 board instructions for this part." https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm64/qemu_kvm_arm64/doc/index.rst#programming-and-debugging and qemu_cortex_a53 works as regular qemu., i.e. twister builds and executes images on this board by default. Therefore I thought qemu_kvm_arm64 is like qemu_cortex_a53 
I don't understand what ` runs on top of a hypervisor ` means. Do you mean there is some extra process for this target, which is not part of regular twister run, which uses images built for qemu_kvm_arm64? 
I am trying to make sure that it is by design that qemu_kvm_arm64 is treated in twister always  as a "non-runnable" target, i.e. twister will always end after building (no execution).