zephyrproject-rtos/zephyr #54986: net: mgmt: Adding events to mgmt queue should not wait for lock with thread

krish2718, 2023-02-20:
Adding any network management events in `mgmt_push_event` currently waits for `net_mgmt_lock`, so, in case the callbacks take too long the event will not get added, this is not a problem in itself, but the if the events are being added using system workqueue this stalls the system and if callbacks also rely on system workqueue it can lead to deadlocks.

We need to come up with a lockless design for adding events, e.g., FIFO where events can be added and `mgmt_thread` processes them.

rlubos, 2023-02-21:
It seems that this particular concern should be addressed with https://github.com/zephyrproject-rtos/zephyr/pull/54701. Once this PR gets merged, adding new events is no longer affected by the mutex protecting callbacks.

rlubos, 2023-11-08:
Fixed with https://github.com/zephyrproject-rtos/zephyr/pull/54701.

closed 2023-11-08 by rlubos