zephyrproject-rtos/zephyr #54990: Zephyr 3.4 Release Checklist

stephanosio, 2023-02-20:
- [ ] Make sure the release plan is kept up to date:
   - See https://github.com/orgs/zephyrproject-rtos/projects/13

- [ ] Pre RC1 Steps
   - [ ] Create v3.4.1 and v3.5.0 milestone so that features that don't make it to the release can be assigned a milestone
   - [ ] Verify that all security-critical external components and external dependencies are up to date (previous: https://github.com/zephyrproject-rtos/zephyr/issues/49678)
   - [ ] Check known vulnerabilities, and fix exploitable vulnerabilities or verify them as un-exploitable (ask Security working group).
   - [ ] Make sure there's no pending ["Release Blocker"](https://github.com/zephyrproject-rtos/zephyr/issues?q=is%3Aissue+is%3Aopen+label%3A%22Release+Blocker%22) issues

See https://docs.zephyrproject.org/latest/project/release_process.html#tagging for tagging details, release specific filters under https://github.com/zephyrproject-rtos/zephyr/wiki/Filters to find pending PRs.

--------
- [ ] Release Notes WIP
  - [ ] Make sure to tell everybody to use the **past tense** (i.e. "X has been added" or "X is now Y")
  - [ ] Create draft `doc/releases/release-notes-3.4.rst` with feature summary from the previous one (if not there already)
  - [ ] Update `doc/releases/release-notes-3.4.rst` detailed sections (relative to 3.3.0 release) as important issues are addressed (subsystem owners). (previous: https://github.com/zephyrproject-rtos/zephyr/issues/54495)
--------
- [ ] Finalize Release Notes: (previous: https://github.com/zephyrproject-rtos/zephyr/pull/54904)
  - [ ] Add list of addressed issues (spell check issue list and fix in GitHub as needed) using `scripts/release/list_issues.py`
  - [ ] Add list of known issues (spell check issue list and fix in GitHub as needed) by downloading the [latest bug pickle](https://github.com/zephyrproject-rtos/zephyr/actions/workflows/bug_snapshot.yaml) and using `scripts/dump_bugs_pickle.py`
  - [ ] Review `doc/releases/release-notes-3.4.rst` overview summary and details sections
    - [ ] Request summary details from subsystem maintainers
- [ ] Update `doc/develop/api/overview.rst` (previous: https://github.com/zephyrproject-rtos/zephyr/pull/50604)
- [ ] Update `doc/conf.py` (add version to relevant list in `html_context`)
- [ ] Update `doc/LICENSING.rst` for new components that do not have a license
--------
- [ ]  Remaining Release Notes (previous: https://github.com/zephyrproject-rtos/zephyr/issues/54495)
- [ ] Check the CI [bi-weekly](https://github.com/zephyrproject-rtos/zephyr/actions/workflows/twister.yaml?query=event%3Aschedule) run for build failures that should be fixed
--------
- [ ] Update version in VERSION ([EXTRAVERSION must be empty, not zero 0](https://docs.zephyrproject.org/latest/project/release_process.html#tagging)) (previous: https://github.com/zephyrproject-rtos/zephyr/pull/54950)
- [ ] Tag both v3.4.0 (see https://docs.zephyrproject.org/latest/project/release_process.html#tagging)
- [ ] Update and publish the [draft GitHub release](https://github.com/zephyrproject-rtos/zephyr/releases) created by the CI.
- [ ] Tag v3.4.0 in [example-application](https://github.com/zephyrproject-rtos/example-application) (previous: https://github.com/zephyrproject-rtos/example-application/pull/31)
--------
- [ ] After Release
  - [ ] Update patchlevel in VERSION (PATCHLEVEL = 99) (previous: https://github.com/zephyrproject-rtos/zephyr/pull/54956)
  - [ ] Create `v3.4-branch` branch for 3.4.x releases (from v3.4.0 tag, don't include the switch to patchlevel 99)
  - [ ] Create `backport v3.4-branch` label
  - [ ] Move all PRs/Issues in the v3.4.0 milestone to the next milestone
  - [ ] Close v3.4.0 milestone 
  - [ ] Update [current release](https://www.zephyrproject.org/developer-resources/#current-release) and GitHub stats on the website (email Brett)
  - [ ] Email devel@lists.zephyrproject.org & announce@lists.zephyrproject.org lists about release and announce the opening of the merge window
  - [ ] Create checklist issue for next release
  - [ ] Add a comment to this issue linking to the next release checklist issue
  - [ ] Marketing Blog for announcing release (https://www.zephyrproject.org/blog)
  - [ ] Store the number of currently open PRs to keep track of post-release PR overhead over time
  - [ ] Update MAINTAINERS.yml "Release Notes" maintainers to point at the next release owners
  - [ ] Update current and supported version information in the [Security Policy](https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/SECURITY.md).

Previous checklist: https://github.com/zephyrproject-rtos/zephyr/issues/50898

nashif, 2023-02-21:
Retrospective:
- submitting issues for each PR or change in the stabilisation phase. 
- Sometimes it is an overkill to ask for trivial changes
- Release blocker definition needs to be refined and clarified
-

kartben, 2023-05-16:
please see #57966

kartben, 2023-05-31:
Re: "Update doc/develop/api/overview.rst" I just opened #58487. Would also agree with @fabiobaltieri's comments in #50604 that this list might not be all that helpful in the end (is it even complete?), and we should probably consider implementing something along the lines of https://github.com/zephyrproject-rtos/zephyr/pull/50604#issuecomment-1257681337

closed 2023-08-14 by nashif