zephyrproject-rtos/zephyr #55013: Can't boot when enabling CONFIG_ENTROPY_GENERATOR in Zephyr v3.3.0 (STM32)

liamclarkza, 2023-02-20:
**Describe the bug**
After updating to Zephyr v3.3.0 from v3.2.0, my applications seem to break when CONFIG_ENTROPY_GENERATOR is enabled. I am using a custom STM32 board which has been tested and is working with Zephyr v3.2.0. 
When debugging, it seems that the device gets stuck trying to read a random byte from the random number generator which has the CECS (clock error current status) flag set.

Additional information:
 - Platform/SoC: STM32L471RG
 - Workarounds: The issue seems to be related to the STM32 entropy driver (entropy_stm32.c) issue where the clocks aren't being configured correctly. This can be seen from the CECS (clock error current status) bit in the RNG's status register, which is set to 1. You can bypass this using the test random generator and disabling the devices random number generator in the devicetree, in which case the issue goes away
 - Is this a regression? Yes.

**To Reproduce**
Steps to reproduce the behavior:
Using an STM32L4 SoC, using the hello_world example as a starting point:
1. Enable the random number generator in a DTS overlay:
```
&rng {
	status = "okay";
};
```
2. In prj.conf, set: `CONFIG_ENTROPY_GENERATOR=y`

**Expected behavior**
The application should pass boot stage and run.

**Impact**
Largely a showstopper as this prevents me doing anything that requires a random number generator.

**Environment (please complete the following information):**
 - OS: macos
 - Toolchain: Zephyr v15.2
 - Zephyr version tag: v3.3.0

FRASTM, 2023-02-21:
The RNG is sourced by the MSI 48MHz from the  dts/arm/st/l4/stm32l4.dtsi
Is the clock MSI configured for 48MHz in your board DTS 
```
&clk_msi {
	status = "okay";
	msi-pll-mode;
	msi-range = <11>; /* 48MHz USB bus clk */
};
```

liamclarkza, 2023-02-21:
Ah, yes that fixes things. Sorry, I see I missed the note regarding entropy devices in the v3.3.0 release notes. Thanks for the help!

closed 2023-02-21 by liamclarkza