zephyrproject-rtos/zephyr #55040: Stack size revert in `samples/philosophers` break some in-tree boards

fkokosinski, 2023-02-21:
**Describe the bug**
Found with Renode while working on Renodepedia.

Changes introduced in #54969 break the Dining Philosophers demo for the following in-tree platforms:
* actinius_icarus
* circuitdojo_feather_nrf9160
* nrf5340_audio_dk_nrf5340_cpuapp
* nrf5340dk_nrf5340_cpuapp

**To Reproduce**
Build and flash (or upload the sample by other means) the sample:
```west build -p -b <board_name> samples/philosophers```

**Expected behavior**
Philosopher statuses should be printed out.

**Impact**
The demo doesn't work properly.

**Logs and console output**
The demo freezes after printing the demo description section; no output for philosophers with `STACK_SIZE=1024`:
```
Demo Description
----------------
An implementation of a solution to the Dining Philosophers
problem (a classic multi-thread synchronization problem).
This particular implementation demonstrates the usage of multiple
preemptible and cooperative threads of differing priorities, as
well as dynamic mutexes and thread sleeping.
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: 0de8c6bf29e53f0b51cfc598739fbf70826bac53

carlocaione, 2023-02-21:
@SakuRautio-ext FYI

SakuRautio-ext, 2023-02-22:
I think the commit should be reverted and we can investigate the issue further internally.

closed 2023-02-22 by nashif