zephyrproject-rtos/zephyr #55066: tests: kernel: timer: timer_behavior: kernel.timer.timer test_jitter_drift fails

nordic-piks, 2023-02-22:
**Describe the bug**
The tests/kernel/timer/timer_behavior/kernel.timer.timer test_jitter_drift fails

Observed for:
- nrf52840dk_nrf52840
- nrf52dk_nrf52832
- nrf9160dk_nrf9160
- nrf5340dk_nrf5340_cpuapp_ns
- nrf5340dk_nrf5340_cpuapp

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/kernel/timer/timer_behavior -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-02-21T20:35:39.527Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.3.0-237-g0151368a92a0 ***[0m
[2023-02-21T20:35:39.527Z] [0mRunning TESTSUITE timer_jitter_drift[0m
[2023-02-21T20:35:39.527Z] [0m===================================================================[0m
[2023-02-21T20:35:39.527Z] [0mSTART - test_jitter_drift[0m
[2023-02-21T20:35:39.527Z] [0mperiodic timer behavior test (approx 10 seconds)[0m
[2023-02-21T20:35:39.527Z] [0mperiodic timer samples gathered, calculating statistics[0m
[2023-02-21T20:35:39.527Z] [0mtimer clock rate 32768, kernel tick rate 32768[0m
[2023-02-21T20:35:39.527Z] [0mAssertion failed at ../../../../../../../../../zephyr/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:177: timer_jitter_drift_test_jitter_drift: expected_period_drift == 0.0 is false[0m
[2023-02-21T20:35:39.527Z] [0mFAIL - test_jitter_drift in 10.234 seconds[0m
[2023-02-21T20:35:39.527Z] [0m===================================================================[0m
[2023-02-21T20:35:39.527Z] [0mTESTSUITE timer_jitter_drift failed.[0m
[2023-02-21T20:35:39.527Z] [0mRunning TESTSUITE timer_tick_train[0m
[2023-02-21T20:35:39.527Z] [0m===================================================================[0m
[2023-02-21T20:35:39.527Z] [0mSTART - test_one_tick_timer_train[0m
[2023-02-21T20:35:39.527Z] [0mInitializing 4 Timers, Tick Rate 32768Hz, Expecting 81920 callbacks in 11000 ms[0m
[2023-02-21T20:35:39.527Z] [0mStarting Timers with Skews[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 0[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 5197[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 10402[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 15609[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 20817[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 26022[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 31224[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 36428[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 41633[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 46837[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 52047[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 57249[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 62453[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 67657[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 72862[0m
[2023-02-21T20:35:39.527Z] [0mFaking busy work, remaining timers is 4, timer callbacks 78067[0m
[2023-02-21T20:35:39.527Z] [0mOne Tick Timer Train Done, took 10066 ms, busy loop ran 16 times[0m
[2023-02-21T20:35:39.527Z] [0mPerfect delta 4 cycles or 122 us[0m
[2023-02-21T20:35:39.527Z] [0mTimer 0 max delta 5 cycles or 153 us, 32769 late callbacks (2.5%)[0m
[2023-02-21T20:35:39.527Z] [0mTimer 1 max delta 5 cycles or 153 us, 32769 late callbacks (2.5%)[0m
[2023-02-21T20:35:39.527Z] [0mTimer 2 max delta 5 cycles or 153 us, 32768 late callbacks (2.5%)[0m
[2023-02-21T20:35:39.527Z] [0mTimer 3 max delta 5 cycles or 153 us, 32768 late callbacks (2.5%)[0m
[2023-02-21T20:35:39.527Z] [0m!! Some ticks were missed.[0m
[2023-02-21T20:35:39.527Z] [0m!! Consider making CONFIG_SYS_CLOCK_TICKS_PER_SEC smaller.[0m
[2023-02-21T20:35:39.527Z] [0m!! The busy loop didn't run as much as expected.[0m
[2023-02-21T20:35:39.527Z] [0m!! Consider making CONFIG_SYS_CLOCK_TICKS_PER_SEC smaller.[0m
[2023-02-21T20:35:39.527Z] [0mPASS - test_one_tick_timer_train in 10.125 seconds[0m
[2023-02-21T20:35:39.527Z] [0m===================================================================[0m
[2023-02-21T20:35:39.527Z] [0mTESTSUITE timer_tick_train succeeded[0m
[2023-02-21T20:35:39.527Z] [0m------ TESTSUITE SUMMARY START ------[0m
[2023-02-21T20:35:39.527Z] [0mSUITE FAIL -   0.00% [timer_jitter_drift]: pass = 0, fail = 1, skip = 0, total = 1 duration = 10.234 seconds[0m
[2023-02-21T20:35:39.527Z] [0m- FAIL - [timer_jitter_drift.test_jitter_drift] duration = 10.234 seconds[0m
[2023-02-21T20:35:39.527Z] [0mSUITE PASS - 100.00% [timer_tick_train]: pass = 1, fail = 0, skip = 0, total = 1 duration = 10.125 seconds[0m
[2023-02-21T20:35:39.527Z] [0m- PASS - [timer_tick_train.test_one_tick_timer_train] duration = 10.125 seconds[0m
[2023-02-21T20:35:39.527Z] [0m------ TESTSUITE SUMMARY END ------[0m
[2023-02-21T20:35:39.527Z] [0m===================================================================[0m
[2023-02-21T20:35:39.527Z] [0mRunID: 4b5c4417d96ae3906303c681b63ef6f5[0m
[2023-02-21T20:35:39.527Z] [0mPROJECT EXECUTION FAILED[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.2
 - Commit SHA or Version used:  zephyr-v3.3.0-237-g0151368a92a0

nordic-piks, 2023-02-22:
Seems to be the same test like in  #54211 but different test case.

teburd, 2023-02-28:
Testing this on my nrf52840dk I see the timer behavior tests passing on main as of commit 9f46db90c832e56363c5d7bb42892651b39b271d

nashif, 2023-02-28:
then let's close this.

closed 2023-02-28 by nashif

nordic-piks, 2023-03-01:
Working fine.