zephyrproject-rtos/zephyr #55076: up_squared: zbus: work_queue: sample output mangling fails regex verification

KamilxPaszkiet, 2023-02-22:
The work_queue test requires orderly console output to verify the proper operation of the program via RegEx. The bug mangles the output of many output lines in total, making RegEx not recognize the required lines and timing out the test.

part of the log:
```
I: Sensor msg processed by CALLBACK fh1: temp = 10, press = 1, humidity = 100
I: Sensor msg processed by CALLBACK fh2: temp = 10, press = 1, humidity = 100
I: Sensor msg processed by CALLBACK fh3: temp = 10, press = 1, humidity = 100
=I::  SSeennssoorr  mmssgg  pprroocceesssseedd  bbyy  WTOHRRKE AQDU EhUaEn dhlaenrd l1e:r  tdehm1p:  =t e1m0p,  =p r1e0s,s  p=r e1s,s  h=u m1i,d ihtuym i=d i1t0y0 
e I1:0 0S
inIs:o rS emnssgo rp rmoscge spsreodc ebsys eTdH RbEyA DW OhRaKn dQlUeErU E2 :h atnedmlpe r=  d1h02,:  ptreemsps  ==  110,,  hpurmeisdsi t=y  1=,  1h0u0m
mdIi:t yS e=n s1o0r0 
1sIg:  pSreoncseosrs emds gb yp rToHcReEsAsDe dh abnyd lWeOrR K3 :Q UtEeUmEp  h=a n1d0l,e rp rdehs3s:  =t e1m,p  h=u m1i0d,i tpyr e=s s1 0=0 
 , humidity = 100
I: Sensor msg processed by CALLBACK fh1: temp = 20, press = 2, humidity = 200
I: Sensor msg processed by CALLBACK fh2: temp = 20, press = 2, humidity = 200
I: Sensor msg processed by CALLBACK fh3: temp = 20, press = 2, humidity = 200
=I::  SSeennssoorr  mmssgg  pprroocceesssseedd  bbyy  WTOHRRKE AQDU EhUaEn dhlaenrd l1e:r  tdehm1p:  =t e2m0p,  =p r2e0s,s  p=r e2s,s  h=u m2i,d ihtuym i=d i2t0y0 
e I2:0 0S
inIs:o rS emnssgo rp rmoscge spsreodc ebsys eTdH RbEyA DW OhRaKn dQlUeErU E2 :h atnedmlpe r=  d2h02,:  ptreemsps  ==  220,,  hpurmeisdsi t=y  2=,  2h0u0m
mdIi:t yS e=n s2o0r0 
2sIg:  pSreoncseosrs emds gb yp rToHcReEsAsDe dh abnyd lWeOrR K3 :Q UtEeUmEp  h=a n2d0l,e rp rdehs3s:  =t e2m,p  h=u m2i0d,i tpyr e=s s2 0=0 
 , humidity = 200
I: Sensor msg processed by CALLBACK fh1: temp = 30, press = 3, humidity = 300
I: Sensor msg processed by CALLBACK fh2: temp = 30, press = 3, humidity = 300
I: Sensor msg processed by CALLBACK fh3: temp = 30, press = 3, humidity = 300
=I::  SSeennssoorr  mmssgg  pprroocceesssseedd  bbyy  WTOHRRKE AQDU EhUaEn dhlaenrd l1e:r  tdehm1p:  =t e3m0p,  =p r3e0s,s  p=r e3s,s  h=u m3i,d ihtuym i=d i3t0y0 
e I3:0 0S
inIs:o rS emnssgo rp rmoscge spsreodc ebsys eTdH RbEyA DW OhRaKn dQlUeErU E2 :h atnedmlpe r=  d3h02,:  ptreemsps  ==  330,,  hpurmeisdsi t=y  3=,  3h0u0m
mdIi:t yS e=n s3o0r0 
3sIg:  pSreoncseosrs emds gb yp rToHcReEsAsDe dh abnyd lWeOrR K3 :Q UtEeUmEp  h=a n3d0l,e rp rdehs3s:  =t e3m,p  h=u m3i0d,i tpyr e=s s3 0=0 
 , humidity = 300
```
This line is corrupted:
```
2sIg:  pSreoncseosrs emds gb yp rToHcReEsAsDe dh abnyd lWeOrR K3 :Q UtEeUmEp  h=a n2d0l,e rp rdehs3s:  =t e2m,p  h=u m2i0d,i tpyr e=s s2 0=0 
 , humidity = 200
```
And it should looks like:
```
"I: Sensor msg processed by THREAD handler 3: temp = 20, press = 2, humidity = 200"
```

**To Reproduce**
Steps to reproduce the behavior:

1. Connect at least one UP Squared board to your controller and map it

1. Run the CHRE test
`<path/to>/zephyrproject/zephyr$ ./scripts/twister --device-testing --hardware-map <path/to>/<up2.map> -T samples/subsys/zbus/work_queue/ -vv`

1. The test should fail.

**Expected behavior**
The test should pass.

**Impact**
Currently, annoyance and lack of clear test output. However, name mangling while using RegEx to verify program operation can lead to further test failures and thus useless tests.

**Logs and console output**
```
I: Sensor msg processed by CALLBACK fh1: temp = 10, press = 1, humidity = 100
I: Sensor msg processed by CALLBACK fh2: temp = 10, press = 1, humidity = 100
I: Sensor msg processed by CALLBACK fh3: temp = 10, press = 1, humidity = 100
=I::  SSeennssoorr  mmssgg  pprroocceesssseedd  bbyy  WTOHRRKE AQDU EhUaEn dhlaenrd l1e:r  tdehm1p:  =t e1m0p,  =p r1e0s,s  p=r e1s,s  h=u m1i,d ihtuym i=d i1t0y0 
e I1:0 0S
inIs:o rS emnssgo rp rmoscge spsreodc ebsys eTdH RbEyA DW OhRaKn dQlUeErU E2 :h atnedmlpe r=  d1h02,:  ptreemsps  ==  110,,  hpurmeisdsi t=y  1=,  1h0u0m
mdIi:t yS e=n s1o0r0 
1sIg:  pSreoncseosrs emds gb yp rToHcReEsAsDe dh abnyd lWeOrR K3 :Q UtEeUmEp  h=a n1d0l,e rp rdehs3s:  =t e1m,p  h=u m1i0d,i tpyr e=s s1 0=0 
 , humidity = 100
I: Sensor msg processed by CALLBACK fh1: temp = 20, press = 2, humidity = 200
I: Sensor msg processed by CALLBACK fh2: temp = 20, press = 2, humidity = 200
I: Sensor msg processed by CALLBACK fh3: temp = 20, press = 2, humidity = 200
=I::  SSeennssoorr  mmssgg  pprroocceesssseedd  bbyy  WTOHRRKE AQDU EhUaEn dhlaenrd l1e:r  tdehm1p:  =t e2m0p,  =p r2e0s,s  p=r e2s,s  h=u m2i,d ihtuym i=d i2t0y0 
e I2:0 0S
inIs:o rS emnssgo rp rmoscge spsreodc ebsys eTdH RbEyA DW OhRaKn dQlUeErU E2 :h atnedmlpe r=  d2h02,:  ptreemsps  ==  220,,  hpurmeisdsi t=y  2=,  2h0u0m
mdIi:t yS e=n s2o0r0 
2sIg:  pSreoncseosrs emds gb yp rToHcReEsAsDe dh abnyd lWeOrR K3 :Q UtEeUmEp  h=a n2d0l,e rp rdehs3s:  =t e2m,p  h=u m2i0d,i tpyr e=s s2 0=0 
 , humidity = 200
I: Sensor msg processed by CALLBACK fh1: temp = 30, press = 3, humidity = 300
I: Sensor msg processed by CALLBACK fh2: temp = 30, press = 3, humidity = 300
I: Sensor msg processed by CALLBACK fh3: temp = 30, press = 3, humidity = 300
=I::  SSeennssoorr  mmssgg  pprroocceesssseedd  bbyy  WTOHRRKE AQDU EhUaEn dhlaenrd l1e:r  tdehm1p:  =t e3m0p,  =p r3e0s,s  p=r e3s,s  h=u m3i,d ihtuym i=d i3t0y0 
e I3:0 0S
inIs:o rS emnssgo rp rmoscge spsreodc ebsys eTdH RbEyA DW OhRaKn dQlUeErU E2 :h atnedmlpe r=  d3h02,:  ptreemsps  ==  330,,  hpurmeisdsi t=y  3=,  3h0u0m
mdIi:t yS e=n s3o0r0 
3sIg:  pSreoncseosrs emds gb yp rToHcReEsAsDe dh abnyd lWeOrR K3 :Q UtEeUmEp  h=a n3d0l,e rp rdehs3s:  =t e3m,p  h=u m3i0d,i tpyr e=s s3 0=0 
 , humidity = 300
```

**Environment:**
- OS: Ubuntu for the controller
- Toolchain: Zephyr SDK 15.0
- The current Zephyr version used

**Additional context**
Multiline RegEx used in the test in question:
```
regex:
      - "I: Sensor msg processed by CALLBACK fh1: temp = 10, press = 1, humidity = 100"
      - "I: Sensor msg processed by CALLBACK fh2: temp = 10, press = 1, humidity = 100"
      - "I: Sensor msg processed by CALLBACK fh3: temp = 10, press = 1, humidity = 100"
      - "I: Sensor msg processed by WORK QUEUE handler dh1: temp = 10, press = 1, humidity = 100"
      - "I: Sensor msg processed by WORK QUEUE handler dh2: temp = 10, press = 1, humidity = 100"
      - "I: Sensor msg processed by WORK QUEUE handler dh3: temp = 10, press = 1, humidity = 100"
      - "I: Sensor msg processed by THREAD handler 1: temp = 10, press = 1, humidity = 100"      
      - "I: Sensor msg processed by THREAD handler 2: temp = 10, press = 1, humidity = 100"      
      - "I: Sensor msg processed by THREAD handler 3: temp = 10, press = 1, humidity = 100"      
      - "I: Sensor msg processed by CALLBACK fh1: temp = 20, press = 2, humidity = 200"
      - "I: Sensor msg processed by CALLBACK fh2: temp = 20, press = 2, humidity = 200"
      - "I: Sensor msg processed by CALLBACK fh3: temp = 20, press = 2, humidity = 200"
      - "I: Sensor msg processed by WORK QUEUE handler dh1: temp = 20, press = 2, humidity = 200"
      - "I: Sensor msg processed by WORK QUEUE handler dh2: temp = 20, press = 2, humidity = 200"
      - "I: Sensor msg processed by WORK QUEUE handler dh3: temp = 20, press = 2, humidity = 200"
      - "I: Sensor msg processed by THREAD handler 1: temp = 20, press = 2, humidity = 200"      
      - "I: Sensor msg processed by THREAD handler 2: temp = 20, press = 2, humidity = 200"      
      - "I: Sensor msg processed by THREAD handler 3: temp = 20, press = 2, humidity = 200" 
```
The problem is similar to this issue:
[https://github.com/zephyrproject-rtos/zephyr/issues/54861](url)
[https://github.com/zephyrproject-rtos/zephyr/issues/55071](url)

nashif, 2023-02-22:
looks like an issue with SMP enabled?

rodrigopex, 2023-02-23:
Is this an issue from zbus or the log subsystem of that platform? 

All the other platform print that correctly. Should I change the zbus test or open an issue for the platform? Does log support SMP @nashif?

```
=I::  SSeennssoorr  mmssgg  pprroocceesssseedd  bbyy  WTOHRRKE AQDU EhUaEn dhlaenrd l1e:r  tdehm1p:  =t e1m0p,  =p r1e0s,s  p=r e1s,s  h=u m1i,d ihtuym i=d i1t0y0 
e I1:0 0S
inIs:o rS emnssgo rp rmoscge spsreodc ebsys eTdH RbEyA DW OhRaKn dQlUeErU E2 :h atnedmlpe r=  d1h02,:  ptreemsps  ==  110,,  hpurmeisdsi t=y  1=,  1h0u0m
mdIi:t yS e=n s1o0r0 
1sIg:  pSreoncseosrs emds gb yp rToHcReEsAsDe dh abnyd lWeOrR K3 :Q UtEeUmEp  h=a n1d0l,e rp rdehs3s:  =t e1m,p  h=u m1i0d,i tpyr e=s s1 0=0 
 , humidity = 100
```

rodrigopex, 2023-02-23:
@KamilxPaszkiet it would be hard (or forced) to change the regex to accept something like that.

gbarkadiusz, 2023-02-23:
According my investigation, this test passed after changing CONFIG_SMP parameter in 
`
<paht/to>/zephyr/boards/x86/up_squared/up_squared_defconfig
`
from `CONFIG_SMP=y` to `CONFIG_SMP=n`.
Moreover, test `<path/to>/tests/kernel/smp has a parameter `platform_allow: qemu_x86_64`. Which can means, that the up2 board doesn't support smp.

rodrigopex, 2023-06-20:
@SzymonRichert, it seems not to be a ZBus issue. Based on the @gbarkadiusz comment, up_squared maybe not support SMP. That can be a up_squared config issue. I have run the sample now, and it is running ok.

rodrigopex, 2023-09-18:
This is not a ZBus issue. The output is unordered, and there is no straightforward way of checking the output with regex.

closed 2023-09-18 by rodrigopex