zephyrproject-rtos/zephyr #55078: mec15xxevb_assy6853 : spi_quad.test_spi_octal_read and spi_quad.test_spi_quad_read failing

pkunieck, 2023-02-22:
**Describe the bug**
Tests spi_octal and spi_quad  from tests/boards/mec15xxevb_assy6853/qspi/ are failing, and this is output from console:

DEBUG   - DEVICE: SUITE FAIL -   0.00% [spi_quad]: pass = 0, fail = 2, skip = 0, total = 2 duration = 0.043 seconds
DEBUG   - DEVICE: - FAIL - [spi_quad.test_spi_octal_read] duration = 0.021 seconds
DEBUG   - DEVICE: - FAIL - [spi_quad.test_spi_quad_read] duration = 0.022 seconds

Assertion failed at WEST_TOPDIR/zephyr/tests/boards/mec15xxevb_assy6853/qspi/src/main.c:656: spi_quad_test_spi_octal_read: (memcmp(buffer_tx, safbuf2, BUF_SIZE) == 0 is false)
Buffer read data is different to write data
FAIL - test_spi_octal_read in 0.021 seconds

Assertion failed at WEST_TOPDIR/zephyr/tests/boards/mec15xxevb_assy6853/qspi/src/main.c:590: spi_quad_test_spi_quad_read: (memcmp(buffer_tx_2, safbuf2, BUF_SIZE_2) == 0 is false)
Buffer read data is different to write data
FAIL - test_spi_quad_read in 0.022 seconds

Problem seems to be old and I can't find a commit that cause the issue because of other problems in the past(no console output and some others in older commits).

**To Reproduce**
Run command on newest commit:
twister --hardware-map /home/zephyr/mec1.file --device-testing -T tests/boards/mec15xxevb_assy6853/qspi/ 

**Expected behavior**
Passing tests

**Impact**
annoyance

**Logs and console output**
See above.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK

pkunieck, 2023-05-18:
Can we have an update on the issue? These tests are still failing

Manimaran-A, 2023-05-24:
Added pinctrl property in overlay file to support spi quad mode.

PR #58231 (2023-05-24): xec_test_qspi_twister_fix
Updated the overlay file to enable spi quad IOs.
Fix: #55078
commit 224a7242 (2023-05-24): test: boards: qspi: mchp: Overlay file updated for Quad mode

Updated the overlay file to enable spi quad IOs.
Fix: #55078
closed 2023-05-24 by nashif