zephyrproject-rtos/zephyr #55085: npcx9m6f_evb: hanging evb while testing.

gbarkadiusz, 2023-02-22:
**Describe the bug**
While testing npx9m6f_evb, there are three test which causes hanging this evo board.
```
<path/to>/tests/kernel/tickless/tickless_concept/kernel.tickless.concept
<path/to>/tests/subsys/pm/device_runtime_api
<path/to>/tests/kernel/timer/timer_api/kernel.timer.tickless
```
After attempt of execute any of this tests, it is passing but cause Device issue (Flash error?) and evo board is hanging for another executation.
```
INFO    - 1/1 npcx9m6f_evb              <path/to>/tests/kernel/tickless/tickless_concept/kernel.tickless.concept FAILED Device issue (Flash error?) (device 0.354s)
```

**Temporary solutions**
The hanging is not permanently. There is the solution to bring nuvoton back to work.
To fix this board we should connect to target via jlink. With the proper commend_file.jlink. (Create it manually).

`
<path/to>/JLinkExe -CommandFile ./command_file.jlink 
`
Content of command_file.jlink
```
device cortex-m4
si 1
speed 4000
r
h
setPC 0x0c
go
exit
```
This jlink's script causes jump the current executing code into Hard_fault_handler. After that we can use npx9m6f_evb again.

The another solution can be adding name of platform as `platform_exclude` parameter to testcase.yml, cause these tests have already some exclusions.

**To Reproduce**
Steps to reproduce the behavior:
`
<path/to>/scripts/twister --device-testing --hardware-map <path/to>/map.yml -T <path/to>/tests/kernel/tickless/  -v
`

**Logs and console output**
For <path/to>/tests/kernel/timer/timer_api/kernel.timer.tickless
```
DEBUG   - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '/home/zephyr/zephyrproject/zephyr/twister-out/npcx9m6f_evb/tests/kernel/timer/timer_api/kernel.timer.tickless', '--runner', 'openocd']
DEBUG   - -- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/zephyr/zephyrproject/zephyr/twister-out/npcx9m6f_evb/tests/kernel/timer/timer_api/kernel.timer.tickless/zephyr/zephyr.npcx.hex

DEBUG   - halted
DEBUG   - run status: npcx9m6f_evb/tests/kernel/timer/timer_api/kernel.timer.tickless error
INFO    - 3/3 npcx9m6f_evb              tests/kernel/timer/timer_api/kernel.timer.tickless FAILED Device issue (Flash error?) (device 0.355s)
ERROR   - see: /home/zephyr/zephyrproject/zephyr/twister-out/npcx9m6f_evb/tests/kernel/timer/timer_api/kernel.timer.tickless/device.log
```
For <path/to>/tests/subsys/pm/device_runtime.api
```
DEBUG   - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '/home/zephyr/zephyrproject/zephyr/twister-out/npcx9m6f_evb/tests/subsys/pm/device_runtime_api/pm.device_runtime.api', '--runner', 'openocd']
DEBUG   - -- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/zephyr/zephyrproject/zephyr/twister-out/npcx9m6f_evb/tests/subsys/pm/device_runtime_api/pm.device_runtime.api/zephyr/zephyr.npcx.hex

DEBUG   - halted
DEBUG   - run status: npcx9m6f_evb/tests/subsys/pm/device_runtime_api/pm.device_runtime.api error
INFO    - 9/9 npcx9m6f_evb              tests/subsys/pm/device_runtime_api/pm.device_runtime.api FAILED Device issue (Flash error?) (device 0.479s)
ERROR   - see: /home/zephyr/zephyrproject/zephyr/twister-out/npcx9m6f_evb/tests/subsys/pm/device_runtime_api/pm.device_runtime.api/device.log
```
For <path/to>tests/kernel/tickless/tickless_concept/
```
DEBUG   - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '/home/zephyr/zephyrproject/zephyr/twister-out/npcx9m6f_evb/tests/kernel/tickless/tickless_concept/kernel.tickless.concept', '--runner', 'openocd']
DEBUG   - -- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/zephyr/zephyrproject/zephyr/twister-out/npcx9m6f_evb/tests/kernel/tickless/tickless_concept/kernel.tickless.concept/zephyr/zephyr.npcx.hex

DEBUG   - halted
DEBUG   - run status: npcx9m6f_evb/tests/kernel/tickless/tickless_concept/kernel.tickless.concept error
INFO    - 1/1 npcx9m6f_evb              tests/kernel/tickless/tickless_concept/kernel.tickless.concept FAILED Device issue (Flash error?) (device 0.357s)
ERROR   - see: /home/zephyr/zephyrproject/zephyr/twister-out/npcx9m6f_evb/tests/kernel/tickless/tickless_concept/kernel.tickless.concept/device.log
INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
```

**Environment:**
 - OS: Linux
 - Toolchain Zephyr SDK 15.0

MulinChao, 2023-02-23:
Hi @gbarkadiusz, thanks for providing this information. When the symptom occurred, we noticed the chip's power state is in [Deep Sleep](https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/nuvoton_npcx/common/power.c#L23) state if CONFIG_PM is enabled. Since the clock of core (Cortex-M4) is stopped, that's why it cannot handle `west flash` command properly. 

Usually, we suggest the user set pin 6 of SW1 to low to enable `FLPRG#` strap pin and press 1VCC1_RST1 button manually. Its functionality is to forbid the core from executing the firmware that enters deep sleep again. It is similar to what you did by `command_file.jlink`. I will discuss how to avoid this issue automatically with my colleague. Any update will inform you.

gbarkadiusz, 2023-02-23:
You're welcome, I was thinking about how to press "SW1" automatically. We can use FT232H module to set pin6 remotely. I know it may not be the best solution but it will probably work. It will be a similar solution with up_squared board.

MulinChao, 2023-02-23:
Add @Michal-Sch in this loop.

ChiHuaL, 2023-02-24:
I have the investigation on why `JLinkExe -CommandFile ./command_file.jlink ` has no such issue.
When JLinkExe polled the DAP fail, it will toggle the  pin 15 of the Jlink connector with a (1->0->1) pulse for 50 milli-second.
```
Connecting to target via SWD                               
Found SW-DP with ID 0x2BA01477
Failed to power up DAP   --> poll  DAP fail , toggle pin15                                   
Found SW-DP with ID 0x2BA01477
DPIDR: 0x2BA01477        
```
As the pin 15 of the Jlink connector connects to EC's VCC1_RST, It cause the EC resets and exits the deep sleep state.
As a result, JLinkExe successfully poll the DAP after the reset.
Looks like the JLinkExe did exactly the same idea that @gbarkadiusz mentioned above:
```
I was thinking about how to press "1VCC1_RST1" automatically. We can use FT232H module to press it remotely. I know it may not be the best solution but it will probably work. It will be a similar solution with MEC board.
```

galak, 2023-04-18:
@ChiHuaL, @MulinChao any updates on this?

gbarkadiusz, 2023-04-19:
Of course, we can use FT232H module to "press"  1VCC1_RST1 button automatically, but it needs added hardware setup.. I did propose to use jlink_script.

jgl-meta, 2023-06-20:
@MulinChao What is the status of this one? It was marked as stale, should it be closed?

ChiHuaL, 2023-07-03:
@gbarkadiusz Do you think this issue can be closed?

gbarkadiusz, 2023-07-03:
@ChiHuaL  According to my latest investigation, there is still problem with these tests.
So, I think that, it couldn't be closed.

gbarkadiusz, 2023-07-26:
I think that @ChiHuaL described this issue properly.

gbarkadiusz, 2024-02-02:
It is still reproducible.

gbarkadiusz, 2024-04-05:
Waiting for fix.
https://github.com/zephyrproject-rtos/zephyr/pull/70870

ChiHuaL, 2024-04-10:
This is not something that can be fixed easily by firmware   because the hardware/chip limitation.
As the npcx9 chip can support openocd to program the flash, I'd suggest to use the method mentioned in the [previous comment](https://github.com/zephyrproject-rtos/zephyr/issues/55085#issuecomment-1441295168) to bypass it.

gbarkadiusz, 2024-06-10:
@ChiHuaL So far it can be treating as fixed. https://github.com/zephyrproject-rtos/zephyr/pull/70870

closed 2024-06-10 by gbarkadiusz