zephyrproject-rtos/zephyr #55086: CAN API test fail on sam_e70_xplained board.

JoshNewell, 2023-02-22:
**Describe the bug**
On the '''sam_e70_xplained''' board when running the drivers/can/api test, it hangs on the test_filters_added_while_stopped test. It seems to occur when a CAN message is trying to be sent.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b sam_e70b_xplained tests/drivers/can/api
2. Open another terminal and execute: picocom -b 115200 /dev/ttyACM0
3. west flash
4. See error

**Expected behavior**
Build succeeds
Flash succeeds
Tests carry out as expected

**Impact**
When running the tests upon trying to send to the CAN, the program hangs.

**Logs and console output**
```
Running TESTSUITE can_api
===================================================================
START - test_add_filter
D: Attached std filter at 0
D: Attached ext filter at 0
D: Attached std filter at 0
D: Attached ext filter at 0
D: Attached std filter at 0
D: Attached ext filter at 0
 PASS - test_add_filter in 0.016 seconds
===================================================================
START - test_filters_added_while_stopped
D: Attached std filter at 0
D: Sending 8 bytes. Id: 0x555, ID type: standard 

```

**Environment (please complete the following information):**
 - OS: Fedora Linux 37
 - Toolchain: Zephyr SDK 0.15.2
 - Zephyr Version 3.2.0

henrikbrixandersen, 2023-02-22:
Unfortunately, I do not have this hardware available for testing. @JoshNewell Have you been able to bisect and find where the bug was introduced?

henrikbrixandersen, 2023-02-22:
Also, are you able to reproduce this issue on v3.3 (or even better, current `main`)?

JoshNewell, 2023-02-22:
I'll try it on the latest version and get the bisect over to you.

JoshNewell, 2023-02-22:
It seems the bug has been fixed in the v3.3, all the tests in the API pass, sorry for the trouble. Is there anything else you would like me to do?

henrikbrixandersen, 2023-02-22:
> It seems the bug has been fixed in the v3.3, all the tests in the API pass, sorry for the trouble. Is there anything else you would like me to do?

Thank you for verifying this. No, nothing more to do here.

closed 2023-02-22 by henrikbrixandersen