zephyrproject-rtos/zephyr #55110: tests: bluetooth: RX overflow in lt_tx_real_no_encode

kruithofa, 2023-02-23:
**Describe the bug**
PR #54119 describes a bug, but a proper fix requires more than is described in the PR.
when compiling tests/bluetooth/controller/ctrl_feature_exchange/bluetooth.controller.ctrl_feature_exchange.test without -ffreestanding (i.e., with memcpy warnings enabled), the following error pops up:

.../tests/bluetooth/controller/common/src/helper_util.c: In function ‘lt_tx_real_no_encode’: .../tests/bluetooth/controller/common/src/helper_util.c:412:9: error: ‘memcpy’ writing 39 bytes into a region of size 32 overflows the destination [-Werror=stringop-overflow=]
412 | memcpy((struct pdu_data *)rx->pdu, pdu, sizeof(struct pdu_data));
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.../tests/bluetooth/controller/common/src/helper_util.c:410:14: note: at offset 32 into destination object of size 64 allocated by ‘malloc’
410 | rx = malloc(PDU_RX_NODE_SIZE);
| ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors

Please also mention any information which could help others to understand
the problem you're facing:
 Most likely there is an issue with the define of PDU_RX_NODE_SIZE in helper_util.c  PDU_DATA_SIZE should use sizeof(struct pdu_data) 

**To Reproduce**
Steps to reproduce the behavior:
see above

**Expected behavior**
no compiler error should happen

**Impact**
low priority

**Logs and console output**
N/A

**Environment (please complete the following information):**
 N/A

**Additional context**
N/A

kruithofa, 2023-04-20:
Haven't reproduced the issue yet, continuing investigation.

keith-packard, 2023-04-20:
Yeah, I found the issue by building with picolibc, but it *should* happen whenever you build without `-ffreestanding`, however that requires a Kconfig hack to switch the default C library from minimal to picolibc as the minimal C library can't be used without that flag.

keith-packard, 2023-04-20:
I've managed to get the usual twister run to pass with the minimal C library without `-ffreestanding` (see #57099). Which means to me that this issue appears to no longer affect the test suite execution?

kruithofa, 2023-04-27:
upon code inspection we suspect that we also may use the wrong size definition. See PR#57328

PR #57328 (2023-04-27): Tests: Bluetooth: fix memcpy out of bounds error
In the helper-module we do a memcpy of a structure but the size is possibly incorrect.

This PR ensures that the correct sizes are used.

fixes #55110