zephyrproject-rtos/zephyr #55115: `DT_FOREACH_PROP_ELEM()` in a `zephyr,user` node inferred property does not work if a single item is supplied

xyzzy42, 2023-02-23:
**Describe the bug**
`DT_FOREACH_PROP_ELEM()` in a `zephyr,user` node inferred property does not work if a single phandle is used.

A related bug is this also does not work for a single string, but the underlying problem is different.

**To Reproduce**
```c
/ {
        user: zephyr,user {
            one-string = "one";
            two-string = "one", "two";
            one-handle = <&user>;
            two-handle = <&user>, <&user>;
        };
};

#define PHANDLE_TO_DEVICE(node_id, prop, idx) \
     DEVICE_DT_GET(DT_PHANDLE_BY_IDX(node_id, prop, idx)),

// This fails
const struct device *one_dev[] = { DT_FOREACH_PROP_ELEM(ZEPHYR_USER_NODE, one_handle, PHANDLE_TO_DEVICE) };
// This works
const struct device *two_dev[] = { DT_FOREACH_PROP_ELEM(ZEPHYR_USER_NODE, two_handle, PHANDLE_TO_DEVICE) };

// This also fails
const char *one_string[] = { DT_FOREACH_PROP_ELEM_SEP(ZEPHYR_USER_NODE, one_string, DT_PROP_BY_IDX, (,)) };
// This works
const char *two_string[] = { DT_FOREACH_PROP_ELEM_SEP(ZEPHYR_USER_NODE, two_string, DT_PROP_BY_IDX, (,)) };
```

**Expected behavior**
All should work.

**Impact**
Code that allows one or more items to be specified in zephyr,user nodes doesn't work.  It needs to be written to either allow exactly one item or two or more items.

**Additional context**
There are two different causes here.

For phandles, the issue is that the inferred type is `phandle` and not `phandles`.  While `phandle` is treated as an array of length 1, and will have `_IDX` and `_LEN` macros generated, the code to that generates the `_FOREACH_PROP_ELEM` macro does not do this.  It does not consider `phandle` to be an array.

For strings, the issue is similar, in that the inferred type is `string` and not `string-array`.  However, the `FOREACH_PROP_ELEM` code however *does* generate a macro for properties of type `string`.  But the rest of the macros, `_IDX` and `_LEN`, are not generated for `string` properties, so it still doesn't work.

xyzzy42, 2023-02-23:
I have a patch which makes phandle properties iterable with `DT_FOREACH_PROP_ELEM`.  This would also apply to properties with explicit bindings of type `phandle`.  `DT_PHANDLE_BY_IDX()` works on these properties, so it seems like foreach should work as well.

It doesn't fix the issue with string and string-array, or int and array.  These are not treated as iterable or as having a length with explicit bindings. So perhaps this is an issue with inferred bindings.

galak, 2023-02-23:
This may end up just being a limitation of `zephyr,user`.  Since we have no way of knowing what the type of the properties under `zephyr,user` are (by definition of `zephyr,user`).

I'm not sure how much pain it would cause to treat every singleton as an array instead to allow the ability to iterate over it.  @mbolivar-nordic thoughts?

xyzzy42, 2023-02-23:
`phandle` is a little different, in that it's already treated as an array.  `DT_PHANDLE_BY_IDX()` works on a singleton phandle and so does `DT_PROP_LEN()`.  It's just the `FOREACH` macros that don't work.

`int` and `string` might be a limitation.

galak, 2023-02-23:
Posted your patch and we'll take a look.

closed 2023-05-16 by carlescufi