zephyrproject-rtos/zephyr #55148: size_report fails in encodings\cp1252.py with cmd.exe (works with PowerShell)

IvanVnucec, 2023-02-24:
I'm using nRF Connect SDK v2.0.2.
Machine is Windows 10.
Before calling `west` commands, I've ran `C:\ncs\v2.0.2\zephyr\zephyr-env.cmd` with console to set needed environment variables.
Then I change directory to the root folder of my app and run `west build -b <some_board>`.
After succesfull build, I've ran `west build -t rom_report` which, after 15 seconds or so, results in FATAL ERROR shown below:
```sh
-- west build: running target rom_report
[0/1] cmd.exe /C "cd /D C:\Users\Korisnik\Desktop\ivan\projekti\art_nr1amp\build && C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/project/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/project/build --workspace=C:/ncs/v2.0.2 -d 99 rom"
Path                                                                                             Size    %
==============================================================================================================
Root                                                                                           212424 100.00%
Traceback (most recent call last):
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 803, in <module>
    main()
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 792, in main
    print_any_tree(root, symsize, args.depth)
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 697, in print_any_tree
    print(f"{row.pre}{cc}{row.node.name} {s} {cr}{Fore.BLUE}{percent:6.2f}%{Fore.RESET}")
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 41, in write
    self.__convertor.write(text)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 162, in write
    self.write_and_convert(text)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 187, in write_and_convert
    self.write_plain_text(text, cursor, start)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 195, in write_plain_text
    self.wrapped.write(text[start:end])
  File "encodings\cp1252.py", line 19, in encode
UnicodeEncodeError: 'charmap' codec can't encode characters in position 0-2: character maps to <undefined>
FAILED: zephyr/cmake/reports/CMakeFiles/rom_report
cmd.exe /C "cd /D C:\project\build && C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/project/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/project/build --workspace=C:/ncs/v2.0.2 -d 99 rom"
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\ncs\toolchains\v2.0.2\opt\bin\cmake.EXE' --build 'C:\project\build' --target rom_report
```

IvanVnucec, 2023-02-24:
You're welcome. Thank you!

IvanVnucec, 2023-02-24:
> Thanks @IvanVnucec for the detailed reproduction steps, very useful.
> 
> As a workaround I recommend you switch to Powershell if possible, cmd.exe is long obsolete anyway.
> 
> Since you reproduced the same with `size_report` directly this failure is unrelated to west. I'm transferring it.

You're welcome. Thank you!

mbolivar-nordic, 2023-02-24:
@marc-hb thanks for the support!

> As a workaround I recommend you switch to Powershell if possible, cmd.exe is long obsolete anyway.

Just a reminder that cmd.exe is our only officially supported windows command line environment, so unless you're proposing to update the GSG to work in powershell, using cmd.exe is still the "blessed" route

marc-hb, 2023-02-24:
Thanks @mbolivar-nordic I'm glad you took that cmd.exe bait, it was for you :-)

More seriously: terminal issues like this one aside I hope there is nothing in Zephyr that is either cmd.exe or Powershell specific. I understand the difference between "officially supported" and "should be OK" but I hope Powershell-specific bug (again: if any at all) would be at least considered.

So if Powershell works for you and cmd.exe does not, you should know what to do...

mbolivar-nordic, 2023-02-24:
> I hope there is nothing in Zephyr that is either cmd.exe or Powershell specific

One example of a possible difference is the command needed to activate a virtual environment in powershell vs cmd.exe

carlescufi, 2023-06-02:
@IvanVnucec I cannot reproduce this locally at all. Can you please give me exact reproducible steps using upstream Zephyr (i.e. not NCS)? I tried with both `cmd.exe` and the new Windows Terminal and both work perfectly with `rom_report`.

PR #58789 (2023-06-02): scripts: Fix a couple of string encoding issues
See commit messages.

Fixes #55148
commit d2fff07b (2023-06-02): scripts: size_report: Use default encoding to process filenames

The linker generates an .elf file wth encoded strings containing
filenames. If those filenames contain non-ASCII characters we need to
know which encoding was used by the linker to store those filenames. Use
the system's preferred encoding for that purpose.

Fixes #55148.
closed 2023-06-02 by nashif