zephyrproject-rtos/zephyr #55166: new samples/basic/hash_map fails on some stm32 boards

FRASTM, 2023-02-24:
**Describe the bug**
A clear and concise description of what the bug is.
(Note that [Github Discussions](https://github.com/zephyrproject-rtos/zephyr/discussions) are available to first verify that the issue
is a genuine Zephyr bug and not a consequence of Zephyr services misuse.)

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

cfriedt, 2023-02-24:
[image]

@FRASTM - which of these board options would be best to try and reproduce?

cfriedt, 2023-02-25:
@FRASTM - cannot reproduce
```
west build -p auto -b nucleo_g071rb samples/basic/hash_map/ -- -DCONFIG_NEWLIB_LIBC=y CONFIG_SYS_HASH_MAP_CHOICE_CXX=y CONFIG_SYS_HASH_FUNC32_CHOICE_DJB2=y
west flash
```
results in
```
CONFIG_TEST_LIB_HASH_MAP_MAX_ENTRIES: 40                                                                 
[00:00:04.001,000] <inf> hashmap_sample: n_insert: 720 n_remove: 6120 n_replace: 6716 n_miss: 0 n_error: 0 max_size: 720
[00:00:04.006,000] <inf> hashmap_sample: success                                                         
```

FRASTM, 2023-02-27:
when running the sample on stm32l562e_dk
$ west build -p always -b stm32l562e_dk samples/basic/hash_map
--> the test  **libraries.hash_map.newlib.cxx_unordered_map.djb2**
add the `extra_configs` to the prj.conf has 
CONFIG_NEWLIB_LIBC=y
CONFIG_SYS_HASH_MAP_CHOICE_CXX=y
CONFIG_SYS_HASH_FUNC32_CHOICE_DJB2=y

cfriedt, 2023-03-01:
@FRASTM - would you be OK to make a PR that adds this `extra_config` to the `testcase.yaml` file for the `cxx` case?

If not, I can make one, but I haven't been able to reproduce it so far.

FRASTM, 2023-03-02:
I've prepared a PR https://github.com/zephyrproject-rtos/zephyr/pull/55349 to adjust the value of  CONFIG_MAIN_STACK_SIZE

closed 2023-03-02 by nashif