zephyrproject-rtos/zephyr #55172: atmel sam0 I2C cannot read registers at a select address and must read entire chip

TomzBench, 2023-02-24:
Describe the bug

When trying to read a single register via sam0 i2c bus, an extra address byte is sent on the wire.

I2C devices support reading a single byte by first writing the address with the RW bit set high, followed by writing a second byte which sets up an internal pointer register, followed by a subsequent read for clocking in the data from the remote device. The i2c_sam0 driver does not support this feature. the implementation of i2c_transfer intentionally sends the address byte with the start of every i2c_msg.

As partial workaround, I am able to use the driver to read the entire chip in one shot in order to target the register I want. However, not being able to select a specific register with a control byte write means I cannot use this driver to manage interrupts in a practical way (which are Clear on Read).

To Reproduce

Build any board that uses the i2c_sam0.c driver and enable the I2C_SHELL. Then try to read a single byte on the bus with the shell command i2c read_byte sercom@xxxxxxxxx ADDR REG and observe that 2 address bytes are sent on the wire.

** Other notes **
moved to issue from: https://github.com/zephyrproject-rtos/zephyr/discussions/55170

TomzBench, 2023-02-24:
As another work around, the gpio-i2c (bitbang) driver seems to work for me when i read a single byte

attie-argentum, 2023-03-01:
Thanks for the report... I suspect this might be an issue with (not) handling `I2C_MSG_RESTART`, but I'll investigate and get back to you.

TomzBench, 2023-03-02:
> @TomzBench - are you able to share a scope/logic analyser trace showing the issue? please could you also let me know which devices you're working with? (uC and I2C device)

@attie-argentum I been using an oscilloscope. I can send an image in a few hours or so.

I'm using custom hardware based off the atsame54_xpro board.

The issue is not so much the extra address byte as described in post. After comparing with the bitbang driver, the sam0 driver does not set the RW bit when doing the write_read pattern.

Will send image when have set up again

attie-argentum, 2023-03-02:
Thanks @TomzBench ... knowing the I2C device would be useful too, please.

I've tested using a SAML21 XPro and a VL53L0x ... while I acknowledge that this doesn't perform a "_repeated start_", I'm not aware of any devices specifically that **require** that.

Here is a trace captured from my setup, and all appears to be in order(?)

```
uart:~$ i2c read_byte sercom@42000800 0x29 0xc0
Output: 0xee
```

[image]

TomzBench, 2023-03-02:
@attie-argentum 

I am using the sercom3@41014000 device with the atmel,i2c-sam0 driver, on the atsame54p20a chip

Below I am reading a MAX5980 chip, register 1B.  The MAX5980 chip is address 0x20. Register 1B is supposed to be the "chip id" register where the ID bits should be 0xD0.  When ever I do a read, the final packet is NACK and the data seems incorrect.  All registers seem to read with the value 0x08.  

Also note in between the Read [0x20] and the read byte, there is no repeated start so perhaps this is bugging up the MAX5980? I tried to sequentially capture all the bytes in multiple screen shots because of the nature of my oscilloscope.

It was hard to count the bits so some of my information reported above was not correct. However perhaps your insight can help diagnose the problem after reviewing the images below. 

Write [0x20] looks ok
[image]

0x1B + ACK looks ok
[image]

Looks like maybe repeated start?
[image]

Read [0x20] seems ok with write bit
[image]

0x08 + NACK (data not correct)
[image]

attie-argentum, 2023-03-02:
Thanks for the info @TomzBench.

Unfortunately I can't follow the screenshots, as there's too much missing data for me to decode the I2C... reducing the time base / zooming out might help capture more at once - e.g: 20µs or maybe 40µs /div.

I've had a look at the MAX5980 datasheet - it uses an atypical approach to internal register addressing... Note that **_both_** the read and write operations have the control byte before the data. The Zephyt `i2c read_byte` command expects only data to be returned for a read operation.

[image]

Please could you try using `i2c read` instead of `i2c read_byte`? I would expect the response will be `0x1b` (i.e: the value of the control byte / pointer register) followed by the data in that register... e.g:

```
i2c read sercom3@41014000 0x20 0x1b 2
```

Do you have multiple I2C initiators / masters on your bus? If not, then not using repeated starts shouldn't be an issue here...

TomzBench, 2023-03-02:
@attie-argentum 

I only have a single master on the bus (the same54p20a).

I took some more images below zoomed out a bit. It is now 2 pictures showing each pair of writes, and I used the cursors to highlight the pairs of bytes being sent so you can more easily track between the two images.  

When i do i2c read 2 bytes i see the 0x08 again. 

> i2c read sercom@41014000 0x20 0x1B 2
> 00000000: 08 e4

[image]
[image]

TomzBench, 2023-03-02:
@attie-argentum 

Note some more information RE the 0x08 0xE4 value.

when using the working driver, reading 0xB1 gives 0xD0 0x0F. However what might be useful infomartion is that 0x08 0xE4 is correctly the first 2 bytes (located at 0x00).  Meaning that the setup byte seems ignored and all reads are starting from 0x00. and it's internal counter will increment from 0, (not from the control setup byte)

TomzBench, 2023-03-02:
@attie-argentum I merged your sam0-i2c-restart branch into my zephyr and now register 0x1B reads 0xD0 as expected. so it looks to me like the datasheet did not document this register reset on stop behavior.

attie-argentum, 2023-03-02:
Great, thanks for letting me know!

> so it looks to me like the datasheet did not document this register reset on stop behavior.

Indeed - I've actually been looking for a device that doesn't treat stop-start the same as restart, so thanks! 😄

TomzBench, 2023-03-02:
> Indeed - I've actually been looking for a device that doesn't treat stop-start the same as restart, so thanks!

And thanks for responding so quickly with a patch!

Much appreciated, potentially shipping with the bitbang driver felt funny  :smile:

PR #55385 (2023-03-09): drivers: i2c: sam0: add support for I2C_MSG_RESTART
This PR reinstates support for I2C repeated starts, that was accidentally removed in the re-shuffle in de18f97...

Fixes #55172
closed 2023-03-09 by carlescufi