zephyrproject-rtos/zephyr #55186: posix: fnmatch: fix known bugs

cfriedt, 2023-02-25:
**Describe the bug**
It appears as though the original `fnmatch()` implementation that was checked-in with 486b731b40717a16f619b07e5d58aaf7d5143e78 has bugs. This is not surprising since it was checked-in without any test coverage.

The actual failing cases are commented-out in `tests/lib/posix/common/src/fnmatch.c` (part of #55184)

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? N/A
 - What have you tried to diagnose or workaround this issue? Comment-out known failure modes in the test
 - Is this a regression? No, the bugs have always been present

**To Reproduce**
Steps to reproduce the behavior:
1. Uncomment failure modes in `tests/lib/posix/common/src/fnmatch.c`
2.`west build -p auto -b qemu_cortex_a53 -t run tests/posix/common/`
3. See error

**Expected behavior**
All test cases should pass, including known failure modes.

**Impact**
The shell has been using a buggy version of `fnmatch()` since approximately 2018.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.3.0-475-gec19849fec26 ***
Running TESTSUITE posix_apis
===================================================================
START - fnmatch

    Assertion failed at WEST_TOPDIR/zephyr/tests/posix/common/src/fnmatch.c:35: posix_apis_fnmatch: fnmatch("[[?*\\]", "\\", 0) is non-zero

 FAIL - fnmatch in 0.001 seconds
===================================================================
```

**Environment (please complete the following information):**
 - OS: Any
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK v0.15.2
 - Commit SHA or Version used: b83cf1f1ee244e48a22844be181fd6ae2366e2c1 but goes as far back as 486b731b40717a16f619b07e5d58aaf7d5143e78

**Additional context**
Noticed in the context of #55185

cfriedt, 2023-03-28:
Changed to priority low, as these are not new bugs by any means, just newly discovered, and so any code that exists already in main is should continue to be stable even without an immediate fix.

AbhinavMir, 2024-01-17:
Hi! Are there any hints for what could be wrong? I see quite a few cases covered in `rangematch`, but it's pretty confusing looking at all the code. I will take a stab at this again

cfriedt, 2024-01-18:
> Hi! Are there any hints for what could be wrong? I see quite a few cases covered in `rangematch`, but it's pretty confusing looking at all the code. I will take a stab at this again later today.

There are just a few cases commented out in the test suite.

For this ticket, we would need to uncomment those cases and ensure that tests pass.

https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/posix/common/src/fnmatch.c

ClaCodes, 2024-09-14:
@cfriedt 
I have started to take a look a this.
1. The following two examples do not even pass on my linux `fnmatch` implementation. They are either wrong, `unspecified` or Linux has a bug. I would propose to delete them.
```C
	/* zassert_ok(fnmatch("[[?*\\]", "\\", 0)); */
	/* zassert_ok(fnmatch("[]?*\\]", "]", 0)); */
```
2. This in fact passes on linux. If it is not `unspecified` it is a bug with the current `fnmacht` as this ticket suggests.
```C
	/* zassert_ok(fnmatch("a[/]b", "a/b", 0)); */
```
3. The following do in fact pass on linux. But I found, `It is unspecified whether an explicit \<period\> in a bracket expression matching list, such as "[.abc]", can match a leading \<period\> in a filename. https://pubs.opengroup.org/onlinepubs/9699919799/utilities/V3_chap02.html#tag_18_13_03 . I interpret this as applying here. Such that these test for something unspecified i.e. tests can be removed.
```C
	/* zassert_equal(fnmatch("[.]a/b", ".a/b", FNM_PATHNAME | FNM_PERIOD), FNM_NOMATCH); */
	/* zassert_equal(fnmatch("a/[.]b", "a/.b", FNM_PATHNAME | FNM_PERIOD), FNM_NOMATCH); */
```
4. Character classes like `:alpha:` are not yet implemented. One would assume, that this pattern matching code is usefull elsewhere. For example in `shell`? If so, it probably make sense to implement one in terms of the other. Probably implement `shell` in terms of `posix`. Is there pattern matching code already elsewhere in the tree?
```C
	/* zassert_ok(fnmatch("*[[:alpha:]]/""*[[:alnum:]]", "a/b", FNM_PATHNAME)); */
```

cfriedt, 2024-09-14:
@ClaCodes 

> 1. The following two examples do not even pass on my linux `fnmatch` implementation. They are either wrong, `unspecified` or Linux has a bug. I would propose to delete them.
> ```C
> 	/* zassert_ok(fnmatch("[[?*\\]", "\\", 0)); */
> 	/* zassert_ok(fnmatch("[]?*\\]", "]", 0)); */
> ```

fnmatch() would typically be implemented in a libc on Linux systems. What C library does your Linux system use? Have you tried fnmatch() from musl?

We'd like to avoid ambiguity (especially in the case of an RTOS), and also gracefully handle most things that are thrown at us, even if the behaviour is unspecified.

I don't believe it's correct to remove / ignore test cases because another implementation has the same bug or also simply categorizes some input as unspecified.

> 2. This in fact passes on linux. If it is not `unspecified` it is a bug with the current `fnmacht` as this ticket suggests.
> ```C
> 	/* zassert_ok(fnmatch("a[/]b", "a/b", 0)); */
> ```

IIRC, this test wouldn't have passed with the current Zephyr implementation (if it is commented-out this way in the testsuite).

So likely it does need to be fixed in Zephyr.

> 3. The following do in fact pass on linux. But I found, `It is unspecified whether an explicit \<period\> in a bracket expression matching list, such as "[.abc]", can match a leading \<period\> in a [...]

Again, we don't want to remove the testcases that stress the implementation. We want to leave breadcrumbs so that we remember to improve the implementation at some point.

If it is unspecified behaviour, we should make it obvious (via tests and docs) how we handle it.

So again, I would disagree that we should remove tests. 

> 4. Character classes like `:alpha:` are not yet implemented. One would assume, that this pattern matching code is usefull elsewhere. For example in `shell`? If so, it probably make sense to [...]

Yes, it's very easy to match both of these specifiers using the ISO C calls [isalpha()](https://en.cppreference.com/w/c/string/byte/isalpha) and [isalnum()](https://en.cppreference.com/w/c/string/byte/isalnum).

ClaCodes, 2024-09-14:
Thank you for the quick response.

> What C library does your Linux system use?

ldd (Ubuntu GLIBC 2.31-0ubuntu9.16) 2.31

> Have you tried fnmatch() from musl?

No not yet

> I don't believe it's correct to remove / ignore test cases because another implementation has the same bug or also simply categorizes some input as unspecified.

Fair enough, The question is, which is right. Is the test correct? I am not sure. Are they testing, that the bracket expression really escapes the special characters and matches them literally as described here (https://pubs.opengroup.org/onlinepubs/9699919799/basedefs/V1_chap09.html#tag_09_03_05). BTW I do not understand why it sais `'\\'` instead of `'\'` in that quote. If so the test should stay and the behavior implemented.
```
The special characters '.', '*', '[', and '\\' ( <period>, <asterisk>, <left-square-bracket>,
and <backslash>, respectively) shall lose their special meaning within a bracket expression 
```

> If it is unspecified behaviour, we should make it obvious (via tests and docs) how we handle it.

Do you agree, that it is unspecified? If so, would you be ok to specify the current behavior and then to assert the contrary of the commented out test, Namly, that we do match those? Personally I would not test for something unspecified as users are not supposed to rely on it and the implementation should be allowed to change it. But both is fine.

> Yes, it's very easy to match both of these specifiers using the ISO C calls [isalpha()](https://en.cppreference.com/w/c/string/byte/isalpha) and [isalnum()](https://en.cppreference.com/w/c/string/byte/isalnum).

Fair enought. I was just wondering, if there is something similar elsewhere in the tree. Character classes are not yet implemented at all and at least according to the spec many more need to be supported (https://pubs.opengroup.org/onlinepubs/9699919799/basedefs/V1_chap09.html#tag_09_03_05)
```
The following character class expressions shall be supported in all locales:

[:alnum:]   [:cntrl:]   [:lower:]   [:space:]
[:alpha:]   [:digit:]   [:print:]   [:upper:]
[:blank:]   [:graph:]   [:punct:]   [:xdigit:]

```

cfriedt, 2024-09-14:
> > I don't believe it's correct to remove / ignore test cases because another implementation has the same bug or also simply categorizes some input as unspecified.
> 
> Fair enough, The question is, which is right. Is the test correct?

My guess is that the test cases (originally from [here](https://git.musl-libc.org/cgit/libc-testsuite/log/fnmatch.c)) are correct. The test-cases themselves were created about 13 years ago and the last update was 10 years ago (some [history](https://github.com/PocketCluster/libc-test/commits)).

> I am not sure. Are they testing, that the bracket expression really escapes the special characters and matches them literally as described here (https://pubs.opengroup.org/onlinepubs/9699919799/basedefs/V1_chap09.html#tag_09_03_05). 

The expectation is that the call would result in `FNM_NOMATCH`, so it seems that the bracket expression `[.]` should _not_ match the `.` character. at the start of a file name.

> BTW I do not understand why it sais `'\\'` instead of `'\'` in that quote. If so the test should stay and the behavior implemented.

In C, if we want to express a backslash character in a string, then it must be escaped. So realistically, the matching pattern is `[[?*\]`, and the input is `\`. This test simply verifies that the backslash is interpreted correctly in the pattern.

So it would seem that this is also a bug in the Zephyr implementation.

> Do you agree, that it is unspecified?

I actually do not believe that it is unspecified behaviour. I think it's most likely a bug.

> ```
> The following character class expressions shall be supported in all locales:
> 
> [:alnum:]   [:cntrl:]   [:lower:]   [:space:]
> [:alpha:]   [:digit:]   [:print:]   [:upper:]
> [:blank:]   [:graph:]   [:punct:]   [:xdigit:]
> ```

It might be helpful to add test cases for each of those scenarios - they are all easy to verify using functions from [ctype.h](https://www.tutorialspoint.com/c_standard_library/ctype_h.htm)

clamattia, 2025-03-18:
The `fnmatch`-function now sees use also in the `http_server_core`

cfriedt, 2025-03-18:
@clamattia - I would be happy to review if you have cycles to implement a fix for the few missing cases.

clamattia, 2025-05-19:
Please keep this open.

nimisht, 2025-05-24:
seems like the tests have now moved to `tests/posix/c_lib_ext/` 

I've got `/lib/posix/options/fnmatch.c` passing all tests except this one: `	/* zassert_ok(fnmatch("*[[:alpha:]]/""*[[:alnum:]]", "a/b", FNM_PATHNAME)); */`

Good enough for a PR? 

btw, seems like some of the tests in `tests/posix/c_lib_ext/getentropy.c` are failing, not sure if its a problem on my end.

cfriedt, 2025-05-24:
> I've got `/lib/posix/options/fnmatch.c` passing all tests except this one: 
> 
> `/* zassert_ok(fnmatch("*[[:alpha:]]/""*[[:alnum:]]", "a/b", FNM_PATHNAME)); */`
> 
> Good enough for a PR? 

@nimisht - Of course :-)

> btw, seems like some of the tests in `tests/posix/c_lib_ext/getentropy.c` are failing, not sure if its a problem on my end.

Those tests rely on an entropy device being present, which isn't always the case. Feel free to bring it up on Discord if you're unsure.

ritesh006, 2025-08-05:
is this issue still open

cfriedt, 2025-09-16:
@DrBinary9 - this testcase needs to pass

https://github.com/zephyrproject-rtos/zephyr/blob/377926ca0d3a65e146db554fcc7321db3a0c6c74/tests/posix/c_lib_ext/src/fnmatch.c#L76

with the implementation here

https://github.com/zephyrproject-rtos/zephyr/blob/377926ca0d3a65e146db554fcc7321db3a0c6c74/lib/posix/options/fnmatch.c#L126

Typically 1 code commit followed by 1 test commit.