zephyrproject-rtos/zephyr #55193: MQTT-SN IPv6: wrong family in zsock_socket

sebarnd, 2023-02-26:
**Describe the bug**

The File
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/mqtt_sn/mqtt_sn_transport_udp.c#L51

contains the line:
```
udp->sock = zsock_socket(PF_INET, SOCK_DGRAM, 0);
```
in order to support IPv6 `PF_INET` needs to replaced by `PF_INET6`. Assuming `AF_INET == PF_INET && AF_INET6 == PF_INET6` a solution could be:

```
udp->sock = zsock_socket(udp->gwaddr.sa_family, SOCK_DGRAM, 0);
```.

Would that be a good solution according to your coding style? Than I could try to prepare a pull request for this...

rlubos, 2023-02-27:
> ```
> udp->sock = zsock_socket(udp->gwaddr.sa_family, SOCK_DGRAM, 0);
> ```
> 
> 
>     
>   
> 
> Would that be a good solution according to your coding style? Than I could try to prepare a pull request for this...

Yes, this was my first thought as well, the AF for socket should be set according to the destination address provided. Please send a PR.

napent, 2023-02-28:
Hi, from what I understand similar bug is introduced here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/mqtt_sn/mqtt_sn_transport_udp.c#L64

rlubos, 2023-03-01:
> Hi, from what I understand similar bug is introduced here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/mqtt_sn/mqtt_sn_transport_udp.c#L64

Well, technically `sockaddr_in`/`sockaddr_in6` structures have the same layout when it comes to family/port, so I'd rather call this one an oversimplification rather than a bug, it still should work fine. However, I'm thinking it'd be cleaner if port was also provided by the `get_ip_str()` helper.

> Also there is a bug in the example project:
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/mqtt_sn_publisher/src/udp.c#L119

This indeed looks wrong.

sebarnd, 2023-03-02:
> > Also there is a bug in the example project:
> > https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/mqtt_sn_publisher/src/udp.c#L119
> 
> This indeed looks wrong.

How would you fix this one? Adding an additional `CONFIG` flag or using `zsock_getaddrinfo()` with `AI_NUMERICHOST` hint?

rlubos, 2023-03-03:
> > > Also there is a bug in the example project:
> > > https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/mqtt_sn_publisher/src/udp.c#L119
> > 
> > 
> > This indeed looks wrong.
> 
> How would you fix this one? Adding an additional `CONFIG` flag or using `zsock_getaddrinfo()` with `AI_NUMERICHOST` hint?

It's just a matter of wrong error code check. As @napent already mentioned: `inet_pton() returns 1 on success` - hence, the check should compare to 1, not 0.

PR #55573 (2023-03-09): net/lib: mqtt_sn: Fix MQTT-SN IPv6 Support and assertation in Example
Fixes `zsock_socket()` getting always PF_INET packet Familiy also when using IPv6. Fixes assertation as  `inet_pton()` returns 1 on success.

Fixes #55193.
closed 2023-03-09 by carlescufi