zephyrproject-rtos/zephyr #55204: Zephyr does not define minimal C++ language standard requirement and does not track it

aborisovich, 2023-02-27:
**Describe the bug**
Zephyr defines C99 as minimal supported standard for C language in [C language support - language standards](https://docs.zephyrproject.org/latest/develop/languages/c/index.html#language-standards).
Similar documentation article: [C++ language support](https://docs.zephyrproject.org/latest/develop/languages/cpp/index.html) does not have "language standard" section. Roadmap #31281 also does not say a word about minimal requirements. Thus, minimal C++ version supported is unknown. This is a serious problem since:
*  C++ language standards have much more changes between releases than C language.
* Custom toolchains for embedded systems often concentrate on C language support (due to its popularity) leaving C++ language  with old standard (like dreadful C++98).

**Concrete problem encountered:**
[SOF project](https://github.com/thesofproject/sof) uses Cadence Xtensa toolchain "xt-xcc" that is based on GNU 4.2.0 compiler for Intel Meteorlake audio firmware.
Meteorlake board has some C++ code that uses Zephyr `#include <zephyr/sys/util.h>` that provides C/C++ interropt helpers at the time of writing. "xt-xcc" is obsolete for some time and latest support language is C++98 standard.
Everything was working fine in our CI until PR #53405 , where author added `constexpr` keyword usage in commit deab09d7a6b0c51a52ffd8bd03f17e04cf39c25c .
Author @cfriedt in that PR claims (selective quote):

> > There should be a announcement for this and the code that suddenly requires new C++

> 
> As mentioned above, this is all in line with Zephyr's suggested ISO C and C++ standards (C11, C++11).

**Prof of wrong:**
Fuction template definition `__z_log2_impl` from `include/zephyr/sys/util.h` in some conditions returns value A and in other conditions value B. According to C++ standard ISO/IEC 14882:2011 §7.1.5.3:

> The definition of a constexpr function shall satisfy the following constraints:
> - it shall not be virtual (10.3);
> - its return type shall be a literal type;
> - each of its parameter types shall be a literal type;
> - its function-body shall be = delete, = default, or a compound-statement that contains only
>   - null statements,
>   - static_assert-declarations
>   - typedef declarations and alias-declarations that do not define classes or enumerations,
>   - using-declarations,
>   - using-directives,
>   - and exactly one return statement;
> - every constructor call and implicit conversion used in initializing the return value (6.6.3, 8.5) shall be
> one of those allowed in a constant expression (5.19).

Also, [cppreference.com - constexpr](https://en.cppreference.com/w/cpp/language/constexpr) confirms this:
Quote marked "(until C++14)":

> the function body must be either deleted or defaulted or contain only the following:
> - [null statements](https://en.cppreference.com/w/cpp/language/statements#Expression_statements) (plain semicolons)
> - [static_assert](https://en.cppreference.com/w/cpp/language/static_assert) declarations
> - [typedef](https://en.cppreference.com/w/cpp/language/typedef) declarations and [alias](https://en.cppreference.com/w/cpp/language/type_alias) declarations that do not define classes or enumerations
> - [using declarations](https://en.cppreference.com/w/cpp/language/namespace#Using-declarations)
> - [using directives](https://en.cppreference.com/w/cpp/language/namespace#Using-directives)
> - if the function is not a constructor, exactly one [return](https://en.cppreference.com/w/cpp/language/return) statement
> 

**To Reproduce**
Steps to reproduce the behavior:
1. Open my minimal reproduction project with Compiler Explorer:
https://godbolt.org/z/78sWY65v4
2. Play around with compiler version and C++ standard.
Note: compiler x86-64 gcc 12.2 compiles minimal example using C++11 without a warning!
Change compiler to: x86-64 11.3 to receive an error (related to multiple return statements in constexpr function):

> header.h:17:1: error: body of 'constexpr' function 'constexpr int __z_log2_impl(T) [with T = int]' not a return-statement
>    17 | }
>       | ^
3. Change CMake line:
`set_property(TARGET output.s PROPERTY CXX_STANDARD 11)` to
`set_property(TARGET output.s PROPERTY CXX_STANDARD 14)`
and watch how everything compiles successfully.

**Expected behavior**
1. Zephyr defines minimal C++ language standard in the official documentation.
Right now it is C++14 if we do not roll back PR #53405 (probably - remaining C++ code needs verification too!)
2. Developers know, and check the C++ standard of the code they wright! Please note, that modern compilers GCC, Clang, MSVC tend to provide "backward compatibility" to old C++ standards what results in non-conforming code being compile without an issue!
If the compiler supports C++17 or C++20 it is possible that it will compile the code marked -std=c++11 that should not compile with that standard due to compiler extensions they provide! We need to watch the conformance carefully, maybe some static analysis tools?

**Impact**
SOF project has to urgently upgrade toolchain to new Cadence Xtensa "xt-clang" based on clang, that allows to use up to C++17 standard. If we do not do this, we are unable upgrade Zephyr revisions since our code will not build!
Luckily, we had been preparing for toolchain change in https://github.com/thesofproject/sof/pull/7027 for some time and have only 2 .cpp source files at the time of writing.

**Logs and console output**
Compiling with xt-xcc:  [xt-xcc.log](https://github.com/zephyrproject-rtos/zephyr/files/10835174/xt-xcc.log)
Compiling with xt-clang -std=c++11:  [xt-clang C++11.log](https://github.com/zephyrproject-rtos/zephyr/files/10835177/xt-clang.C%2B%2B11.log)

**Environment (please complete the following information):**
 - OS: Windows, Linux

**Additional context**
N/A

cfriedt, 2023-02-27:
@aborisovich - documentation only does so much. Why not contribute a PR? Perhaps a test so you are not inconvenienced the same way in the future. It will probably amount to only a few lines of code.

aborisovich, 2023-02-27:
> @aborisovich - documentation only goes so far.
> 
> Why not contribute a PR? Perhaps a test so you are not inconvenienced the same way in the future.
> 
> It will probably amount to only a few lines of code.

Subject is certainly not easy and we need some tool that would run as part of Zephyr CI. And it will certainly not be only a few lines of code. I'm looking for potential solutions. Maybe something like [cppcheck](https://cppcheck.sourceforge.io/) added to Compliance Checks workflow would be good for checking standard conformance...

cfriedt, 2023-02-28:
@aborisovich - can you extract just the relevant portion of your error from the log and put it into a markdown code block, so that the error is a bit more visible / searchable?

aborisovich, 2023-02-28:
@cfriedt I never, ever meant to discredit anyone. I assume that as obvious thing, that we developers work with mutual respect.
Bugs, unspecified behaviors are constant part of our work. Moreover, I strongly approve what you do - C++ is a very needed language in Zephyr and I hope it will be replacing or complimenting C in future. When I wrote "you are doing something wrong" I meat it as a quote, not as an expression targeting you directly. By this quote I meant "Zephyr project systems providing API compatibility". I wanted to indicate importance of the problem I described above and potentially severe impact, so we can harden the processes in Zephyr.

cfriedt, 2023-02-28:
@aborisovich - no harsh feelings. I'm glad that we can work together to make things better.

Please let me know if you need any help.

jwakely, 2023-03-01:
Using `-std=c++11 -pedantic` should get you similar earnings with GCC. I'm not sure which backwards compatibility extensions you're worried about, generally GCC and clang will warn if you use language features that are not supported by the standard dialect selected with -std

teburd, 2023-03-01:
Zephyr docs don't explicitly say C11 or even C99 is required. C11 is still on the table as a pending requirement in RFC #30105

The docs simply state that C99 features are widely used but there is no requirement on it.

andyross, 2023-03-01:
(and having said that it's clear I should just put that text in the docs and mark it as fixing this bug)

stephanosio, 2023-03-01:
> Zephyr defines C11 as minimal supported standard for C language in [C language support - language standards](https://docs.zephyrproject.org/latest/develop/languages/c/index.html#language-standards).

That is not necessarily true. The documentation says that most parts of Zephyr only make use of the features available in C99 with some additional components requiring C11 -- i.e. the "minimum" compiler C standard support requirement varies depending on which component(s) you require.

The documentation does not designate a specific "minimum supported" (or "maximum allowed" for that matter) C standard version because some components may benefit from the features available in a more recent version of the standard; nevertheless, the recommendation is to stick with C99 in general and C11 at most.

> **Expected behavior**
> 
> 1. Zephyr defines minimal C++ language standard in the official documentation.

For the same reason described above, I would refrain from defining a hard limit; especially for C++ because newer C++ standards have many appealing features.

The documentation should only provide the general recommendation (and the de facto requirement for the core components of Zephyr) for the C++ standard conformance, which would be to stick with C++11 in general. It would be preferable to allow a more recent version of the C++ standard for specific components and modules.

aborisovich, 2023-03-03:
I reopen this issue because it looks to me that you may missed the key point I am trying to show here.
This statement:

> For the same reason described above, I would refrain from defining a hard limit; especially for C++ because newer C++ standards have many appealing features.

Is not a resolution to problem with using C++. I agree that new language feature are appealing, but please note what I wrote in the issue description:

> Custom toolchains for embedded systems often concentrate on C language support (due to its popularity) leaving C++ language with old standard (like dreadful C++98).

@andyross, filling up documentation with "We support C++ standard whatever found in the give file you compile" is not a solution either. I would rather call it "discrediting a problem" than solving it.
This problem is not related to C, because Zephyr does great job of hiding C language features under #ifdef macros.
Example: What would happen to all zephyr applications if you suddenly add to very commonly used header (like some util header) [alignof](http://ld2014.scusa.lsu.edu/cppreference/en/c/align/alignof.html) keyword (supported since C11).
All zephyr applications using C99 would stop to work and would be forced to use C11 compiler right?
You basically do the same for C++ right now, and cover the issue with a blanked - do not do this.

andyross, 2023-03-03:
Hm... as of right now we guarantee that all the OS headers build correctly with C++98 compilers (specifically gcc and clang with -std=c++98), and we're testing this along with all the other .  It's true it was a hole, but I'm pretty sure it's been plugged.  Obviously the discussion as to when to deprecate C++98 will be contentious, but for now at least that's the requirement.

Which is to say: if you push a PR right now using `alignof` or rvalue references or whatever inside an `#ifdef __cplusplus`, it will fail in CI.  I think that's what you want, no?

(Note that there are other issues with C++ support at the toolchain level, like "how to link in the standard library" that are quite a bit farther behind.  So while the OS has support lots of platforms won't quite build C++98 correctly even if their compilers technically support it; xcc and xt-clang are both in this category for now.)

aborisovich, 2023-03-03:
Maybe I need to make it simple and more concise:
Commit https://github.com/zephyrproject-rtos/zephyr/commit/deab09d7a6b0c51a52ffd8bd03f17e04cf39c25c should not every compile without setting Kconfig option `CONFIG_STD_CPP14=y` but it does (using `CONFIG_STD_CPP11=y`) because the compilers you use does not check standard conformity correctly.

andyross, 2023-03-03:
Yeah, those problems are fixed as of #55290 

Please open a new bug if you find a way to add non-C++98 code to an OS header without doing proper version detection.

closed 2023-03-03 by andyross

aborisovich, 2023-03-03:
Thanks @andyross . Did not notice a whole PR just a single commit modifying docs. Sorry for the hussle.

> Using `-std=c++11 -pedantic` should get you similar warnings with GCC. I'm not sure which backwards compatibility extensions you're worried about, generally GCC and clang will warn (or error) if you use language features that are not supported by the standard dialect selected with -std. You just need to use the option to ask for warnings about extensions, which is `-pedantic`.

Indeed adding a `-pedantic` flag to newest gcc also resolves the issue.

commit 10b9aab8 (2023-03-02): doc: Clarify C++ standards support

We weren't explicit about what we mean by "C++ support" as it relates
to standards versions, and it's been causing some friction in the tree
vs. some of our cruftier toolchains.  Add a paragraph to clarify
things.

Fixes #55204