zephyrproject-rtos/zephyr #55205: spi: stm32 support for asynchronous DMA transfer

semihalf-niedzwiecki-dawid, 2023-02-27:
**Is your feature request related to a problem? Please describe.**
The SPI STM32 drivers [supports](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/spi/spi_ll_stm32.c#L833) using DMA only for synchronous transfers. The DMA support is needed for high-speed SPI communication. The problem occurs when the chip that runs Zephyr is a SPI slave. The start of a SPI communication is unknown and the `spi_transceive` (which supports DMA) call blocks the program flow. Using ASYNC SPI API would be a much better solution in this case.

**Describe the solution you'd like**
The request is to support using DMA also for the asynchronous transfers, as it is done for [UART](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/serial/uart_stm32.c#L1310).

**Describe alternatives you've considered**
Create a separate thread for the `spi_transceive` blocking call, but it is not the way to go. 

**Additional context**
NA

semihalf-niedzwiecki-dawid, 2023-03-07:
Are there any chances to implement that?

Benni77, 2024-04-03:
@erwango any news on this?

erwango, 2024-07-04:
FYI https://github.com/zephyrproject-rtos/zephyr/pull/73855