zephyrproject-rtos/zephyr #55209: samples: basic: hash_map: libraries.hash_map.newlib.cxx_unordered_map.djb2 fails

nordic-piks, 2023-02-27:
**Describe the bug**
The samples/basic/hash_map/libraries.hash_map.newlib.cxx_unordered_map.djb2 fails

Observed for:
- nrf5340dk_nrf5340_cpunet
- nrf9160dk_nrf9160_ns
- nrf9160dk_nrf9160

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf9160dk_nrf9160 connected
2. go to your zephyr dir
3. call ./scripts/twister -T samples/basic/hash_map -p nrf9160dk_nrf9160--device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-02-26T21:04:05.371Z] [0mERROR   -  CONFIG_TEST_LIB_HASH_MAP_MAX_ENTRIES: 40[0m
[2023-02-26T21:04:05.371Z] [0m[00:00:01.766,143] <err> os: ***** USAGE FAULT *****[0m
[2023-02-26T21:04:05.371Z] [0m[00:00:01.766,143] <err> os:   Stack overflow (context area not valid)[0m
[2023-02-26T21:04:05.371Z] [0m[00:00:01.766,174] <err> os: r0/a1:  0x20001cb8  r1/a2:  0x00000000  r2/a3:  0x00004070[0m
[2023-02-26T21:04:05.371Z] [0m[00:00:01.766,174] <err> os: r3/a4:  0x00000000 r12/ip:  0x20001ef8 r14/lr:  0x000020ed[0m
[2023-02-26T21:04:05.371Z] [0m[00:00:01.766,174] <err> os:  xpsr:  0x49000000[0m
[2023-02-26T21:04:05.371Z] [0m[00:00:01.766,204] <err> os: Faulting instruction address (r15/pc): 0x000014f6[0m
[2023-02-26T21:04:05.371Z] [0m[00:00:01.766,235] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0[0m
[2023-02-26T21:04:05.371Z] [0m[00:00:01.766,265] <err> os: Current thread: 0x20000d60 (unknown)[0m
[2023-02-26T21:04:05.371Z] [0m[00:00:01.828,125] <err> os: Halting system[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.2
 - Commit SHA or Version used:  zephyr-v3.3.0-472-gb2512d2f534e

FRASTM, 2023-02-27:
found similar issue in https://github.com/zephyrproject-rtos/zephyr/issues/55166

cfriedt, 2023-02-28:
@nordic-piks - are you able to suggest a `min_ram` setting or minimal stack size that does make the example work on your board (or at least be filtered out due to ram constraints)?

Not able to reproduce currently because I cannot flash my nRF device from macOS (text below is from discord).

----
Does anyone know if it's possible to flash the `nrf52840dongle_nrf52840` using macOS? I tried using `west build -p auto -b nrf52840dongle_nrf52840 -t flash sample/hello_world` but I get the error below. Just trying to repro / fix a stack overflow on nrf HW.
```
-- west flash: using runner nrfjprog
FATAL ERROR: "nrfjprog --ids" did not find a board; is the board connected?
```
But I actually do have `nrfjprog` installed:
```
nrfjprog -v
nrfjprog version: 10.17.2 external
JLinkARM.dll version: 7.70a
```

It seems to just not work at all on macOS (yes, dongle is plugged-in)
```
% nrfjprog --ids
%
```

nordic-piks, 2023-03-01:
@PerMac @anangl Can you help with those 2 issues? (stack size and nrfjprog issue).
@cfriedt What device you have?

cfriedt, 2023-03-01:
> @PerMac @anangl Can you help with those 2 issues? (stack size and nrfjprog issue).
> @cfriedt What device you have?

I ended up downloading the prebuilt mac release from Nordic and was able to get things almost working then, but had to switch tasks.

I do have the 52840 dongle and devkit. I think nrfprog should work with the latter, right?

nordic-piks, 2023-03-01:
Yes, using devkit with be the best option - should work out of the box.

PerMac, 2023-03-01:
@cfriedt I think there was some misunderstanding. The test works on nrf52840dk, you won't be able to reproduce the failure with it. Failures were observed for:
- nrf5340dk_nrf5340_cpunet
- nrf9160dk_nrf9160_ns
-nrf9160dk_nrf9160
Not sure about nrf52840 dongle, we don't have it in our CI, nor on my desk. I can ask my colleagues if they can try this on 52 dongle.

cfriedt, 2023-03-01:
Ah, I see. That's all I have for Nordic parts locally

PerMac, 2023-03-01:
It seems you want be able to reproduce it with dongle either. From what I got it doesn't have programator on it, so without soldering something there you won't be able to use nrfjprog to program. Dfu/mcu is needed. I will try to help figuring out is some stack can be increased to simple solve the issue.

cfriedt, 2023-03-01:
It's possible that simply setting `min_ram` for the cxx test permutation might solve the issue. Is C++ STL used a great deal on these parts?

PerMac, 2023-03-01:
I played with stack sizes and found that adding 
CONFIG_MAIN_STACK_SIZE=2048
to prj.conf solved the issue on all the platforms we observed it. Setting 1024 was not yet enough. Not sure will it be efficient on stm platforms
 @FRASTM Is setting such stack size solves your issues? When debugging I found that nrf5340_cpunet was also either passing or printing nothing

FRASTM, 2023-03-01:
> I played with stack sizes and found that adding CONFIG_MAIN_STACK_SIZE=2048 to prj.conf solved the issue on all the platforms we observed it. Setting 1024 was not yet enough. Not sure will it be [...]

Yes, I noticed the STACK SIZE was too low (https://github.com/zephyrproject-rtos/zephyr/issues/55166), but I did not find the lower limit.

cfriedt, 2023-03-01:
> I played with stack sizes and found that adding CONFIG_MAIN_STACK_SIZE=2048 to prj.conf solved the issue on all the platforms we observed it. Setting 1024 was not yet enough. Not sure will it be efficient on stm platforms @FRASTM Is setting such stack size solving your issues? When debugging I found that nrf5340_cpunet was also either passing or printing nothing

@PerMac - if you are ok with it, please make a PR to add `extra_configs` the `testcase.yaml` for the `cxx` test case. If not, I can, but I'm not able to reproduce on any platforms atm.

FRASTM, 2023-03-02:
see https://github.com/zephyrproject-rtos/zephyr/pull/55349

closed 2023-03-02 by nashif