zephyrproject-rtos/zephyr #55210: samples: subsys: usb: mass; sample.usb_device_next.mass_flash_fatfs fails

nordic-piks, 2023-02-27:
**Describe the bug**
The samples/subsys/usb/mass/sample.usb_device_next.mass_flash_fatfs fails

Observed for:
- nrf52840dk_nrf52840
- nrf5340dk_nrf5340_cpuapp

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T samples/subsys/usb/mass -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-02-27T01:21:38.236Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.3.0-472-gb2512d2f534e ***[0m
[2023-02-27T01:21:38.236Z] [0mArea 0 at 0x0 on mx25r6435f@0 for 131072 bytes[0m
[2023-02-27T01:21:38.236Z] [0m[00:00:00.380,676] <err> os: ***** MPU FAULT *****[0m
[2023-02-27T01:21:38.236Z] [0m[00:00:00.380,706] <err> os:   Data Access Violation[0m
[2023-02-27T01:21:38.236Z] [0m[00:00:00.380,706] <err> os:   MMFAR Address: 0x200044bc[0m
[2023-02-27T01:21:38.236Z] [0m[00:00:00.380,737] <err> os: r0/a1:  0x20004550  r1/a2:  0x0000000e  r2/a3:  0x00000000[0m
[2023-02-27T01:21:38.236Z] [0m[00:00:00.380,737] <err> os: r3/a4:  0x2000450c r12/ip:  0x00000800 r14/lr:  0x00001cf9[0m
[2023-02-27T01:21:38.236Z] [0m[00:00:00.380,767] <err> os:  xpsr:  0x61000000[0m
[2023-02-27T01:21:38.236Z] [0m[00:00:00.380,767] <err> os: Faulting instruction address (r15/pc): 0x00000a10[0m
[2023-02-27T01:21:38.236Z] [0m[00:00:00.380,798] <err> os: >>> ZEPHYR FATAL ERROR 19: Unknown error on CPU 0[0m
[2023-02-27T01:21:38.236Z] [0m[00:00:00.380,828] <err> os: Current thread: 0x20000bd0 (unknown)[0m
[2023-02-27T01:21:38.236Z] [0m[00:00:01.585,632] <err> os: Halting system[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.2
 - Commit SHA or Version used:  zephyr-v3.3.0-472-gb2512d2f534e

tmon-nordic, 2023-03-01:
This seems to be related to main stack size. Increasing `MAIN_STACK_SIZE` to 1137 or more (zephyr commit ``865ee855d11f76c764356a6fe9643abae58b8f20``) results in sample working fine. I have to check if it is a stack overflow issue or something happens to be writing out of bounds.

tmon-nordic, 2023-03-01:
When I increased stack and enabled stack initialization, the last stack write seems to be at 1144 bytes from top. Therefore 1024 bytes long stack is definitely too small. Peak stack usage is in cbprintf functions called from logger functions.

PR #55348 (2023-03-02): samples: usb: mass: Increase main stack size
Increase main stack size for next USB device stack config to prevent stack overflow when FatFs is used. The new main stack size value matches the value used in old stack mass storage sample config.

Fixes: #55210
commit e776241f (2023-03-02): samples: usb: mass: Increase main stack size

Increase main stack size for next USB device stack config to prevent
stack overflow when FatFs is used. The new main stack size value matches
the value used in old stack mass storage sample config.

Fixes: #55210
closed 2023-03-02 by carlescufi