zephyrproject-rtos/zephyr #55230: k_sched_lock() needs to lock out also meta-IRQ threads

bartagcx, 2023-02-27:
By design `k_sched_lock()` doesn’t lock out meta-IRQ threads. It prevents only cooperative and preemptive threads from interrupting the running thread. This is however not sufficient for certain low level use cases and leads to spurious system crashes.

The use case, which I ran into, is by chance well described in the documentation of the [TZ_THREAD_SAFE_NONSECURE_ENTRY_FUNC()](https://github.com/zephyrproject-rtos/zephyr/blob/0476d6ad69f1954c5e10e48f57515b881e5a5ee1/arch/arm/include/aarch32/cortex_m/tz_ns.h#L135) wrapper.

This wrapper aims to protect a call from non-secure to secure firmware against thread switches by wrapping the function call into `k_sched_lock()` / `k_sched_unlock()`. Obviously it is assumed that this locks out all threads including meta-IRQ threads. As this is not the case currently, it can happen that a meta-IRQ thread gets ready and a thread switch happens, while execution is in the secure binary. In that case the system crashes, as described in the documentation.

In my real-world system this issue occurs every few minutes. Scheduler lock is taken, call to secure binary is done, while ongoing, a hardware interrupt triggers and makes a bottom half interrupt handler with meta-IRQ priority ready. After that the system crashes, as described, right after returning from secure binary.

Can we change the behavior of `k_sched_lock()` to lock out all threads and update the documentation to make clear that `k_sched_lock()` should be used only for this kind of low level use cases? Using `k_sched_lock()` for other use cases, like synchronization, should be discouraged in my opinion, as meta-IRQ threads should never be blocked by some likely non-critical task.

Using some other method to protect the secure call, like `irq_lock()`, is not possible, as a secure call may take significant time and blocking hardware interrupts for that duration is not acceptable.

Thanks for looking into it.

andyross, 2023-02-27:
Sorry, this is exactly by design.  "Meta" IRQ act like IRQs (it's right there in the name), and can preempt cooperative threads.  That's the whole point: you use them for tasks (like  your bottom half interrupt processing[1]) that need to run based on hardware events without interference from application scheduling decisions.

And, just like IRQs, the synchronization tools you have available are limited.  You can't use a k_mutex or k_sem to lock out an MetaIRQ either.

If I had to guess, your problem is that you're using MetaIRQs for something you shouldn't, or that you're making demands on your synchronization primitives that the OS can't meet[2] and should be using a spinlock instead.

[1] FWIW: note that true bottom halves have this problem too.  Your bottom half callback in Linux doesn't care what userspace process might be scheduled and will run to completion before the ISR returns.

[2] Among other problems, you're aware that priority-based synchronization like k_sched_lock doesn't work across SMP processors, right?

andyross, 2023-02-27:
> And, just like IRQs, the synchronization tools you have available are limited. You can't use a k_mutex or k_sem to lock out an MetaIRQ either.

I should clarify: you absolutely can do that, but those are blocking abstractions that will suspend the MetaIRQ thread on contention, which is likely not what you want your bottom half to be doing.

andyross, 2023-02-27:
> I should clarify: you absolutely can do that, but those are blocking abstractions that will suspend the MetaIRQ thread on contention, which is likely not what you want your bottom half to be doing.

But... heh, **maybe this is what you want**, since your desire seems to be that your secure binary be able to run to completion before the MetaIRQ gets a chance to finish its work.  So in that case, maybe just using a k_sem/mutex as a lock is exactly what you want.  Just be aware that by doing this, you are breaking the priority promise you made to the driver that is trying to use that MetaIRQ.  So you'll likely be coming back with another bug report like "k_mutex in a MetaIRQ suspends the thread!".

To be blunt: you have an app bug here.  The critical section between MetaIRQ and whatever code is running in your secure binary isn't safe vs. the priority and synchronization decisions you have made for it.

bartagcx, 2023-02-28:
Thanks a lot for looking into it and your quick feedback.

Maybe I didn’t explain the problem well enough, seems you didn’t get my point. Let me try to explain it in more detail.

I’m currently working on an interface driver. I’m using a meta-IRQ thread to run a bottom half handler for a GPIO interrupt. So that is exactly, what meta-IRQs are intended for.

The call to the secure binary is completely unrelated to my work. There is no relation at all between my driver and the secure binary. Just by chance another completely unrelated component in the system is obtaining random numbers by doing a call to a secure binary.

Based on measurements, I can tell, that the call to the secure binary takes about 230 µs on my system. It is performed every few minutes to generate some secure tokens for network communication. As told, it is completely unrelated and completely asynchronous to my driver.

Under full load the GPIO interrupt in my driver typically triggers every 416 µs. Due to the relatively long duration of the secure call and the high frequency of the GPIO interrupt, the probability is especially high that at some point a GPIO interrupt by chance triggers, while a secure call is ongoing.

As the GPIO interrupt unblocks its bottom half handler, which is a meta-IRQ thread, the system crashes, as switching threads is not allowed, while a secure call is ongoing.

This is not a synchronization problem and nothing that could be solved with any synchronization primitive. It is just a technical limitation of secure calls, that they must not be interrupted by thread switching. Secure calls may be interrupted only by interrupt handlers. I don’t know, why this is the case, I’m not an expert for secure calls. Secure calls are totally unrelated to my work, but this is what the documentation says and it also matches, what I see at run-time.

We would need to involve some expert for secure calls, in case you want to have an explanation, why a secure call must not be interrupted by thread switching, but I think the bottom line is that a secure call simply requires some way to lock the scheduler in a way, which makes sure that no thread switch will happen. And that is exactly what k_sched_lock() should do in my opinion and that is obviously also what the author of the mentioned wrapper for secure calls expected to happen.

Of course you could claim that this is not a scheduler problem, but a bug in the implementation of secure calls, but I don’t see any way to fix that without any method to completely lock the scheduler without blocking interrupts at the same time.

I understand that you want meta-IRQs to act like IRQs, but that is technically simply not possible and not desired to my understanding, as meta-IRQs are still threads and scheduling a thread simply has different side effects compared to running an IRQ handler.

There is another occurrence of k_sched_lock() in [efi_call()](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/x86/core/efi.c#L95). I can’t test that one, but I suppose, it could run into similar problems with meta-IRQs.

Another instance can be found in [tfm_ns_interface_dispatch()](https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/trusted-firmware-m/interface/interface.c#L47). This function as well obviously assumes that all threads are locked out.

Another instance can be found in [do_page_fault()](https://github.com/zephyrproject-rtos/zephyr/blob/main/kernel/mmu.c#L1362). I’m quite sure that this function also expects to lock out all threads including meta-IRQ threads.

Thanks again for looking into it and looking forward for your feedback.

andyross, 2023-02-28:
> As the GPIO interrupt unblocks its bottom half handler, which is a meta-IRQ thread, the system crashes, as switching threads is not allowed, while a secure call is ongoing.

Can you explain this part?  What's panicking and why?  That doesn't sound like a zephyr feature at all, more like a hardware security model that requires a modified interrupt exit path that Zephyr doesn't support?

Again, we can't modify MetaIRQ scheduling without breaking the feature that they're intended to provide.  We might as well not have them at all.

Consider this workaround:  If you can't have a MetaIRQ scheduled reliably, just make your bottom half a regular cooperative thread and it will have the behavior you want.  My guess is you don't actually want that, because then you can't rely on it running when you need it to.  Now do you see my point?  You can't have what you're asking for.

andyross, 2023-02-28:
Alternatively, you could make your "secure call" (again, do you have a link or something to the specific technology in question?) run as a highest-priority MetaIRQ thread itself, which would prevent preemption without disabling interrupts entirely.  That is, instead of k_sched_lock() it would use `k_thread_priority_set(k_current_get(), K_HIGHEST_THREAD_PRIO);` (with some logic to save off and restore the previous priority, of course).

This is almost what k_sched_lock() does already, actually, but without the (once more: intentional) exclusion of the MetaIRQ priorities and without the cute performance trick it uses to implement nesting with a single byte decrement.

andyross, 2023-02-28:
Oh... this is TrustZone.  Sorry, I finally clicked through links.  So it is a zephyr bug, just not here.  :)

The TrustZone code indeed assumes k_sched_lock() will prevent context switching by elevating a thread to a cooperative priority, but it doesn't[1].  It needs to implement the "set priority to highest" mechanism I mention above, though in this case since there's no nesting required (only one TZ call at a time until it completes) and only one CPU, it can just stick `k_thread_priority_get(k_current_get())` into a local variable and restore it on return.

@carlescufi I think you guys are still responsible for the TrustZone stuff?  Can I give this to you to find an owner?

[1] To repeat again: the ability to defeat cooperative scheduling is the only reason MetaIRQ priorities exist in the first place.

bartagcx, 2023-03-01:
Thanks. Please keep in mind that this issue is not limited to TrustZone. As mentioned in my last comment, several other components make the same assumption. I think we should involve corresponding developers, as this needs to be fixed everywhere. A few examples are listed below (with links to the affected code!), but there are even more uses of `k_sched_lock()` in the code, which all need to be reviewed, if you really want to stick to the current behavior of this function.

[TrustZone (tz_ns.h#L138)](https://github.com/zephyrproject-rtos/zephyr/blob/865ee855d11f76c764356a6fe9643abae58b8f20/arch/arm/include/aarch32/cortex_m/tz_ns.h#L138): @oyvindronningstad
[TrustZone (interface.c#L47)](https://github.com/zephyrproject-rtos/zephyr/blob/8b4407ab7cfb8319c3bf57cb01cbcb8cf9b7eb46/modules/trusted-firmware-m/interface/interface.c#L47): @joerchan @ioannisg 

[EFI calls (efi.c#L95)](https://github.com/zephyrproject-rtos/zephyr/blob/865ee855d11f76c764356a6fe9643abae58b8f20/arch/x86/core/efi.c#L95): @enjiamai @cfriedt 

[memory management (mmu.c#L1362)](https://github.com/zephyrproject-rtos/zephyr/blob/865ee855d11f76c764356a6fe9643abae58b8f20/kernel/mmu.c#L1362): @andrewboie @nashif 

[power management (pm.c#L267)](https://github.com/zephyrproject-rtos/zephyr/blob/865ee855d11f76c764356a6fe9643abae58b8f20/subsys/pm/pm.c#L267): @ceolin 

[static threads (thread.c#L780)](https://github.com/zephyrproject-rtos/zephyr/blob/865ee855d11f76c764356a6fe9643abae58b8f20/kernel/thread.c#L780): @andyross

[CMSIS RTOS v2 API (kernel.c#L48)](https://github.com/zephyrproject-rtos/zephyr/blob/865ee855d11f76c764356a6fe9643abae58b8f20/subsys/portability/cmsis_rtos_v2/kernel.c#L48): @rgundi 

I personally still believe that `k_sched_lock()` should do exactly, what the name tells and what everybody is obviously expecting, i.e. lock the scheduler completely.

The current functionality should be covered by something separate like `k_sched_mark_coop()` / `k_sched_unmark_coop()`.

andyross, 2023-03-01:
Most of those look OK to me, actually.  They're dealing with thread subclasses that don't subsume MetaIRQs, which are (sorta obviously) special: the EFI call is just a hack to reduce the likelihood of failure (the firmware re-enables interrupts and we can't stop that, long story), demand paging doesn't work for MetaIRQ contexts and shouldn't, the PM code likewise clearly wants "interrupt" handling (presumably including bottom halves) to continue on the suspending CPU, static threads can't have MetaIRQ properties, nor can CMSIS threads.

I mean.... I get what you're asking.  You got burned by a special case.  But adjusting an API for the benefit of one special case such that it hurts others is almost always a bad idea.  For sure we should improve the docs.  But IMHO the worst solution here is to make **two** "scheduler lock" calls and force everyone to understand what they mean.   Recognize that k_sched_lock() is already an oddball historical API that production code shouldn't be using except with good reason (as mentioned: it doesn't work to synchronize anything on SMP platforms).

andyross, 2023-03-02:
They... aren't any more special.  That's exactly the behavior: the scheduler will select a MetaIRQ in preference to a cooperative priority.  That's the only difference[1].  Here exactly is where it is in code: https://github.com/zephyrproject-rtos/zephyr/blob/main/kernel/sched.c#L154

And in fact this design was chosen[2] to be as close as possible to the way bottom halves work in other kernels.  The bottom half runs at the end of the interrupt, immediately before returning to user code.

> Is there any specific reason, why you want meta-IRQs in Zephyr to behave differently compared to threaded IRQs in Linux?

Because without MetaIRQ logic, "threaded IRQs" wouldn't work.   A regular user thread could block them simply by choosing a cooperative priority.  Linux doesn't have a clear equivalent to Zephyr's cooperative scheduling.  The reason we have MetaIRQs is to plug that hole and allow regular threads to do this work.

Again, the solution here is relatively straightforward: make your secure calls into MetaIRQs themselves (which they are, essentially: they're "traps" that expect to preempt all thread mode processing while they run), either by walling them off into separate threads (maybe that's the way it already works) or by directly elevating their priority.

[1] Well, also there's a "who I preempted" pointer kept, as the cooperative thread that was preempted expects that no other cooperative threads will run before it yields.

[2] Well, in addition to the fact that it could be done in 2 lines of code.  Modifying the interrupt return path is complicated as it still happens in the architecture layer on some of the older platforms.

bartagcx, 2023-03-03:
Sorry, I’m just not getting your point. I apologize upfront for all the questions, but I’m just not getting it. On one hand you are saying that meta-IRQs are super special and need very special treatment, on the other hand you are saying they are not.

> And in fact this design was chosen[2] to be as close as possible to the way bottom halves work in other kernels.

That is exactly my point, meta-IRQs don’t behave like threaded IRQs in Linux. To be more precise, it seems you don’t want them to behave like threaded IRQs. It seems you want them to behave like tasklets. But due to the fact that meta-IRQs are essentially threads, I think the design goal should be to achieve the behavior of threaded IRQs, not tasklets. And in fact the behavior is already very close to threaded IRQs, the missing scheduler locking is the only design inconsistency, which I see.

Did you already think about priority inheriting mutexes in this context? As you told in one of your previous comments, a meta-IRQ thread can request a mutex. In case this mutex is currently hold by some other low priority thread, the priority of the other thread needs to be increased to the priority of the meta-IRQ thread in order not to block the meta-IRQ thread by a low priority thread. That is also what currently happens in practice. I have just tested it on hardware. The problem is, what to do in case a third thread has just done a `k_sched_lock()` before. As it is implemented now, the second originally low priority thread, which holds the mutex, but is blocked due to the scheduler lock, will get scheduled, as it is now temporary a meta-IRQ thread. But this completely breaks your scheduler locking, as suddenly a low priority non-meta-IRQ thread is running, despite having been locked out by scheduler lock before. Always talking about a single core system of course. You see, the design is inconsistent currently.

> Because without MetaIRQ logic, "threaded IRQs" wouldn't work.

I understand the need for meta-IRQ threads and the need to preempt cooperative threads. There is no question about that. This is however conceptional independent from the question, if locking the scheduler should lock out meta-IRQ threads or not. The only reasons, why this is related, are the current implementation of the scheduler lock and the fact that some application could use `k_sched_lock()` for synchronization. That would indeed break the concept of meta-IRQs, but as you are already saying, `k_sched_lock()` should never be used by production code. It should be used only for very special cases (in low level kernel code). I fully agree to that. This however means that, as it is never used, except for locking the scheduler for a few lines in critical low level kernel code, it effectively wouldn’t break the concept of meta-IRQs, even if it locks out meta-IRQ threads. That means the proposed behavior change of `k_sched_lock()` can be safely done, this doesn’t break anything and it makes the design consistent, just by removing this unnecessary special treatment of meta-IRQ threads.

> demand paging doesn't work for MetaIRQ contexts and shouldn't

I understand that doing paging in a bottom half interrupt handler is a bad idea from performance perspective, but except from that I don’t see any reason, why it shouldn’t be possible. Let’s assume you have some A/D converter attached to your SOC over I2C. When a new value gets available, the A/D chip signals this to SOC by asserting some GPIO interrupt line. After that the SOC is expected to read the value over I2C. That is one of the most typical use cases for a threaded interrupt handler. Your GPIO interrupt handler will wake up your bottom half to do the actual I2C transfer, as this can’t be done by the top half handler. An I2C transfer obviously will take significant time, maybe in the range of milliseconds and the threaded bottom half handler needs to sleep for this duration. Swapping in and out memory pages in such a use case won’t hurt the performance any further. The swapping time is negligible compared to the overall run-time of the bottom half. So why shouldn’t it be possible?

> PM code likewise clearly wants "interrupt" handling (presumably including bottom halves) to continue on the suspending CPU

You are saying that PM code presumably wants meta-IRQs to continue running, while all other threads are blocked. That again makes meta-IRQs very special, but on the other hand you are saying, they are not special, so why do they deserve special treatment?

> static threads can't have MetaIRQ properties

As I see it, a static thread is simply a thread, which is statically defined with `K_THREAD_DEFINE()` in the code. I see there is an assert checking the priority, but I don’t see any reason for it. Why shouldn’t it be possible to use `K_THREAD_DEFINE()` to define a thread with meta-IRQ priority? In case meta-IRQ threads are not special, as you are saying, why this special treatment?

Btw. the use of `K_HIGHEST_THREAD_PRIO` is inconsistent. It is defined as
[#define K_HIGHEST_THREAD_PRIO (-CONFIG_NUM_COOP_PRIORITIES)](https://github.com/zephyrproject-rtos/zephyr/blob/5e9d3c75223fe9b008428b61654fd4252e30450e/include/zephyr/kernel.h#L54), but the check in [is_metairq()](https://github.com/zephyrproject-rtos/zephyr/blob/5e9d3c75223fe9b008428b61654fd4252e30450e/kernel/sched.c#L71) expects it to be `#define K_HIGHEST_THREAD_PRIO (-(CONFIG_NUM_COOP_PRIORITIES + CONFIG_NUM_METAIRQ_PRIORITIES))`. Let’s assume you set `CONFIG_NUM_CO [... cut ...] RIORITY_CEILING` is `0`, but the [code](https://github.com/zephyrproject-rtos/zephyr/blob/afd434696560274b8e2bf22a9f4f7d13d540e506/kernel/Kconfig#L85) tells `-127`.

For our system I have a solution. I have just disabled cooperative and meta-IRQ priorities in kernel configuration and now I’m running everything as preemptive threads with my bottom half as highest priority thread. That works perfectly fine for me. Nevertheless getting scheduler locking consistent in general seems desirable for me.

andyross, 2023-03-03:
> I have just disabled cooperative and meta-IRQ priorities in kernel configuration and now I’m running everything as preemptive threads with my bottom half as highest priority thread.

Uh... that was my suggestion?  :)  Yet again, in a system without cooperative priorities, there is no semantic meaning to "MetaIRQ" and you'd never introduce them.  They exist to defeat the scheduling behavior of cooperative threads and permit threads to be used as interrupt bottom halves.  I've sorry this was confusing.

I'm going to close this.  We're going in circles and I don't think there's anything further you don't understand.  I'll open a separate bug vs. TrustZone.

closed 2023-03-03 by andyross

oyvindronningstad, 2023-03-07:
@bartagcx Sorry I didn't read every message here, but I can't see that you've said what kind of secure binary you have. Are you using TF-M or something else? It's been a while since I worked on this, but from what I can tell, the TZ_THREAD_SAFE_NONSECURE_ENTRY_FUNC() macro is outdated. TF-M, and indeed Nordic's SPM (now deprecated) also call z_arm_save_fp_context()/z_arm_restore_fp_context() before/after nonsecure entry functions, which should prevent such crashes.

bartagcx, 2023-03-07:
@oyvindronningstad, thanks for looking into it. The TrustZone specific issue is now handled in #55412. The change proposed there should fix it. Thanks.