zephyrproject-rtos/zephyr #55232: Discovery fails to start with new LLCP implementation.

dchat-nordic, 2023-02-27:
Bluetooth failes to launch discovery with new LLCP implementation.
The case fails when the peripheral device has 128bit UUID and MTU exchange
took place.

The peripheral used is nrf52dk_nrf52832 with smp_svr sample.
The central is nrf52840dk_nrf52840 with the attached code (cropped central_hr sample, just to showcase issue).
The issue appears with new LLCP implementation. With Kconfig `CONFIG_BT_LL_SW_LLCP_LEGACY=y` the issue disappears.
The issue requires the service to have 128bit UUID *and* MTU exchange to happen.
When either of this things the issue is not present.

The issue appeared with 3.2 release.

Steps to reproduce the behavior:
1. Program the peripheral board with smp_svr sample with bluetooth overlay.
2. Build central_hr with main swapped for the attached one.
3. Program central with code from point 2.
4. See for central logs - there should be no discovery logs (for example "discover_func called").
5. Redo point 2-4 with `CONFIG_BT_LL_SW_LLCP_LEGACY=y` and the issue should disappear.

**Logs and console output**
Incorrect behavior (step 1-4):
```
Connected: C4:4B:CB:99:79:F0 (random)

MTU exchange pending

MTU exchange successful

Current MTU: 65
```

Correct behavior (step 5):
```
Connected: C4:4B:CB:99:79:F0 (random)

MTU exchange pending

MTU exchange successful

Current MTU: 65

discover_func called

[ATTRIBUTE] handle 1
```

**Environment:**
 - OS: Ubuntu 20.04
 - Toolchain Zephyr SDK 0.15.1
 - 3.2 release

kruithofa, 2023-03-07:
Not reproducable using nrf52833dk and nrf52840dk V1.1.0, but the bug report was for nrf52840DK v. 2.0.1. Further investigation required to see if it is version related

carlescufi, 2023-03-23:
@dchat-nordic is the issue present with 3.3.0? Otherwise could you bisect?

kruithofa, 2023-04-27:
Issue does not exist in 3.3.0. After discussion it is concluded that no fix for this issue is required in 3.2.0
Therefor closing the issue

closed 2023-04-27 by kruithofa