zephyrproject-rtos/zephyr #55242: drivers: xbar: Dedicated xbar driver

jeppenodgaard, 2023-02-27:
**Is your enhancement proposal related to a problem? Please describe.**

Some features requires usage of the crossbar (xbar) on some nxp processors e.g. mimxrt1050.
Currently the quadrature decoder driver configures the xbar and xbar connections.

**Describe the solution you'd like**

I would like to define the xbar connection in the pinctrl and also relate it to a group.
For example:

mimxrt1050_evk-pinctrl.dtsi
```
	pinmux_qdec1: pinmux_qdec1 {
		group0 {
			pinmux = <&iomuxc_gpio_ad_b0_09_xbar1_xbar_in21>,
				<&iomuxc_gpio_ad_b0_10_xbar1_xbar_in22>;
			xbar-connections =
				/* kXBARA1_InputIomuxXbarIn21 <-> kXBARA1_OutputEnc1PhaseAInput */
				<&xbar1 (21|0x100) (66|0x100)>,
				/* kXBARA1_InputIomuxXbarIn22 <-> kXBARA1_OutputEnc1PhaseBInput */
				<&xbar1 (22|0x100) (67|0x100)>;
			...
		};
	};

	pinmux_qdec1_single_phase: pinmux_qdec1_single_phase {
		group0 {
			pinmux = <&iomuxc_gpio_ad_b0_09_xbar1_xbar_in21>;
			xbar-connections =
				/* kXBARA1_InputIomuxXbarIn21 <-> kXBARA1_OutputEnc1PhaseAInput */
				<&xbar1 (21|0x100) (66|0x100)>,
				/* kXBARA1_InputLogicLow <-> kXBARA1_OutputEnc1PhaseBInput */
				<&xbar1 (0|0x100) (67|0x100)>;
			...
		};
	};
```
mimxrt1050_evk.overlay:
```
&qdec1 {
	status = "okay";
	pinctrl-0 = <&pinmux_qdec1>;
	pinctrl-1 = <&pinmux_qdec1_single_phase>;
	pinctrl-names = "default", "single-phase";
	counts-per-revolution = < 120 >;
};

&xbar1 {
	status = "okay";
};
```

**Additional context**
https://github.com/zephyrproject-rtos/zephyr/pull/53068#pullrequestreview-1236278435

henrikbrixandersen, 2023-02-28:
I am unsure if pinctrl is the best way to configure internal crossbars. How does Linux handle similar configuration?
CC: @gmarull, @galak

gmarull, 2023-04-17:
I'm not familiar with this peripheral, some more details would be useful to provide some guidance/ideas. But if pin related, pinctrl driver sounds like a good candidate.

jeppenodgaard, 2023-04-27:
@gmarull  I was able to find [this describtion](https://community.nxp.com/t5/i-MX-Processors/i-MX6-What-is-XBAR-How-it-works-and-%EF%BD%97hat-kind-of-structure-is-it/m-p/1314134/highlight/true#M177543) from NXP. I think it describes the XBAR pretty well:
> XBAR is a module intended to provide a flexible crossbar switch function that allows any input (typically from external GPIO or internal module outputs) to be connected to any output (typically to external GPIO or internal module inputs) under user control.

FelixWang47831, 2026-08-25:
This requirement has been done by https://github.com/zephyrproject-rtos/zephyr/pull/112088

closed 2026-08-25 by FelixWang47831