zephyrproject-rtos/zephyr #55243: Mesh sample on Xiao SEEED NRF52840 (not sense)

ShawnRobertson, 2023-02-27:
**Describe the bug**
When attempting to run the Zephyr/Bluetooth/Mesh sample on the Xiao Seeed NRF52840 the device broadcasts Mesh packets however appears to provision itself with a Network ID of 0x00, then 0x01, and so on.  It does not broadcast a mesh network name but most definitely broadcasts mesh packets.  The attached image shows, in the NRF connect app, the packets being broadcast by the device.  It does not allow for provisioning however.   I found in the main.c for the sample a self provisioning call when a button is pushed and attempted to fix this by commenting out that section, assuming that somehow the device was sending a true on a button push despite the device having only a reset button.  This did not work.  I have been able to run the sample on an NRF528040 DK from makerdiary.   

[image]

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using: Xiao Seeed NRF52840
 - What have you tried to diagnose or workaround this issue: Code manipulation on the main.c and examination of the included board files as well as using the sample on other boards.
 - Is this a regression? no

**To Reproduce**
Steps to reproduce the behavior:
We are using nrfConnect for VS Code to build a UF2 file and drag and drop to the device.
1. Create a new application with nRF SDK 2.2 and Toolchain 2.2
2. Select the  zephyr/samples/bluetooth/mesh sample
3. Create application
4. Select the xiao_ble board and the prj.conf config file. 
5. Drag and drop to xiao seeed device 

**Expected behavior**
Capability to provision the device to a ble mesh network.  

**Impact**
Showstopper

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: NRF SDK 2.2 Zephyr
 - Zephyr version 2.2

PavelVPV, 2023-03-01:
Hi @ShawnRobertson, this is true, the sample provisions itself upon button press. To disable this, comment out this code: https://github.com/zephyrproject-rtos/zephyr/blob/8b4407ab7cfb8319c3bf57cb01cbcb8cf9b7eb46/samples/bluetooth/mesh/src/main.c#L335-L374, then erase the device (so that it doesn't restore the previously stored provisioning data) and flash the sample again.

ShawnRobertson, 2023-03-01:
Hi @PavelVPV, thanks for your response.  We were able to get the code working on a new device however are unable to deprovision a device once it is provisioned once.  We utilized the nRF mesh app to provision the device to test, then deprovisioned it based on the apps guidance, however it did not actually deprovision as we can still see it broadcasting secure packets associated with the provisioned network.  We tried reflashing the sample as well as reburning the bootloader.  Neither worked.  Do you know of a way to completely wipe these?  I know this is more of a question for  Seeed but was hoping you might know.