zephyrproject-rtos/zephyr #55248: samples: bluetooth: peripheral_hids: bt_enable fails on the nucleo_wb55rg board.

2songh, 2023-02-28:
**Describe the bug**
After building the peripheral_hids sample on the nucleo_wb55rg board, an error occurs when bt_enable().

**Environment**
OS: Linux ubuntu 22.04
cmake: 3.21.1
https://docs.zephyrproject.org/3.2.0/develop/getting_started/index.html

**To Reproduce**
Steps to reproduce the behavior:

1. mkdir build; cd build
2. cmake -DBOARD=nucleo_wb55rg ..
3. west flash

**Expected behavior**
bt_enable() should operate normally.

**Logs and console output**
[image]

**Additional context**
I flashed the board with:
stm32wb5x_FUS_fw_1_0_2.bin at 0x080EC000
stm32wb5x_BLE_Stack_full_fw.bin at 0x080CB000
using STM32CubeProgrammer.

Also,
I tried this method.
https://github.com/zephyrproject-rtos/zephyr/issues/47991

erwango, 2023-02-28:
Looks better on my side:
```
*** Booting Zephyr OS build zephyr-v3.3.0-298-g258e6f07c6ec ***                 
[00:00:00.019,000] <inf> fs_nvs: nvs_mount: 2 Sectors of 4096 bytes             
[00:00:00.019,000] <inf> fs_nvs: nvs_mount: alloc wra: 0, f80                   
[00:00:00.019,000] <inf> fs_nvs: nvs_mount: data wra: 0, a0                     
Bluetooth authentication callbacks registered.                                  
[00:00:00.040,000] <inf> bt_hci_core: bt_init: No ID address. App must call set)
Bluetooth initialized                                                           
[00:00:00.040,000] <err> settings: settings_call_set_handler: set-value failure)
[00:00:00.041,000] <err> settings: settings_call_set_handler: set-value failure)
[00:00:00.042,000] <inf> bt_hci_core: bt_dev_show_info: Identity: 02:80:E1:00:0)
[00:00:00.042,000] <inf> bt_hci_core: bt_dev_show_info: HCI: version 5.2 (0x0b)0
[00:00:00.042,000] <inf> bt_hci_core: bt_dev_show_info: LMP: version 5.2 (0x0b)b
Advertising successfully started    
```

> stm32wb5x_BLE_Stack_full_fw.bin at 0x080CB000

See related note here:
https://docs.zephyrproject.org/latest/boards/arm/nucleo_wb55rg/doc/nucleo_wb55rg.html#bluetooth-and-compatibility-with-stm32wb-copro-wireless-binaries
> Note that since STM32WB Cube package V1.13.2, “full stack” binaries are not compatible anymore for a use in Zephyr and only “HCI Only” versions should be used on the M0 side.

So I suggest updating FW to HCI Only versions and report the error seen with this version if any.

carlescufi, 2023-03-02:
@2songh please close this if @erwango's solution works for you.

2songh, 2023-03-03:
I'm sorry to late confirmation.

I checked three binary data.
**stm32wb5x_BLE_HCI_AdvScan_fw.bin**
**stm32wb5x_BLE_HCILayer_extended_fw.bin**
**stm32wb5x_BLE_HCILayer_fw.bin**
Which one should I use?

Can I know exact address?
(ex. stm32wb5x_BLE_Stack_full_fw.bin at **0x080CB000**)

erwango, 2023-03-03:
> Can I know exact address?

Information is in STM32WB Cube package, under [Projects/STM32WB_Copro_Wireless_Binaries/STM32WB5x](https://github.com/STMicroelectronics/STM32CubeWB/tree/master/Projects/STM32WB_Copro_Wireless_Binaries/STM32WB5x)

2songh, 2023-05-03:
I guess I won't have to check the issue anymore.
Close the issue.

closed 2023-05-17 by kartben