zephyrproject-rtos/zephyr #55267: ARC linker script: application overflows the ICCM

jacoba-at, 2023-02-28:
Hi,
I'm in the progress of migrating our code base from Zephyr v2.7.4 to v3.3.0. We are using the GNU toolchain from SDK 0.15.
I noticed that when I build our app, the map file shows that the 'datas' section exceeds the size of the ICCM. This did not happen when I built the same app in 2.7.4. 
I see in the map file that this happens because the linker creates a *fill* area at the end of the rodata section, which causes  __rodata_region_end to point to the end of the ICCM.
 The LOADADDR of the datas section is right after this, leaving no space to the datas section in the ICCM. This leads me to lose a lot of valuable ICCM area and it is critical for us as it is our ICCM ROM.
Was there any padding or alignment that was added between these versions that causes this issue?
How can I overcome this?

I think the offending line is include/zephyr/arch/arc/v2/linker.ld:140:
           MPU_ALIGN(ABSOLUTE(.) - __rom_region_start);

Once I remove it the build passes.
What do you guys think?

ruuddw, 2023-02-28:
This was added in #46775 to support XIP with MPU protections. And with power-of-two size/alignment restrictions this indeed may be wasting some memory. If MPU protection is not relevant, or if you can e.g. merge more regions with compatible attributes, the alignment is probably not relevant to you.

evgeniy-paltsev, 2023-03-02:
Hi @jacoba-at,

may I ask you to provide build log of v2.7.4 and v3.3.0?

Most interesting part is memory region statistics (which is shown at the end of the build log) and the error message you see with  v3.3.0. 

It also would be good to have `.config` file to check Zephyr features / configuration.

Thanks!

jacoba-at, 2023-03-02:
[build_2.7.4.txt](https://github.com/zephyrproject-rtos/zephyr/files/10873546/build_2.7.4.txt)
[config_2.7.4.txt](https://github.com/zephyrproject-rtos/zephyr/files/10873556/config_2.7.4.txt)
[zephyr_2.7.4.map.txt](https://github.com/zephyrproject-rtos/zephyr/files/10873561/zephyr_2.7.4.map.txt)
[build_v3.3.0.txt](https:
[zephyr_pre0_3.3.0.map.txt](https://github.com/zephyrproject-rtos/zephyr/files/10873571/zephyr_pre0_3.3.0.map.txt)
//github.com/zephyrproject-rtos/zephyr/files/10873567/build_v3.3.0.txt)
[config_3.3.0.txt](https://github.com/zephyrproject-rtos/zephyr/files/10873579/config_3.3.0.txt)

jacoba-at, 2023-03-02:
@evgeniy-paltsev Our ICCM ROM starts at 0x60050000 and ends at 0x60060000.
Note in the 3.3.0 map that __rodata_region_end is set to be the same as __rom_region_end, with a fill of 0x6680 bytes, while in the 2.7.4 map, the __rodata_region_end  is before  __rom_region_end.

PR #59207 (2023-06-14): ARC: fix excessive ROM memory consumption 
This extra MPU alignment of RAMABLE_REGION is only required if we put ROMABLE_REGION and RAMABLE_REGION into the same (continuous) memory (i.e. SRAM) - so we won't get beginning of the RAMABLE_REGION in the end of ROMABLE_REGION MPU aperture.

If we use different regions (ICCM & DCCM, FLASH & SRAM, etc...) we don't need this extra MPU alignment.

Let's drop it to decrease ROM region memory usage.

Fixes: #55267
closed 2023-06-14 by nashif