zephyrproject-rtos/zephyr #55278: CSI driver and camera driver initialization priorities

jeronimoagullo, 2023-02-28:
**Environment and board**
I am using the platform and camera recommended by the sample readme file:
- board: mimxrt1064_evk
- camera: mt9m114
- Zephyr version: 3.3.99 (main branch)
- OS: Windows

**Describe the bug**
The sample app **samples/subsys/video/capture** fails to get the camera ready with the following output:

```
[00:00:00.100,000] <err> mt9m114: Unable to read chip ID
*** Booting Zephyr OS build zephyr-v3.3.0-65300-ge5744284d44a ***
[00:00:00.100,000] <err> main: csi@402bc000: device not ready.
```

**Reason of the Issue**
I added some log messages to the code and I found out the root of the issue, which is **the initialization priority of mcux_csi and mt9m114 drivers**.

The `mt9m114` camera needs **an external clock to work**, otherwise it is not possible to communicate with it through i2c bus. It is the reason of the error message `mt9m114: Unable to read chip ID`. Then, since the video sensor is not ready, the CSI driver fails to initialize and the app main function returns. 

The required external clock signal is generated by the CSI driver in the following line, in which the _Control Pin API_ apply the CSI configuration to the pins.
```c
err = pinctrl_apply_state(config->pincfg, PINCTRL_STATE_DEFAULT);
```

However, the above code line never runs because the `video_mcux_csi_init` function returns before reaching this line since the camera sensor is not ready.

**Propose solution**
The `CONFIG_VIDEO_INIT_PRIORITY` must be higher (lower value) than the `CONFIG_KERNEL_INIT_PRIORITY_DEVICE` to initialize the CSI driver before the camera driver. For example, with value 49.

Besides, the sensor cannot be checked in the `video_mcux_csi_init` function, it should be removed from here. This checking step must be done once the camera sensor is initialized, to be compliance with the Zephyr documentation, the code snippet
```c
if (!device_is_ready(config->sensor_dev)) {
		return -ENODEV;
}
```
can be call in the function `video_mcux_csi_stream_start` for instance, to check this object before using the pointer.

**Next step**
If you agree, I will open a pull request to fix this issue.

dleach02, 2023-03-10:
@jeronimoagullo, thank you for the report. Please go ahead and provide a PR to address this.

dleach02, 2023-05-17:
Fixed with #53381

closed 2023-05-17 by dleach02