zephyrproject-rtos/zephyr #55281: Coding style documentation

jori-nordic, 2023-02-28:
Reviewers for Bluetooth patches seem to have a clear idea of the coding standard that the Bluetooth subsystem is supposed to follow, but this isn't documented anywhere.

Create a "coding style" page with all the do's and don'ts here https://docs.zephyrproject.org/latest/connectivity/bluetooth/index.html

It can include stuff like doxygen commenting style, newline before return, scope of vars, etc..

Thalley, 2023-02-28:
We used to have a discussion that was closed due to inactivity too: https://github.com/zephyrproject-rtos/zephyr/discussions/41120

There are a lot of code styles that we haven't agreed on, and thus probably shouldn't be enforced.

Thalley, 2023-03-01:
> the purpose of this task really is that once we have that document, a style comment on a PR should have a corresponding entry in that document, or shouldn't be made. 

That assumes that we can agree on such rules :D And we would also need to ensure that they do not conflict with any of the non-BT Coding Guidelines (https://docs.zephyrproject.org/latest/contribute/coding_guidelines/index.html)

> That way we cut down on the review noise and can focus on the actual behavior/functionality.
IMO the only way we can really achieve this, is if we can have a tool in CI run these tests, but that would require general support from Zephyr, as such a tool should apply for the entire Zephyr project. 

Personally I agree with your statements, but having tried to walk down the path of BT specific guidelines, I don't have high hopes for a solution. I won't take the lead on this, but I'll support anyone that will :)

jori-nordic, 2023-03-01:
Alright, I'll assign it to myself then. I'll try to start with a small subset we can all agree on.

jori-nordic, 2023-03-20:
another instance of this https://github.com/zephyrproject-rtos/zephyr/pull/55952#issuecomment-1476146602

jori-nordic, 2024-04-23:
More material: https://github.com/zephyrproject-rtos/zephyr/pull/71638#discussion_r1575745662

Thalley, 2024-05-10:
Another: https://github.com/zephyrproject-rtos/zephyr/pull/72090#discussion_r1596897603

jori-nordic, 2024-05-14:
why, when and where to use `const`
https://github.com/zephyrproject-rtos/zephyr/pull/72090#discussion_r1599731797

jori-nordic, 2024-05-27:
What about `#ifdef` comments: https://github.com/zephyrproject-rtos/zephyr/pull/72090#discussion_r1603242269

jori-nordic, 2024-05-28:
implicit casts to bool: https://github.com/zephyrproject-rtos/zephyr/pull/72090#discussion_r1616716521

Thalley, 2024-05-28:
> implicit casts to bool: [#72090 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/72090#discussion_r1616716521)

This is required by MISRA 14.4: https://docs.zephyrproject.org/latest/contribute/coding_guidelines/index.html#id29

We don't yet require PRs to comply with the MISRA rules IIRC

jori-nordic, 2024-05-28:
`retval`, `ret` or `err`? hmmm https://github.com/zephyrproject-rtos/zephyr/pull/72674#discussion_r1601351663

Thalley, 2024-05-31:
Should values in enum use the enum name as prefix? 
https://github.com/zephyrproject-rtos/zephyr/pull/73527#discussion_r1621925972