zephyrproject-rtos/zephyr #55283: Interfacing multiple DS18b20

CodingGhost, 2023-02-28:
I am wondering how to Interface multiple ds18b20 by using the sensor API.
I already have one sensor working, and I think I could get multiple to work by putting more sensors in the devicetree, but the problem is, that I wont know which is which, as I dont have the ID (ROM address) of the sensors.

Is there a way to Interface multiple ds18b20 in a way that lets me permanently identify them?
Another thing to think of, might be the situation where you dont know how many sensors are connected. In my case, I know there is a max of 3 sensors though so it should not be a problem.
@mbolivar 
@str4t0m

CodingGhost, 2023-03-09:
any info/opinions about this?

str4t0m, 2023-03-09:
Yes interfacing multiple ds18b20 is possible. And putting multiple sensors in the device tree will allow you to handle the w1 sensors like any other sensors during runtime. It is also used as an indication to the 1-wire subsystem to use a match rom command instead of the skip rom command in order to select a specific slave.

However due to the fact that 1-wire slaves can only be identified via a unique rom-id, an additional setup step is required where you assign a rom-id to a device. For the ds18b20 this can be done via the sensor attribute interface.
You might have a look at [w1_sensor.h](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/sensor/w1_sensor.h).

>  I dont have the ID (ROM address) of the sensors.

You can use `w1_search_bus()` to list all devices on the bus.

> Is there a way to Interface multiple ds18b20 in a way that lets me permanently identify them?

This duty really depends on the use case and is therefore the responsibility of the application.
For example, in order to be able to address a specific sensor(i.e. front sensor, rear sensor), you could execute a routine once, where you assign a rom to a specific "object". Store that assignment on a permanent memory (i.e. nvs, settings). Your application then can assign the roms during program initialization.

> the situation where you dont know how many sensors are connected

Yeah this would be a valid use case, and you could detect them via a search. However if such slaves don't have programmed any identifier onto them it would be hard to find out where they are connected and what they measure.

CodingGhost, 2023-03-09:
thanks for your answer!
so in my case, as I have 1-3 sensors connected, I put 3 sensors in the device tree, correct?
then I can use w1_search_rom to find out how many sensors are currently connected and what rom they have.
but how do I get the device now? Previously I did DEVICE_DT_GET_ANY(maxim_ds18b20);
but thats not correct now I guess.
I dont quite unserstand what w1_rom_to_sensor_value does either to be honest. Do I use that function instead of the normal sensor_sample_fetch?

CodingGhost, 2023-03-11:
I tried to understand how todo it, but I cant figure it out... I can search and get the ROMs of all connected devices, and I think I understand what w1_rom_to_sensor_value does, but I dont know why I would want to do this..?
I somehow need to get a struct device that I can use with sensor_channel_get, so I would want a w1_rom_to_device function or not?