zephyrproject-rtos/zephyr #55292: Add Binary blobs for Infineon PSoC 6 BLE Middleware

sreeramIfx, 2023-03-01:
## Origin

I would like to add blobs for BLE middleware to enable Bluetooth support on CY8CPROTO_063_BLE board. These blobs originate from https://github.com/Infineon/bless

## Type

- [ ] Precompiled library
- [ ] Firmware image

## Module

hal_infineon: https://github.com/zephyrproject-rtos/hal_infineon/

## Purpose

PSoC 6 BLE Middleware contains a comprehensive API to configure the BLE Controller and the underlying chip hardware.

## Pull Request

Yet to create the PR

## Dependencies

None

## License

CYPRESS (AN INFINEON COMPANY) END USER LICENSE AGREEMENT

PLEASE READ THIS END USER LICENSE AGREEMENT ("Agreement") CAREFULLY BEFORE
DOWNLOADING, INSTALLING, COPYING, OR USING THIS SOFTWARE AND ACCOMPANYING
DOCUMENTATION.  BY DOWNLOADING, INSTALLING, COPYING OR USING THE SOFTWARE,
YOU ARE AGREEING TO BE BOUND BY THIS AGREEMENT.  IF YOU DO NOT AGREE TO ALL
OF THE TERMS OF THIS AGREEMENT, PROMPTLY RETURN AND DO NOT USE THE SOFTWARE.
IF YOU HAVE PURCHASED THIS LICENSE TO THE SOFTWARE, YOUR RIGHT TO RETURN THE
SOFTWARE EXPIRES 30 DAYS AFTER YOUR PURCHASE AND APPLIES ONLY TO THE ORIGINAL
PURCHASER.

1. Definitions.

    "Software" means this software and any accompanying documentation,
      including any upgrades, updates, bug fixes or modified versions provided
      to you by Cypress.

    "Source Code" means software in human-readable form.

    "Binary Code" means the software in binary code form such as object code or
      an executable.

    "Development Tools" means software that is intended to be installed on a
      personal computer and used to create programming code for Firmware,
      Drivers, or Host Applications.  Examples of Development Tools are
      Cypress's PSoC Creator software, Cypress's AIROC SDKs, and Cypress's
      ModusToolbox software.

    "Firmware" means software that executes on a Cypress hardware product.

    "Driver" means software that enables the use of a Cypress hardware product
      on a particular host operating system such as GNU/Linux, Windows, MacOS,
      Android, and iOS.

    "Host Application" means software that executes on a device other than a
      Cypress hardware product in order to program, control, or communicate
      with a Cypress hardware product.

    "inf File" means a hardware setup information file (.inf file) created by
      the Software to allow a Microsoft Windows operating system to install
      the driver for a Cypress hardware product.

2. License.  Subject to the terms and conditions of this Agreement, Cypress
[... 143 lines cut ...]
or revised from time to time, and may be provided to its third party sales
representatives, distributors and other entities conducting sales activities
for Cypress for sales-related and other business purposes.

12. General.  This Agreement will bind and inure to the benefit of each
party's successors and assigns, provided that you may not assign or transfer
this Agreement, in whole or in part, without Cypress' written consent.  This
Agreement shall be governed by and construed in accordance with the laws of
the State of California, United States of America, as if performed wholly
within the state and without giving effect to the principles of conflict of
law.  The parties consent to personal and exclusive jurisdiction of and venue
in, the state and federal courts within Santa Clara County, California;
provided however, that nothing in this Agreement will limit Cypress' right to
bring legal action in any venue in order to protect or enforce its
intellectual property rights.  No failure of either party to exercise or
enforce any of its rights under this Agreement will act as a waiver of such
rights.  If any portion of this Agreement is found to be void or
unenforceable, the remaining provisions of this Agreement shall remain in full
force and effect.  This Agreement is the complete and exclusive agreement
between the parties with respect to the subject matter hereof, superseding and
replacing any and all prior agreements, communications, and understandings
(both written and oral) regarding such subject matter.  Any notice to Cypress
will be deemed effective when actually received and must be sent to Cypress
Semiconductor Corporation, ATTN: Chief Legal Officer, 198 Champion Court, San
Jose, CA 95134 USA.

ericsomto, 2023-03-01:
Waiting for you to push pr

nashif, 2023-05-03:
approved by the TSC.

closed 2023-07-05 by sreeramIfx