zephyrproject-rtos/zephyr #55296: CONFIG_ROM_START_OFFSET change to be added to cause west flash reset fails, in NXP RT11xx platforms

hakehuang, 2023-03-01:
**Describe the bug**
for RT1170 use below command to download the image, debug reset does not work, but hardwarereset works.
`
west build -b mimxrt1170_evk_cm7
west flash --runner jlink --tool-opt='-SelectEmuBySN 000725371294' --  '--device=MIMXRT1176xxxA_M7' '--reset-after-load'
`

the serial console has no output at all, press reset key works. but this blocks twister testing. bisect found below commit introduce such problem

`
commit 44628735b870b2806bbea47477c3300bb624ad31 (refs/bisect/bad)
Author: Daniel Leung <daniel.leung@intel.com>
Date:   Mon Feb 13 13:31:26 2023 -0800

    linker: rom_start_offset: add to address instead of set

    The CONFIG_ROM_START_OFFSET is supposed to be added to
    the current when linking, instead of having the current
    address set to it. So fix that.

    Not sure why it worked up to this point, but llvm/clang/lld
    complained that it could not move location counter backward.

    Signed-off-by: Daniel Leung <daniel.leung@intel.com>

diff --git a/arch/common/rom_start_offset.ld b/arch/common/rom_start_offset.ld
index 2e82f30d71..8546391614 100644
--- a/arch/common/rom_start_offset.ld
+++ b/arch/common/rom_start_offset.ld
@@ -4,5 +4,5 @@
  * SPDX-License-Identifier: Apache-2.0
  */

-. = CONFIG_ROM_START_OFFSET;
+. += CONFIG_ROM_START_OFFSET;
 . = ALIGN(4);
`

**Expected behavior**
west flash can works

**Impact**
twister testing

**Logs and console output**
whent this happen no console output

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  zephyr-v3.3.0-530-g1751c8f0f5

this impacts mimxrt1170_evk_cm7 and mimxrt1160_evk_cm7

hakehuang, 2023-03-01:
@dleach02 @mmahadevan108 this blocks all testing for RT11xx series, need check.

dleach02, 2023-03-01:
Pinging @dcpleung for awareness

dcpleung, 2023-03-01:
Just revert it and we will deal with llvm/lld later.

PR #55422 (2023-03-30): Revert "linker: rom_start_offset: add to address instead of set"
Revert commit 44628735b870b2806bbea47477c3300bb624ad31

This commit broke the ability for nxp rt series to reset except with power cycle

fixes #55296
closed 2023-03-30 by nashif