zephyrproject-rtos/zephyr #55331: footprint: actions: footprint/track.py should not silently fail

kartben, 2023-03-01:
I am not sure of the history behind this script, but right now it doesn't look like it's meant to fail on errors. 
While running it, I caught #55328, which is a great example of what I think the intention of the script and GH actions was in the first place. 
Apparently the "official" Bluetooth Mesh sample on micro:bit has stopped fitting in RAM, and it would have been great to know when/why to potentially catch unnecessary bloating of a particular subsystem.

stephanosio, 2023-03-01:
This script is not meant to "fail." It is there for generating footprint reports (to be used locally or uploaded to the footprint tracking dashboard).

See https://github.com/zephyrproject-rtos/zephyr/issues/55328#issuecomment-1450395452

closed 2023-03-01 by stephanosio

kartben, 2023-03-01:
> This script is not meant to "fail." It is there for generating footprint reports (to be used locally or uploaded to the footprint tracking dashboard).
> 
> See [#55328 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/55328#issuecomment-1450395452)

Gotcha, thanks. Didn't realize that this could/should be treated the way you suggest in the linked comment, thanks for the context!

BTW, how does one get access to the footprint tracking dashboard? I'm definitely interested!

stephanosio, 2023-03-01:
> BTW, how does one get access to the footprint tracking dashboard? I'm definitely interested!

It can be accessed here:
https://stats.zephyrproject.org/testing/d/SSxr5P6Gk/footprint-tracking?orgId=1