zephyrproject-rtos/zephyr #55336: doc: Consider listing/promoting the various project health dashboards in the documentation

kartben, 2023-03-01:
It feel like I'm discovering new hidden gems every day in terms of publicly accessible dashboards that put the spotlight on some interesting metrics that can help people evaluate Zephyr's overall "health".

I don't think these are advertised anywhere (or are they?), so I would like to use this issue as a way to 1/ list them and 2/ eventually come with a right place to add them more prominently in the documentation so that folks evaluating Zephyr, and anyone really, can easily browse them.

Here's what I'm aware of, others should feel free to jump in:

- https://stats.zephyrproject.org: bug stats, footprint stats
- https://app.codecov.io/gh/zephyrproject-rtos/zephyr: code coverage

simhein, 2023-03-02:
@kartben Here is another Dashboard: https://kibana.zephyrproject.io
This one is WIP and was published not long ago in the discord testing channel(https://discord.com/channels/720317445772017664/733037944922964069/1069647539621597275)
As far as I understood, this dashboard shall be the central one for zephyr in the future.

stephanosio, 2023-03-03:
> As far as I understood, this dashboard shall be the central one for zephyr in the future.

Correct. It will be eventually replacing `stats.zephyrproject.org`.