zephyrproject-rtos/zephyr #55340: nrfjprog runner only supports extra parameters for program commands

mrfuchs, 2023-03-01:
The nrfjprog runner only supports extra parameters (passed by "--tool-opt") for program commands, but ignores it for others like reset and recover.
This leads to an error like shown below when trying to access remote adapters (which are not officially supported by nrfjprog, as it seems, but anyway...):

```console
root@a6fdc28834bc:/workdir/app/nrf_netcore# west flash --skip-rebuild -r nrfjprog --dev-id 1050057458 --erase --tool-opt="--ip 192.168.56.1"
-- west flash: using runner nrfjprog
-- runners.nrfjprog: mass erase requested
-- runners.nrfjprog: Flashing file: /workdir/app/nrf_netcore/build/zephyr/zephyr.hex
[ #################### ]   0.736s | Erase file - Done erasing                                                          
[ #################### ]   0.892s | Program file - Done programming                                                    
[ #################### ]   0.874s | Verify file - Done verifying                                                       
ERROR: No debuggers were discovered.
NOTE: For additional output, try running again with logging enabled (--log).
NOTE: Any generated log error messages will be displayed.
FATAL ERROR: command exited with status 41: nrfjprog --pinreset -f NRF53 --snr 1050057458
```

When "tool-opt" parameters are taken into account for all commands, remote debuggers (connected via Segger J-Link Remote Server) can successfully be used for erasing, programming and verifying nRF boards (like my nRF5340-DK):

```console
root@a6fdc28834bc:/workdir/app/nrf_netcore# west flash --skip-rebuild -r nrfjprog --dev-id 1050057458 --erase --tool-opt="--ip 192.168.56.1"
-- west flash: using runner nrfjprog
-- runners.nrfjprog: mass erase requested
-- runners.nrfjprog: Flashing file: /workdir/app/nrf_netcore/build/zephyr/zephyr.hex
[ #################### ]   0.775s | Erase file - Done erasing                                                          
[ #################### ]   0.899s | Program file - Done programming                                                    
[ #################### ]   0.891s | Verify file - Done verifying                                                       
Applying pin reset.
-- runners.nrfjprog: Board with serial number 1050057458 flashed successfully.
```

These are the changes I applied to [nrfjprog.py](https://github.com/zephyrproject-rtos/zephyr/blob/a2f203d/scripts/west_commands/runners/nrfjprog.py):

```diff
--- a/scripts/west_commands/runners/nrfjprog.py
+++ b/scripts/west_commands/runners/nrfjprog.py
@@ -205,10 +205,10 @@ class NrfJprogBinaryRunner(ZephyrBinaryRunner):
         if self.family == 'NRF53':
             self.check_call(['nrfjprog', '--recover', '-f', self.family,
                              '--coprocessor', 'CP_NETWORK',
-                             '--snr', self.dev_id])
+                             '--snr', self.dev_id] + self.tool_opt)
 
         self.check_call(['nrfjprog', '--recover',  '-f', self.family,
-                         '--snr', self.dev_id])
+                         '--snr', self.dev_id] + self.tool_opt)
 
     def program_hex(self):
         # Get the nrfjprog command use to actually program self.hex_.
@@ -354,14 +354,14 @@ class NrfJprogBinaryRunner(ZephyrBinaryRunner):
     def reset_target(self):
         if self.family == 'NRF52' and not self.softreset:
             self.check_call(['nrfjprog', '--pinresetenable', '-f', self.family,
-                             '--snr', self.dev_id])  # Enable pin reset
+                             '--snr', self.dev_id] + self.tool_opt)  # Enable pin reset
 
         if self.softreset:
             self.check_call(['nrfjprog', '--reset', '-f', self.family,
-                             '--snr', self.dev_id])
+                             '--snr', self.dev_id] + self.tool_opt)
         else:
             self.check_call(['nrfjprog', '--pinreset', '-f', self.family,
-                             '--snr', self.dev_id])
+                             '--snr', self.dev_id] + self.tool_opt)
 
     def do_run(self, command, **kwargs):
         self.require('nrfjprog')

```

carlescufi, 2023-03-01:
I noticed this as well during a rework I am currently working on, so I will be addressing it.

closed 2023-03-06 by mbolivar-nordic

mrfuchs, 2023-03-07:
Thanks, @carlescufi. I can confirm your patch in #55484 fixes this issue.

carlescufi, 2023-03-07:
> Thanks, @carlescufi. I can confirm your patch in #55484 fixes this issue.

Welcome, next is [nrfutil ](https://www.nordicsemi.com/Products/Development-tools/nrf-util)support.

PR #55484 (2023-03-06): scripts: runners: nrfjprog: Multiple improvements
```
    scripts: runners: nrfjprog: Use --tool-opt in all operations

    Pass on the tool-specific options to nrfjprog during all operations, and
    not only when programming. This is useful when combined with options
    like --ip, that allows west flash to be used over the network.

    Fixes #55340.

    Signed-off-by: Carles Cufi <carles.cufi@nordicsemi.no>
```
```
    scripts: runners: tests: nrfjprog: Test --tool-opt

    Add tests for the --tool-opt command-line switch, to ensure that if the
    user includes additional tool options th [... cut]
commit e073210e (2023-03-06): scripts: runners: nrfjprog: Use --tool-opt in all operations

Pass on the tool-specific options to nrfjprog during all operations, and
not only when programming. This is useful when combined with options
like --ip, that allows west flash to be used over the network.

Fixes #55340.