zephyrproject-rtos/zephyr #55358: STM32H7 SDMMC clock requires HSI48

mjaun, 2023-03-02:
**Describe the bug**
With #53898 the SD card stopped working for me. I am working with a STM32H743 controller. The current definition in `stm32h7.dtsi` specifies HSI48 as clock source. According to the reference manual the clock source for the SDMMC peripherals is either PLL1Q (default) or PLL2R, therefore I think the current definition is wrong. As a workaround I can simply enable the HSI48 clock then everything works fine, but I guess the HSI48 is not actually used.

**To Reproduce**
1. Enable `sdmmc1`, but don't enable `clk_hsi48` in the device tree.
2. Access the SD card (in my case LittleFS is used)

**Expected behavior**
I can run the SDMMC driver with PLL1Q as default clock source without having to enable additional clocks.

**Impact**
Annoyance

**Logs and console output**
Upon mounting the file system on the SD card I get the following log output:
```
[00:00:09.398,000] <inf> littlefs: LittleFS version 2.5, disk version 2.0                                        
[00:00:09.405,000] <err> stm32_sdmmc: Failed to enable SDMMC domain clock                                                                              
[00:00:09.413,000] <err> stm32_sdmmc: failed to init clocks                                                                                            
[00:00:09.419,000] <err> littlefs: Storage init ERROR!                                                                                                 
[00:00:09.425,000] <err> fs: fs mount error (-5)
```

**Environment (please complete the following information):**
 - Ubuntu 20.04
 - Zephyr SDK 0.15.2
 - Zephyr 3.3.0

FRASTM, 2023-03-07:
Right, sm32H7 refMan (RM0468 or RM 0433 or RM455)  do not specify HSI48 as clock source for the sdmmc but 

> Bit 16 SDMMCSEL: SDMMC kernel clock source selection
> 0: pll1_q_ck clock is selected as kernel peripheral clock (default after reset)
> 1: pll2_r_ck clock is selected as kernel peripheral clock

 So that DTS of the stm32H7 sholud have : 
```
clocks = <&rcc STM32_CLOCK_BUS_AHB2 0x00000100>,
				 <&rcc STM32_SRC_PLL1_Q SDMMC_SEL(0)>;

```

FRASTM, 2023-03-07:
@mjaun maybe you could just use  this patch in your environment when disabling the HSI48

```
diff --git a/dts/arm/st/h7/stm32h7.dtsi b/dts/arm/st/h7/stm32h7.dtsi
index 4dee62727c..b67e93b6c0 100644
--- a/dts/arm/st/h7/stm32h7.dtsi
+++ b/dts/arm/st/h7/stm32h7.dtsi
@@ -861,7 +861,7 @@
 			compatible = "st,stm32-sdmmc";
 			reg = <0x52007000 0x400>;
 			clocks = <&rcc STM32_CLOCK_BUS_AHB3 0x00010000>,
-				 <&rcc STM32_SRC_HSI48 SDMMC_SEL(0)>;
+				 <&rcc STM32_SRC_PLL1_Q SDMMC_SEL(0)>;
 			resets = <&rctl STM32_RESET(AHB3, 16U)>;
 			interrupts = <49 0>;
 			status = "disabled";
@@ -871,7 +871,7 @@
 			compatible = "st,stm32-sdmmc";
 			reg = <0x48022400 0x400>;
 			clocks = <&rcc STM32_CLOCK_BUS_AHB2 0x00000100>,
-				 <&rcc STM32_SRC_HSI48 SDMMC_SEL(0)>;
+				 <&rcc STM32_SRC_PLL1_Q SDMMC_SEL(0)>;
 			resets = <&rctl STM32_RESET(AHB2, 8U)>;
 			interrupts = <124 0>;
 			status = "disabled";
```

closed 2023-03-30 by fabiobaltieri