zephyrproject-rtos/zephyr #55366: Blutooth: Mesh: missing upper tester interface for reporting received beacons

KKopyscinski, 2023-03-02:
**Describe the bug**
Some tests (like MESH/NODE/IVU/BI-06-C) require Upper tester to verify if beacon with given parameters was received (for example with certain IV index). There should be a way to report to application that beacon was received and what it was.

**To Reproduce**
This is required by PTS tests.

**Expected behavior**
There should be a way for application to be notified about received beacons. This could be a user defined callback, which can be hidden under `CONFIG_BT_TESTING` (like  `bt_test_mesh_model_recv`).

**Impact**
This is not explicitly blocking testing, but proper verification must be added.

**Logs and console output**
N/A

**Environment (please complete the following information):**
PTS tests can be executed using `tester` application together with [autopts](https://github.com/auto-pts/auto-pts) or with mesh_shell

KKopyscinski, 2023-04-24:
https://github.com/zephyrproject-rtos/zephyr/pull/57162

kruithofa, 2023-05-25:
@KKopyscinski please re-open this issue if #57162 does not fix it

closed 2023-05-25 by kruithofa