zephyrproject-rtos/zephyr #55379: flash_mcux_flexspi_nor: incorrect enabling quad mode

GrixaYrev, 2023-03-02:
**Describe the bug**
For enable quad mode `flash_mcux_flexspi_nor` driver [send](https://github.com/zephyrproject-rtos/zephyr/blob/69966dd452e5ebf9f057b393b1e5cdac9ea2034b/drivers/flash/flash_mcux_flexspi_nor.c#L340) value `0x40` to status register 1. It is valid value for **IS25WP064**, which is used in `mimxrt1064_evk` board), but for [Winbond **W25Q128JV**](https://www.winbond.com/hq/product/code-storage-flash-memory/serial-nor-flash/?__locale=en&partNo=W25Q128JV) need set bit 2 in status register 2.

**Expected behavior**
Driver should select way to enable quad mode accordance vendor id. List of ways (vendors) should be extensible.

------

Later i will try to propose PR with fix. Please, give me comments.

fabiobaltieri, 2023-03-07:
Hi! Thanks for reporting the issue, can you send the PR and then we can comment and review on it?

henrikbrixandersen, 2023-03-07:
Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

mbrunnen, 2023-07-28:
Would be a dedicated driver for `nxp,imx-flexspi-w25q128jv` acceptable?

GrixaYrev, 2023-07-31:
@mbrunnen, i think it is redundant and violates principle DRY.

PR #61905 (2024-02-01): drivers: flash: flash_mcux_flexspi_nor: support reading SFDP for flash configuration
Support SFDP probe in flexspi nor driver. This probe will allow the
flash driver to dynamically configure quad spi flashes for 1-4-4 mode,
expanding the flash chips supported with this driver.

The following data is read from the SFDP header:
- quad enable method
- fast read command (1-4-4 is maximum supported)

This PR has been validated with the following Flash chips:
- IS25WP064
- W25Q512NWEIQ
- W25Q32JV

Fixes #55379
~Note: this PR currently depends on #61625~
commit 9d7a3fb6 (2024-02-01): drivers: flash: flash_flexspi_nor: support SFDP probe

Support SFDP probe in flexspi nor driver. This probe will allow the
flash driver to dynamically configure quad spi flashes for 1-4-4 mode,
expanding the flash chips supported with this driver.

The following data is read from the SFDP header:
- quad enable method
- fast read command (1-4-4 is maximum supported)

Fixes #55379
closed 2024-02-01 by carlescufi