zephyrproject-rtos/zephyr #55384: QSPI (1_4_4) Flash not working for board stm32f723e_disco

whati001, 2023-03-02:
**Describe the bug**
The [st,stm32-qspi-nor](https://docs.zephyrproject.org/latest/build/dts/api/bindings/flash_controller/st,stm32-qspi-nor.html) driver seems to have some issue with the board stm32f723e_disco if we use it with four (1_4_4) data lines. In this mode, the driver does either not write or read successfully from the flash.

**Reproduce**
This can be tested by running the [flash_shell](https://docs.zephyrproject.org/latest/samples/drivers/flash_shell/README.html) sample:
```bash
$ west build -p -b stm32f723e_disco samples/drivers/flash_shell
$ west flash
```

```dtc
&quadspi {
	pinctrl-names = "default";
	pinctrl-0 = <&quadspi_clk_pb2 &quadspi_bk1_ncs_pb6
                &quadspi_bk1_io0_pc9 &quadspi_bk1_io1_pc10
                &quadspi_bk1_io2_pe2 &quadspi_bk1_io3_pd13>;
    flash-id = <1>;
    status = "okay";

	mx25r512: qspi-nor-flash@0 {
		compatible = "st,stm32-qspi-nor";
        reg = <0>;
        qspi-max-frequency = <80000000>;
        size = <DT_SIZE_M(512)>;
        reset-gpios = <&gpiod 13 GPIO_ACTIVE_LOW>;
        reset-gpios-duration = <10>;
        spi-bus-width = <4>;
        writeoc = "PP_1_4_4";
        status = "okay";
		
	};
};

/ {
	chosen {
		zephyr,flash-controller = &mx25r512;
	};
};
```

Shell commands based on sample page:
```bash
[00:00:00.010,000] <inf> flash_stm32_qspi: qspi-nor-flash@0: SFDP v 1.6 AP ff with 3 PH
[00:00:00.010,000] <inf> flash_stm32_qspi: qspi-nor-flash@0: SFDP v 1.6 AP ff with 3 PH
[00:00:00.010,000] <inf> flash_stm32_qspi: PH0: ff00 rev 1.6: 16 DW @ 30
[00:00:00.010,000] <inf> flash_stm32_qspi: qspi-nor-flash@0: 64 MiBy flash
[00:00:00.010,000] <inf> flash_stm32_qspi: Flash - address mode: 4B
[00:00:00.010,000] <inf> flash_stm32_qspi: Quad read mode 4 instr [0x6b] supported
[00:00:00.010,000] <inf> flash_stm32_qspi: Quad read mode 7 instr [0xeb] supported
[00:00:00.010,000] <inf> flash_stm32_qspi: Quad read mode 7 instr [0xeb] will be used
[00:00:00.010,000] <inf> flash_stm32_qspi: QE requirement mode: 2
[00:00:00.010,000] <inf> flash_stm32_qspi: Quad mode enabled
[00:00:00.010,000] <inf> flash_stm32_qspi: PH1: ffc2 rev 1.0: 4 DW @ 110
*** Booting Zephyr OS build 15999a4b0fac ***
Found flash controller qspi-nor-flash@0.
Flash I/O commands can be run.
uart:~$ flash_sample page_count
Flash device contains 16384 pages.
uart:~$ flash_sample page_erase 16383
Erasing page 16383 (start offset 0x3fff000, size 0x1000)
uart:~$ flash_sample page_write 16383 8 17 19 28 39
Reading back written bytes:
11 13 14 27 | 
uart:~$ flash_sample page_write 16383 4 77 9 1 2
Reading back written bytes:
45 01 01 02 | 
uart:~$ flash_sample page_read 16383 4 12
45 01 01 02 | 77 77 77 77 | 77 77 77 77 | 
uart:~$ flash_sample page_read 16383 0 16
11 13 14 27 | 45 01 01 02 | 77 77 77 77 | 77 77 77 77
uart:~$ 
```

As we can see, the read value do not match with those we actually write.

**Expected behavior**
Same as shown on the sample page:
```bash
uart:~$ flash_sample page_count
Flash device contains 16384 pages.
uart:~$ flash_sample page_erase 16383
Erasing page 16383 (start offset 0x3fff000, size 0x1000)
uart:~$ flash_sample page_write 16383 8 17 19 28 39
Reading back written bytes:
11 13 1c 27
uart:~$ flash_sample page_write 16383 4 77 9 1 2
Reading back written bytes:
4d 09 01 02
uart:~$ flash_sample page_read 16383 4 12
4d 09 01 02 | 11 13 1c 27 | ff ff ff ff
uart:~$ flash_sample page_read 16383 0 16
ff ff ff ff | 4d 09 01 02 | 11 13 1c 27 | ff ff ff ff
uart:~$ 
```

**Impact**
Interacting with the NOR-Flash does not work properly. This also implies that file systems on top of the NOR-Flash do not work properly.

**Environment (please complete the following information):**
 - OS: Linux Arch
 - Toolchain SDK
 - Version: 15999a4b0fac

FRASTM, 2023-03-03:
Testing quadspi with a stm32l496 disco kit instead (no stm32f723e_disco available)
I set the DTS as follows:
```
&quadspi {
	pinctrl-0 = <&quadspi_bk1_io0_pb1 &quadspi_bk1_io1_pb0
			&quadspi_bk1_io2_pa7 &quadspi_bk1_io3_pa6
			&quadspi_bk1_ncs_pb11 &quadspi_clk_pa3>;
	pinctrl-names = "default";
	flash-id = <1>;
	status = "okay";
	
	mx25r6435: qspi-nor-flash@0 {
		compatible = "st,stm32-qspi-nor";
		reg = <0>;
		qspi-max-frequency = <8000000>;
		size = <DT_SIZE_M(64)>; /* 8 MBytes */
		status = "okay";
		label = "QSPI-NOR";
		spi-bus-width = <4>;
		writeoc = "PP_1_4_4";
		jedec-id = [c2 28 17];
		sfdp-bfp = [
			e5 20 f1 ff  ff ff ff 03  44 eb 08 6b  08 3b 04 bb
			ee ff ff ff  ff ff 00 ff  ff ff 00 ff  0c 20 0f 52
			10 d8 00 ff  23 72 f5 00  82 ed 04 cc  44 83 68 44
			30 b0 30 b0  f7 c4 d5 5c  00 be 29 ff  f0 d0 ff ff
		];

	};
};
```

I run the sample  : west build -p auto -b stm32l496g_disco samples/drivers/flash_shell/

```
uart:~$ flash_sample set_device QSPI-NOR                                        
Leaving behind device QSPI-NOR                                                  
uart:~$ flash_sample page_erase 2036                                            
Erasing page 2036 (start offset 0x7f4000, size 0x1000)                          
uart:~$ flash_sample page_read 2036 0x10                                        
ff ff ff ff | ff ff ff ff | ff ff ff ff | ff ff ff ff                           
uart:~$ flash_sample page_write 2036 1 2 3 4 5 6 7 8 9 0xa 0xb 0xc 0xd 0xe 0xf  
Reading back written bytes:                                                     
02 03 04 05 | 06 07 08 09 | 0a 0b 0c 0d | 0e 0f    
```

FRASTM, 2023-03-06:
or
```
uart:~$ flash_sample write_pattern 0x000fe000 100                                         
Reading back written bytes:                                                               
00 01 02 03 | 04 05 06 07 | 08 09 0a 0b | 0c 0d 0e 0f                                     
10 11 12 13 | 14 15 16 17 | 18 19 1a 1b | 1c 1d 1e 1f                                     
20 21 22 23 | 24 25 26 27 | 28 29 2a 2b | 2c 2d 2e 2f                                     
30 31 32 33 | 34 35 36 37 | 38 39 3a 3b | 3c 3d 3e 3f                                     
40 41 42 43 | 44 45 46 47 | 48 49 4a 4b | 4c 4d 4e 4f                                     
50 51 52 53 | 54 55 56 57 | 58 59 5a 5b | 5c 5d 5e 5f                                     
60 61 62 63 |                                                                             

```

whati001, 2023-03-06:
Hi @FRASTM,
thank you very much for your instant support  and please apologize my late reply.
You are right, the NOR-Flash works perfectly fine with our devicetree file. If I have not misunderstood the driver, there is no need to specify the `jedec-id` and `sfdp-bfp` because it pulls the values at runtime during the `flash_stm32_qspi_init` function is running. Is this correct?

However, it seems like the reset configuration in the device tree breaks the functionality. So after removing the two lines below, the sample works perfect for me.
````
reset-gpios = <&gpiod 13 GPIO_ACTIVE_LOW>;
reset-gpios-duration = <10>;
```

So here are the final device tree construct for the QSPI-Flash device from the stm32f723e_disco.
```
// QSPI Flash Controller
&quadspi {
	pinctrl-names = "default";
	pinctrl-0 = <&quadspi_clk_pb2 &quadspi_bk1_ncs_pb6
                &quadspi_bk1_io0_pc9 &quadspi_bk1_io1_pc10
                &quadspi_bk1_io2_pe2 &quadspi_bk1_io3_pd13>;
    flash-id = <1>;
    status = "okay";

	mx25r512: qspi-nor-flash@0 {
		compatible = "st,stm32-qspi-nor";
		reg = <0>;
		qspi-max-frequency = <8000000>;
		size = <DT_SIZE_M(512)>; /* 64 MBytes */
		status = "okay";
		spi-bus-width = <4>;
		writeoc = "PP_1_4_4";
	};
};

// Use the QSPI driver as flash
/ {
    chosen {
		zephyr,flash-controller = &mx25r512;
	};
};
```

whati001, 2023-03-06:
Should I open a PR to add the QSPI and flash definition to the board device tree?

whati001, 2023-03-08:
Ok, I will create a PR for the device tree, however I am currently blocked and I will need some time.

FRASTM, 2023-03-09:
> Ok, I will create a PR for the device tree, however I am currently blocked and I will need some time.

I can proceed, if you accept. It's just updating the DTS of the board.

whati001, 2023-03-15:
> > Ok, I will create a PR for the device tree, however I am currently blocked and I will need some time.
> 
> I can proceed, if you accept. It's just updating the DTS of the board.

Sure, please go ahead

closed 2023-03-29 by carlescufi