zephyrproject-rtos/zephyr #55399: samples/kernel/metairq_dispatch fails on stm32l475 disco kit

FRASTM, 2023-03-03:
**Describe the bug**
Running the samples/kernel/metairq_dispatch/  on the disco_l475_iot1 board will fail

Please also mention any information which could help others to understand
the problem you're facing:

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b disco_l475_iot1 samples/kernel/metairq_dispatch/
2. west flash
3. See error

**Expected behavior**
```
I:         ---------- Latency (cyc) ----------
I:             Best    Worst     Mean    Stdev
I: MetaIRQ     3421     3538     3428       17
I: Thread0  4893169  5462699  5173453   233645
I: Thread1     5948  3375853  1646489  1034312
I: Thread2  7694920 10304480  8821696   833991
I: Thread3 12227306 14814442 13461505   812224
I: MetaIRQ Test Complete

```

**Impact**
CI failure

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.3.0-686-g1be74ac6d71a ***
I: Starting Thread0 at priority -2
I: Starting Thread1 at priority -1
II: M0 T1 mirq 3494 disp 6004 proc 19970 real 21183
I: M4 T1 mirq 3375 disp 254786 proc 4040 real 5228
I: M16 T1 mirq 3376 disp 118743 proc 14605 real 15044
I: M24 T1 mirq 3376 disp 150333 proc 1I: Thread 2ERROR: Queue full, event dropped!
 qERROR: Queue full, event dropped!
uERROR: Queue full, event dropped!
eue full, message 35 dropped
2258 real 12590
I: M25 T1 mirq 3392 disp 854626 proc 37153 real 38357
I: M27 T1 mirq 3376 disp 1198853 proc 29622 real 29768
I: M31 T1 mirq 3377 disp 1413430 proc 21783 real 22406
I: M33 T1 mirq 3376 disp 1724646 proc 27235 real 28541
I: M5 T0 mirq 3391 disp 3338574 proc 18599 real 18729
I: M6 T0 mirq 3376 disp 3719168 proc 33746 real 34676
I: M10 T0 mirq 3375 disp 3955714 proc 5287 real 6455
I: M18 T0 mirq 3375 disp 3954473 proc 11746 real 12590
I: M23 T0 mirq 3376 disp 4135874 proc 28466 real 29768
I: M28 T0 mirq 3375 disp 4350455 proc 23144 real 23633
I: M29 T0 mirq 3376 disp 4694897 proc 17872 real 18725
: Starting Thread2 at priority 0
I: M2 T2 mirq 3391 disp 6422335 proc 28026 real 28541
I: M7 T2 mirq 3376 disp 6692744 proc 10293 real 11363
I: M8 T2 mirq 3375 disp 7050449 proc 30304 real 30995
I: M9 T2 mirq 3375 disp 7395306 proc 9238 real 10136
I: M11 T2 mirq 3378 disp 7682070 proc 20314 real 21179
I: M12 T2 mirq 3376 disp 8016053 proc 9197 real 10136
I: M13 T2 mirq 3376 disp 8357760 proc 32879 real 33449
I: M15 T2 mirq 3376 disp 8656019 proc 24534 real 24860
I: M17 T2 mirq 3376 disp 8969687 proc 33582 real 34676
I: M20 T2 mirq 3375 disp 9245175 proc 17466 real 18725
I: M21 T2 mirq 3375 disp 9600709 proc 11310 real 12590
I: M22 T2 mirq 3375 disp 9934107 proc 13500 real 13817
I: M26 T2 mirq 3375 disp 10196731 proc 34660 real 35903
I: M30 T2 mirq 3376 disp 10424381 proc 38141 real 38357
I: M32 T2 mirq 3376 disp 10726484 proc 14399 real 15044
I: M34 T2 mirq 3392 disp 11085274 proc 21927 real 22406
I: Starting Thread3 at priority 1
I: M1 T3 mirq 3375 disp 13070273 proc 27671 real 28541
I: M3 T3 mirq 3375 disp 13443624 proc 35337 real 35903
I: M14 T3 mirq 3376 disp 13408337 proc 19169 real 19952
I: M19 T3 mirq 3376 disp 13580036 proc 7157 real 7682
I: M36 T3 mirq 336619 disp 13297736 proc 31037 real 32222
I: M37 T3 mirq 313866 disp 13703627 proc 36574 real 37130
```

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain Zephyr SDK 0.15.2
 - Commit SHA  1be74ac6d71af062c3f8cee57a17423d4288548e

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2023-03-03:
possible WA: 
differ the __ASSERT_NO_MSG(ret == 0); of the k_msgq_get(&threads[id].msgq, &m, K_FOREVER); in the samples/kernel/metairq_dispatch/src/main.c

```
diff --git a/samples/kernel/metairq_dispatch/src/main.c b/samples/kernel/metairq_dispatch/src/main.c
index a767cb8d76..b388a77b46 100644
--- a/samples/kernel/metairq_dispatch/src/main.c
+++ b/samples/kernel/metairq_dispatch/src/main.c
@@ -188,8 +188,6 @@ static void thread_fn(void *p1, void *p2, void *p3)
 		int ret = k_msgq_get(&threads[id].msgq, &m, K_FOREVER);
 		uint32_t start = k_cycle_get_32();
 
-		__ASSERT_NO_MSG(ret == 0);
-
 		/* Spin on the CPU for the requested number of cycles
 		 * doing the "work" required to "process" the event.
 		 * Note the inner loop: hammering on k_cycle_get_32()
@@ -202,6 +200,7 @@ static void thread_fn(void *p1, void *p2, void *p3)
 			for (volatile int i = 0; i < 100; i++) {
 			}
 		}
+		__ASSERT_NO_MSG(ret == 0);
 
 		uint32_t dur = k_cycle_get_32() - start;
```

erwango, 2023-03-07:
@fabiobaltieri Can we have someone from kernel to look at proposed workaround ?

erwango, 2023-03-07:
Note: Not all issues encountered on STM32 are STM32 issues. It would be nice not to assigned them all to me by default. Specifically in current case where a pre-analysis shows link to other subsystem

fabiobaltieri, 2023-03-07:
Hi @erwango, sorry we triaged this with a bunch of other bugs and triaged based on the fact that it's unlikely that the kernel maintainers would have a board to test. Since there's a proposed workaround, @FRASTM how about you send that as a PR and then it can be discussed there?

closed 2023-03-09 by fabiobaltieri