zephyrproject-rtos/zephyr #55404: L2CAP data gets truncated if MTU is larger than 245 bytes

hermabe, 2023-03-03:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/55380

<div type='discussions-op-text'>

<sup>Originally posted by **fbzhomi** March  2, 2023</sup>
Hi Folks,
Currently I am running on Zephyr 3.2 at a vendor specific platform and create an application to run BLE L2CAP stress TX test. I found an issue that when the applications tries to send a L2CAP SDU data with 2000 bytes which is beyond 245 bytes(a full L2CAP complete packet size) data via zephyr API bt_l2cap_chan_send call every 15ms, device has a chance to send a corrupted L2CAP data over the air. The symptom is like whole trunk of L2CAP start traffic is missing, and L2CAP continue packet becomes the first L2CAP packet to the air and when remote device received the corrupted L2CAP data, it wrongly consider L2CAP continue to be L2CAP start and try to parse L2CAP SDU length field and complains about invalid SDU length, then force disconnect the L2CAP channel.

My current prj.conf for L2CAP setting are
CONFIG_BT_BUF_ACL_RX_SIZE=251
CONFIG_BT_L2CAP_TX_MTU=2000
CONFIG_BT_BUF_ACL_TX_SIZE=251
CONFIG_BT_CTLR_DATA_LENGTH_MAX=251
CONFIG_BT_BUF_ACL_TX_COUNT=255

I can reproduce this issue in two platforms using Zephyr 2.3 and Zephyr 3.2 so it seems like a bluetooth host issue in subsys of Zephyr.
I was wondering that if Zephyr stack has been running L2CAP stress test per each version upstreams. I check the L2CAP shell zephyr/subsys/bluetooth/shell/l2cap.c only support 230 bytes L2CAP SDU and only support single TX via CLI instead of TX stress test. So it might be a mis-coverage.
</div>

hermabe, 2023-03-03:
@fbzhomi Are you able to push your stress test somewhere so that we can use it to reproduce the issue?

jori-nordic, 2023-03-06:
We have downstream tests with 1230-byte SDUs: https://github.com/jori-nordic/zephyr/blob/66d9ddd9dc587f299c530a5aea307f1b68075e59/tests/bluetooth/bsim/host/l2cap/stress/src/main.c#L1
We also have the same in hardware downstream. This could still be a corner-case that is missing coverage. 

@fbzhomi So knowing how exactly you call the l2cap APIs will help us (ie your stress-test app sources):
If you are using nordic hardware and don't want to share the sources publicly, you can open a private devzone ticket and mention my name so I get added to it (Jonathan Rico).

fbzhomi, 2023-03-06:
Hi 
Thanks for the reply.
I am sorry that I cannot share a complete code logic for you due to the company IP policy. I am running host only build using zephyr BT stack 3.2 and use a vendor controller build. I believe you can use nRF5340 DK to test it since I used to reproduce this issue using zephyr 2.3 one year ago and started to hit this issue.

My l2cap application will be similar like l2cap shell implementation in zephyr.
I have global tx pool for L2CAP tx purpose.
NET_BUF_POOL_FIXED_DEFINE(data_tx_pool, L2CAP_TX_BUFF_NUM,
			  BT_L2CAP_SDU_BUF_SIZE(BSL_MAX_TX_BUFF_SIZE), 8, NULL);

My core l2cap send os like 
sEffectiveTxMTU = 491;

`int l2cap_send(uint8_t *buf_data, uint16_t len) {
	int ret;
	if (len == 0) {
		BSL_LOG("0 length buf_data to send!");
		return -1;
	}
	uint16_t l = 0; // length to be transmit in one SDU
	int count = (len - 1) / sEffectiveTxMTU + 1;
  struct net_buf *buf;
	if (l2ch_chan_server.ch.tx.credits < count) {
		return -ENOEXEC;
	}
	while (count--) {
		buf = net_buf_alloc(&data_tx_pool, K_NO_WAIT);
		if (!buf) {
			BSL_LOG("no buffers in data_tx_pool");
			return -1;
		}
		net_buf_reserve(buf, BT_L2CAP_SDU_CHAN_SEND_RESERVE);
		l = MIN(sEffectiveTxMTU, len);
		net_buf_add_mem(buf, buf_data, l);
		len -= l;
		ret = bt_l2cap_chan_send(&l2ch_chan_server.ch.chan, buf);
		if (ret < 0) {
			BSL_LOG("unable to send: %d", -ret);
			net_buf_unref(buf);
			return -ENOEXEC;
		}
	}
	return 0;
}`
And I am calling this function with a buffer size 491 bytes every 15ms and LE connection interval is 15ms as well. I enable 2m phy and data length extension feature to 251 TX octets to maximize the TX throughput.

jori-nordic, 2023-03-07:
Yep, I was working on that yesterday, already found two bugs in l2cap. Don't know yet if your problems will be fixed by those fixes, but I'll probably have a PR you can try in the next days.

jori-nordic, 2023-03-07:
@fbzhomi could you try with the changes in https://github.com/zephyrproject-rtos/zephyr/pull/55519 ?

fbzhomi, 2023-03-07:
> ZjQcmQRYFpfptBannerStart
> This Message Is From an External Sender
>
> ZjQcmQRYFpfptBannerEnd

Hi Jonathan,
What is the fix you want me to try? Is there some fix in L2CAP TX side?

jori-nordic, 2023-03-08:
Yes there are changes in both TX and RX sides, you need to apply all the commits with the message "Bluetooth: host: ..." from here https://github.com/zephyrproject-rtos/zephyr/pull/55519/commits.
You can do that by doing `git fetch pull/55519/head` and then `git rebase -i FETCH_HEAD`.

fbzhomi, 2023-03-08:
Hi Jonathan,
Thanks. In our end, we have narrowed down the issue seems to happen below the Zephyr BT host stack and it could be some where in hci driver or even controller build. Because these two areas are both customized implementation. I think your fix won't help solve my problem.
I will continue looking into this issue and will keep you posted if we confirm there is an issue in Zephyr BT host stack.
Thanks.

jori-nordic, 2023-03-09:
Sounds good. Thanks for the bug report though, it prompted me to find and fix more issues in l2cap :)

jori-nordic, 2023-04-11:
Closing as invalid, please re-open if you find out the bug is indeed coming from the host.