zephyrproject-rtos/zephyr #55405: Remove the use of off_t for offsets

Laczen, 2023-03-03:
Several drivers (e.g. flash_disk, eeprom and flash) are using the `off_t` type for specifying addresses. `off_t` however is a signed type, which does not correspond to addresses. This results in unnecessary checks (verify that the address is positive) and is confusing.

A simple modification would be to replace `off_t` with `unsigned long`, this would result in a `uint32_t` on 32-bit and `uint64_t` on 64 bit architectures.

**Describe alternatives you've considered**
Leave it as is and make sure all drivers implement a check for positive addresses.

mniestroj, 2023-03-03:
Should we consider `size_t` instead of `unsigned long`? Offset is usually paired with size/length, which is `size_t`. There are often calculations including both offset and size. If both would be the same type, then they should further simplify implementation.

Laczen, 2023-03-03:
> Should we consider `size_t` instead of `unsigned long`? Offset is usually paired with size/length, which is `size_t`. There are often calculations including both offset and size. If both would be the same type, then they should further simplify implementation.

Using `size_t` is confusing as this is meant to be a length/size and not an offset/address.

nordicjm, 2023-03-03:
uint*_t gets my vote, but this would need to be checked because there might be a driver that actually allows and uses negative offsets

Laczen, 2023-03-03:
@nordicjm, yes this needs to be checked.

The disadvantage of using a unsigned offset is that it would not allow creating a driver that keeps its reading/writing location internally and where the offset has the meaning "from the current" read/write location.

mniestroj, 2023-03-03:
> I also don't see the further simplification of the implementation.

@Laczen I just meant narrowing conversion when going from one type to the other or vice versa. But maybe it isn't a problem, just pointing it out for consideration.

de-nordic, 2023-03-07:
> Using `size_t` is confusing as this is meant to be a length/size and not an offset/address. I also don't see the further simplification of the implementation.

@Laczen is correct here, `size_t` is only supposed to cover array size the architecture can index directly, so on 32bit devices this will be 32bit, on 64bit device 64 and so on.

The `off_t` (https://pubs.opengroup.org/onlinepubs/9699919799/basedefs/sys_types.h.html) has been used for file size, although I think that it has been defined for use with `fseek`
`off_t`  may be defined as `long` or `int` , and there is separate `off64_t`, `long long`,  to cover 64bit values.
Zephyr works with systems that are at least 32bit so `long`/`int` is not a problem - same thing.
`fseek` and `ftell` use `long` anyway.

Drawback of `off_t` that it covers half of 32bit range mostly because negative offsets  rarely make sense:
 - they make sense in fseek, when `SEEK_CUR` is used;
 - they make sense in ftell (to return errno);
 - they do not make sense when devices are addressed as devices usually use restrict access to some area, often in range [0, MAX_DEV_RANGE -1], and map the area to [0, MAX_AREA -1], and do not allow indexing beyond that range; which means that negative offset makes no sense;
 - in file systems, except for seeking, access is done rather by addressing blocks/sectors directly and negative offsets are rarely used in anything.

Using `size_t` limits storage offsets to architecture set length of `long` and may not fulfill requirements of peripherals: for example using 8GiB storage (flash, SD) will be limited on 32bit device to first 4GiB, as it will not be even possible to get the real size of the device. So in case when 32bit device is expected to access larger devices we would need to make APIs work, less efficiently, with 64bit sizes and offsets.

We have at least these options available
 - settle on highest possible uintx_t, which is reasonable, so probably `uint64_t` and accept that API calls will have to take extra cost of pushing 64bit values on 32bit devices;
 - create storage specific type for size/offset which will be independently set from `size_t`/`off_t`, by user, to support required size of devices; always minimal size would be equal to `size_t` or architecture native `uint`; of course when size bigger than native for architecture would be selected, then all APIs using the type will uniformly take efficiency hit across all devices;
 - have two types of calls supporting short and long offsets; this option is less convenient in usage

We probably should

stephanosio, 2023-03-07:
My suggestion would be to introduce a new Zephyr-specific offset type that is:

* unsigned
* configurable via some kind of Kconfig to be 32-bit or 64-bit

Relying on the POSIX `off_t` has been quite controversial and problematic because it is currently defined by the libc and therefore not fully under Zephyr's control.

Laczen, 2023-03-07:
> My suggestion would be to introduce a new Zephyr-specific offset type that is:
>
> * unsigned
> * configurable via some kind of Kconfig to be 32-bit or 64-bit
>
> Relying on the POSIX `off_t` has been quite [...]

I agree to define some zephyr specific offset type: Why not
`typedef address_t unsigned long`

henrikbrixandersen, 2023-03-07:
FWIW, the Linux kernel uses `unsigned int` for address offset in e.g. EEPROM drivers.

de-nordic, 2023-03-07:
> FWIW, the Linux kernel uses `unsigned int` for address offset in e.g. EEPROM drivers.

FreeBSD uses `off_t` for QSPI devices: https://svnweb.freebsd.org/base/head/sys/dev/flash/qspi_if.m

Approaches differ. Question is how does custom type, if we choose one, will affect porting other software to Zephyr.

de-nordic, 2023-03-07:
> My suggestion would be to introduce a new Zephyr-specific offset type that is:
>
>     * unsigned
>
>     * configurable via some kind of Kconfig to be 32-bit or 64-bit
>
>
> Relying on the POSIX `off_t` has [...]

We can go with single unsigned type, but this will require us to adapt `fs_seek` by adding direction flags. Currently you are able to set offset relatively from selected origin using negative and positive values of offsets.
I have nothing against that, but we will differ from other systems But does it matter that much? Win32API differs and people use it (do they still use it?)

carlescufi, 2023-03-07:
Architecture WG:

- @Laczen mentions that Linux uses `off_t` or `long off_t` but there everything is a file, so it's different in Zephyr. A device that keeps its position and increments its position by itself then it makes sense to have negative values
- @henrikbrixandersen asks if the only issue is that `off_t` needs additional size checks. @Laczen says that EEPROM and flash are using `off_t` but memory retained uses `size_t`
- We would like to standardize to a type for offsets/addresses. `off_t` and `size_t` don't really make sense for this case
- @stephanosio [proposes a solution](https://github.com/zephyrproject-rtos/zephyr/issues/55405#issuecomment-1458308204)
- @Laczen mentions that the proposal applies only to addresses, not things like files

Laczen, 2023-03-13:
Maybe a simple solution could be:
a. We standardize on unsigned long (being uint32_t or uint64_t),
b. We add a check in the driver based upon the address register from the dts (that tells the required size),
c. For systems that would need support for uint64_t while working on a 32 bit platform (support for very large devices) a modified driver interface could be added (based on the address register layout) that provides unsigned long long as parameters.

(Both for offset and size this could be applied)

closed 2023-04-27 by Laczen