zephyrproject-rtos/zephyr #55412: TrustZone scheduler behavior isn't MetaIRQ-aware

andyross, 2023-03-03:
It was discovered in #55230 that the TrustZone entry code is using k_sched_lock() to prevent other OS threads from running.

This doesn't work: what that does is elevate the thread to cooperative priority to prevent premption by other preemptive or cooperative threads.  But there is a higher class of priorities in Zephyr, MetaIRQ, which are be design able to preempt cooperative threads.  Making a MetaIRQ thread runnable during TrustZone processing will cause the scheduler to try to context switch and cause a panic.  (Though I'm not TZ expert).

What the code needs to do instead is elevate the thread priority to the highest available priority, e.g. something like:

```c
static int saved_prio;

static inline tz_enter_lock(void)
{
    saved_prio = k_thread_priority_get(k_current_get());
    k_thread_priority_set(k_current_get(), K_HIGHEST_THREAD_PRIO);
}

static inline tz_exit_unlock(void)
{
    k_thread_priority_set(k_current_get(), saved_prio);
}
```

As an optimization, we could fall back to k_sched_un/lock() (which uses a cute trick to be extremely small/fast) if CONFIG_NUM_METAIRQ_PRIORITIES == 0.

carlescufi, 2023-03-03:
@joerchan  FYI

microbuilder, 2023-03-03:
Thanks for the detailed notes, @andyross ... I'll dig into this Monday 👍

oyvindronningstad, 2023-03-07:
@bartagcx (Followup from https://github.com/zephyrproject-rtos/zephyr/issues/55230#issuecomment-1458090119) The reason why I'm asking is that TF-M at least should handle interruption by unrelated threads and IRQs. If you are seeing a crash we should ideally document or investigate the details.

bartagcx, 2023-03-07:
As explained, I'm not working with the secure calls directly. I'm writing a completely unrelated driver. The system is using sdk-nrf version 2.0.2, sorry a little bit old, I know.

I ran into this issue just by chance. Another completely unrelated application in my system is periodically requesting random numbers every few minutes. It uses spm_request_random_number(). That is using

```
void before_nse(void)
{
	k_sched_lock();
	z_arm_save_fp_context(&context_buffer);
}

void after_nse(void)
{
	z_arm_restore_fp_context(&context_buffer);
	k_sched_unlock();
}
```

This turned out not to be sufficient to prevent a crash. The actual secure call takes about 230 µs. In my setup there is a high chance, that during this time a GPIO interrupt triggers. That itself is not causing any issue. But the interrupt handler wakes up a meta-IRQ thread. As k_sched_lock() intentionally doesn't lock out meta-IRQ threads, scheduling happens, while the secure call is ongoing and the system crashes after returning from secure call. That exactly matches the description at tfm_ns_interface_dispatch() and TZ_THREAD_SAFE_NONSECURE_ENTRY_FUNC().

As I was considering this a generic issue of k_sched_lock(), I have reported this bug against Zephyr and not sdk-nrf, which actually doesn't contain the specific code any more.

If TF-M supports interruptions by unrelated threads (that would be really great), do we need to protect the call at all with the priority increase?

oyvindronningstad, 2023-03-08:
> As explained, I'm not working with the secure calls directly. I'm writing a completely unrelated driver. The system is using sdk-nrf version 2.0.2
> ...
> This turned out not to be sufficient to prevent a crash.

Thanks for the info!

> As I was considering this a generic issue of k_sched_lock(), I have reported this bug against Zephyr and not sdk-nrf, which actually doesn't contain the specific code any more.

Feel free to open a support request with Nordic if you wish. It will help us track it, we've had other related reports recently.

> If TF-M supports interruptions by unrelated threads (that would be really great), do we need to protect the call at all with the priority increase?

Exactly. Trying to find out if this is necessary.

PR #55441 (2023-03-08): modules: tf-m: Fix thread preemption toggling
Fixes an issue where thread pre-emption was not being disabled correctly, failing to taking into account MetaIRQ, which can be used to pre-empt any cooperative thread.

The updated code sets the current thread to `K_HIGHEST_THREAD_PRIO` before calling the secure function, and restores the thread priority level once secure execution has terminated, allowing the thread to be pre-empted once again.

Fixes #55412
closed 2023-03-08 by carlescufi