zephyrproject-rtos/zephyr #55428: CONFIG_THREAD_ANALYZER_AUTO crash in native_posix

thucon, 2023-03-04:
I have made a very simple demo application based on the `samples/basic/threads/src/main.c`, where I just send simple text messages via a queue between a couple of threads. Nothing fancy here (full source is shown in the end of this issue report)

I then want to check stack and cpu usage by using the builtin [thread analyzer](https://docs.zephyrproject.org/latest/services/debugging/thread-analyzer.html).

The analyzer support to automatically create an "analyzer thread" via the following option `CONFIG_THREAD_ANALYZER_AUTO`.

However if I enable this option the application crash.
It specifically crash with a zero-division in `zephyr/subsys/debug/thread_analyzer.c` in the function `static void thread_analyze_cb(const struct k_thread *cthread, void *user_data)` in the snippet below - more precisely in the `info.utilization = (info.usage.execution_cycles * 100U) / rt_stats_all.execution_cycles;` calculation.

Because `rt_stats_all.execution_cycles` is zero it fails.

```c
#ifdef CONFIG_THREAD_RUNTIME_STATS
	ret = 0;

	if (k_thread_runtime_stats_get(thread, &info.usage) != 0) {
		ret++;
	}

	if (k_thread_runtime_stats_all_get(&rt_stats_all) != 0) {
		ret++;
	}
	if (ret == 0) {
		info.utilization = (info.usage.execution_cycles * 100U) /
			rt_stats_all.execution_cycles;
	}
#endif
```

My assumption is that it is because no threads have executed yet (either because the `CONFIG_THREAD_ANALYZER_AUTO` thread has higher priority or doesn't wait until somethings have executed). In general the best approch might be to check if `rt_stats_all.execution_cycles` is zero or not as part of the `if (ret == 0) {` statement.

Of course you could disable the calculation by disable the `CONFIG_THREAD_RUNTIME_STATS` option, but this is part of the reason why you do thread analysis... and it is automatically enabled when `CONFIG_THREAD_ANALYZER=y`. 

The output before the crash is:

     [00:00:00.000,000] <inf> thread_analyzer: Thread analyze:

If I implement the analyzer myself and make sure all other threads have executed, I can implement the analyzer with the code shown below

```c
void analyzer_out(void)
{
	while (1) {
		k_msleep(5000);
		thread_analyzer_print();
	}
}
```

To verify / test this bug add the following configs in `prj.conf`:

```
CONFIG_THREAD_ANALYZER=y
CONFIG_THREAD_ANALYZER_USE_PRINTK=y
CONFIG_THREAD_ANALYZER_USE_LOG=y
CONFIG_THREAD_ANALYZER_AUTO=y
CONFIG_THREAD_ANALYZER_AUTO_INTERVAL=5
CONFIG_THREAD_ANALYZER_AUTO_STACK_SIZE=1048
CONFIG_THREAD_NAME=y
CONFIG_THREAD_RUNTIME_STATS=y
```

Next do create a `src/main.c` with the following content:

```c
#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/debug/thread_analyzer.h>
#include <zephyr/drivers/gpio.h>
#include <zephyr/sys/printk.h>
#include <zephyr/sys/__assert.h>
#include <string.h>

/* size of stack area used by each thread */
#define STACKSIZE 1024

/* scheduling priority used by each thread */
#define PRIORITY 7

struct printk_data_t {
	void *fifo_reserved; /* 1st word reserved for use by fifo */
	uint32_t led;
	uint32_t cnt;
};

K_FIFO_DEFINE(printk_fifo);

void blink(uint32_t led, uint32_t sleep_ms)
{
	const size_t size = sizeof(struct printk_data_t);
	int cnt = 0;
	int gpio_state = 0;

	while (1) {
		gpio_state = cnt % 2;

		struct printk_data_t tx_data = { .led = led, .cnt = cnt };

		char *mem_ptr = k_malloc(size);
		__ASSERT_NO_MSG(mem_ptr != 0);

		memcpy(mem_ptr, &tx_data, size);

		k_fifo_put(&printk_fifo, mem_ptr);

		k_msleep(sleep_ms);
		cnt++;
	}
}

void blink0(void)
{
	blink(1, 100);
}

void blink1(void)
{
	blink(2, 1000);
}

void uart_out(void)
{
	while (1) {
		struct printk_data_t *rx_data = k_fifo_get(&printk_fifo,
							   K_FOREVER);
		printk("Toggled led%d; counter=%d\n",
		       rx_data->led, rx_data->cnt);
		k_free(rx_data);
	}
}

K_THREAD_DEFINE(blink0_id, STACKSIZE, blink0, NULL, NULL, NULL,
		PRIORITY, 0, 0);
K_THREAD_DEFINE(blink1_id, STACKSIZE, blink1, NULL, NULL, NULL,
		PRIORITY, 0, 0);
K_THREAD_DEFINE(uart_out_id, STACKSIZE, uart_out, NULL, NULL, NULL,
		PRIORITY, 0, 0);
```

The zephyr version (commit hash) is: bc5fbc929bb805e3c562c7280dd02a6f9f1f9087

aescolar, 2023-03-04:
Hi @thucon ,
I don't expect the thread analyzer to provide you any usable information in native_posix. In the POSIX architecture :
* Code takes 0 simulated time to execute. So the analyzer will report 0 cycles being used. (I presume the division by zero you see is the result of this)
* The stack allocated by Zephyr is not actually used (except for a tiny part used by the arch code itself to do a bit of thread bookkeeping). The POSIX architecture uses a separate stack (from an underlying Linux pthread) which Zephyr is blind about. So the thread analyzer is going to only report a tiny stack utilization.

Maybe the thread analyzer kconfig option should have a depends on !POSIX_ARCH to avoid people trying to use it in native_posix.
CC @joerchan

thucon, 2023-03-04:
Hi @aescolar 

First of all thank you very much for a very quick and concise answer.

What you write makes sense, and the idea of making a compiler flag to avoid using Thread Analyzer for native builds might be the best way to deal with it.

PR #55492 (2023-03-07): Thread analyzer: Prevent on POSIX ARCH
The thread analyzer cannot be really used with the POSIX architecture.

As:
* Code takes 0 simulated time to execute. So the analyzer will report 0 cycles being used. (and result in a division by zero exception)
* The stack allocated by Zephyr is not actually used (except for a tiny part used by the arch code itself to do a bit of thread bookkeeping). The POSIX architecture uses a separate stack (from an underlying Linux pthread) which Zephyr is blind about. So the thread analyzer is going to only report a tiny stack utilization.

Prevent users from selec [... cut]
closed 2023-03-07 by nashif