zephyrproject-rtos/zephyr #55431: Bluetooth: Disabling `BT_DEBUG_LOG` doesn't disable logs

JordanYates, 2023-03-05:
**Describe the bug**
When `CONFIG_BT_DEBUG_LOG` is not enabled, the log level (`CONFIG_BT_HCI_DRIVER_LOG_LEVEL`) of Bluetooth drivers can actually increase.

Disabling `CONFIG_BT_DEBUG_LOG` results in `BT_DEBUG` not being defined, resulting in `BT_HCI_DRIVER` not being defined. https://github.com/zephyrproject-rtos/zephyr/blob/bc5fbc929bb805e3c562c7280dd02a6f9f1f9087/subsys/bluetooth/common/Kconfig#L382-L385

This logging level is used in quite a few locations, for example:
https://github.com/zephyrproject-rtos/zephyr/blob/bc5fbc929bb805e3c562c7280dd02a6f9f1f9087/subsys/bluetooth/controller/hci/hci.c#L97-L99

Because of the behaviour of the logging subsystem when log levels are undefined, in the general case the logs will go to `LOG_LEVEL_INF`:
https://github.com/zephyrproject-rtos/zephyr/blob/bc5fbc929bb805e3c562c7280dd02a6f9f1f9087/include/zephyr/logging/log.h#L290-L293
https://github.com/zephyrproject-rtos/zephyr/blob/bc5fbc929bb805e3c562c7280dd02a6f9f1f9087/subsys/logging/Kconfig.filtering#L13-L19

**To Reproduce**

Modify `samples/bluetooth/scan_adv/prj.conf` with
```
CONFIG_LOG=y
# CONFIG_BT_DEBUG_LOG=y
```
`west build -b nrf52840dk_nrf52840 zephyr/samples/bluetooth/scan_adv/`

**Expected behavior**

Disabling `CONFIG_BT_DEBUG_LOG` should not result in more logs being printed.
I'm not sure if `CONFIG_BT_DEBUG_LOG` even makes sense now in the context of logging transitioning to standard `LOG_*` calls in #51897

**Impact**
Annoyance

**Environment (please complete the following information):**
 - Zephyr v3.3

alwa-nordic, 2023-03-14:
I confirm; this is undesirable behavior. The logging macros `LOG_INF` and the like should ultimately be controlled by the templated log configs. I suggest the following resolution:

- Remove `config BT_DEBUG`. Assume BT_DEBUG=y when removing uses.
- Move `config BT_DEBUG_LOG` out of `choice BT_DEBUG_TYPE`. `config BT_DEBUG_LOG` becomes the legacy symbol for templated log config "BT". Make `choice BT_DEBUG_LOG` select `DEPRECATED`.
- Change/define meaning of `choice BT_DEBUG_TYPE`  to "BT_MONITOR output destination".
- Change/define meaning of `config BT_DEBUG_NONE` to "No BT_MONITOR".

alwa-nordic, 2023-03-14:
I think we should just remove `config BT_DEBUG_LOG` as well. It has no purpose after `select BT_DEBUG` is removed.

closed 2023-04-04 by JordanYates