zephyrproject-rtos/zephyr #55434: ARC IDU is set incorrectly

jacoba-at, 2023-03-05:
I see 2 issues in drivers/interrupt_controller/intc_arcv2_irq_unit.c:
1. The function arc_shared_intc_update_post_smp is called also when CONFIG_SMP is off. I believe that the function and the following SYS_INIT should be surrounded by #ifdef CONFIG_SMP
2. Inside this function, the call:
       z_arc_connect_idu_set_dest(i, GENMASK(CONFIG_MP_NUM_CPUS, 0));
   should be replaced with 
      z_arc_connect_idu_set_dest(i, GENMASK(CONFIG_MP_NUM_CPUS - 1, 0));

  Right now, if you have 2 CPUs, the dest register in the IDU will be set to 0x7, instead of 0x3.

I'm running with a zephyr v3.3.0 code base.

henrikbrixandersen, 2023-03-07:
Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

evgeniy-paltsev, 2023-03-08:
Thanks for reporting, I'll check.

> Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from 

Currently there is enough information for investigate - so probaby it's not required for its issue :)

evgeniy-paltsev, 2023-03-08:
About the first point - it's kinda expected as can have ARConnect even on single-core system - in that case we still want to setup it even if SMP is disabled. Alternatively it can be multi-core system but with only one core enabled (and interrupts routed via ARConnect) - it can be used for testing purposes.

Do you have some issues with your setup due to this or it's just general notice?

jacoba-at, 2023-03-09:
The function arc_shared_intc_update_post_smp(), by it's name, and by it's SYS_INIT stage (smp), should get called once the kernel transits to SMP mode, yet it is called even when SMP is not configured. That looks like a bug to me.
It took me a couple of days to figure out why my ISR is missing interrupts when I'm running with a single core .
Only the simulation waves gave me the clue.

evgeniy-paltsev, 2023-03-20:
Hi @jacoba-at ,

That sounds to me like some misconfiguration.

The INIT_LEVEL_SMP doesn't exist if we have CONFIG_SMP disabled, see:
https://github.com/zephyrproject-rtos/zephyr/blob/f2d9b74da135bf0170dbe2109ad616543b9e529a/kernel/init.c#L69
https://github.com/zephyrproject-rtos/zephyr/blob/f2d9b74da135bf0170dbe2109ad616543b9e529a/kernel/init.c#L313

May I ask you to share your configuration (.config)?

Thanks!

jacoba-at, 2023-03-20:
[config.txt](https://github.com/zephyrproject-rtos/zephyr/files/11021221/config.txt)
added config file

evgeniy-paltsev, 2023-03-21:
Hi @jacoba-at ,

at least this combination from your .config doesn't seem to be correct:
```
CONFIG_MP_NUM_CPUS=2
CONFIG_MP_MAX_NUM_CPUS=1
```

This indeed may cause incorrect IDU setup if the code misses #55956 fix (but still this incorrect setup should happen if `CONFIG_SMP` is enabled)

jacoba-at, 2023-03-26:
Hi @evgeniy-paltsev ,
Regarding the SMP configuration: I checked that CONFIG_SMP is turned off in my config, and I changed the first line of the function arc_shared_intc_update_post_smp() to: 
  __ASSERT(z_arc_v2_core_id() == 2,

and indeed I got an assertion in this line.
This indicates to me that 
SYS_INIT(arc_shared_intc_update_post_smp, SMP, 0);
is not working as you described.
-Jacob

closed 2023-08-10 by github-actions[bot]