zephyrproject-rtos/zephyr #55438: The module for fetcher "http" could not be imported

cptjpk, 2023-03-05:
```
etching blob hal_espressif: /Users/***/zephyrproject/modules/hal/espressif/zephyr/blobs/lib/esp32c3/libbtdm_app.a
The module for fetcher "http" could not be imported (No module named 'requests'). This most likely means it is not handling its dependencies properly. Please report this to the zephyr developers.
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.11/bin/west", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/west/app/main.py", line 866, in main
    app.run(argv or sys.argv[1:])
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/west/app/main.py", line 111, in run
    self.run_command(argv)
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/west/app/main.py", line 385, in run_command
    self.run_extension(args.command, argv)
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/west/app/main.py", line 467, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/west/commands.py", line 194, in run
    self.do_run(args, unknown)
  File "/Users/***/zephyrproject/zephyr/scripts/west_commands/blobs.py", line 140, in do_run
    subcmd(args)
  File "/Users/***/zephyrproject/zephyr/scripts/west_commands/blobs.py", line 121, in fetch
    self.fetch_blob(blob['url'], blob['abspath'])
  File "/Users/***/zephyrproject/zephyr/scripts/west_commands/blobs.py", line 107, in fetch_blob
    fetcher = fetchers.get_fetcher_cls(scheme)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/***//zephyrproject/zephyr/scripts/west_commands/fetchers/__init__.py", line 43, in get_fetcher_cls
    raise ValueError('unknown fetcher for scheme "{}"'.format(scheme))
ValueError: unknown fetcher for scheme "https"
```

nordicjm, 2023-03-06:
Python 3.11 is not supported

KenthJohan, 2023-03-16:
> Python 3.11 is not supported

Which python version should I use?

CrispyCrafter, 2023-09-19:
Just install requests.

`pip install requests`

_The module for fetcher "http" could not be imported (No module named '**requests**'). This most likely means it is not handling its dependencies properly. Please report this to the zephyr developers_

nordicjm, 2023-09-19:
> Just install requests.
> 
> `pip install requests`
> 
> _The module for fetcher "http" could not be imported (No module named '**requests**'). This most likely means it is not handling its dependencies properly. Please report this to the zephyr developers_

Except that this is part of the requirements file https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/requirements-base.txt#L23, so if you get this error it means you haven't followed the setup guide